freerouting/freerouting #8: Use Oracle JDK

qume, 2017-01-24:
Buids and runs fine with Oracle JDK, but had problems under Ubuntu using openjdk (both 8 and 9)

felipesanches, 2017-02-03:
Or better: bugfix openjdk based on the aforementioned problems. By the way... "had problems" is far from guiding us towards a solution. If possible, please provide a more detailed description of the problems at hand so that someone can do something about it.

qume, 2017-02-06:
Understood. Java environment is something I have never had experience with so not much I can help with. The only insight I can add is that running java -jar ... resulted in a segfault. I don't have access to those machines right now but can note that this happened twice on two different machines with different distros for me around 3 months apart, so unlikely to be isolated to just my setup.

In any case I hope this note can save someone time. If there is a more prominent place to put it, perhaps in the readme markdown, it could help someone or better still maybe someone with the skills to fix openJDK as you say.

closed 2017-04-11 by qume