freerouting/freerouting #9: java.awt.geom.Rectangle2D$Double cannot be cast to java.awt.Rectangle

burggraf1966, 2017-05-22:
I copied that issue from [this site](https://github.com/Engidea/FreeRoutingNew/issues/5). I have the same issue on Mac OS X Sierra (but not on windows 10) with this version of freerouting. The patch proposed on that site worked for me on both OS. I suggest you include the patch as well. The attached file already contains the patches.
[GraphicsContext.java.zip](https://github.com/freerouting/freerouting/files/1020256/GraphicsContext.java.zip)

kjyv, 2017-09-06:
I've recompiled the jar with your changed file since the included binaries still don't work on macOS.

[freeroute.jar.zip](https://github.com/freerouting/freerouting/files/1282814/freeroute.jar.zip)

stanleyseow, 2018-01-13:
The freeroute.jar.zip works on Mac OS X, run with java -jar freeroute.jar 
Thanks !!!!

sherief, 2019-10-04:
This has been the only way I found to use FreeRouting on a Windows 10 machine. Thank you so much @jb-42 (and thanks @kjyv for providing a binary)!

rusefi, 2020-03-14:
@jb-42 I hope I've merged your fix

@miho I assume a new release is needed - there is at least this fix

rusefi, 2020-03-15:
@miho this did not make it into 1.4.1 right?

andrasfuchs, 2020-03-23:
@rusefi, no I didn't, it got pulled only yesterday: https://github.com/miho/freerouting/pull/23.

I'll try to solve the merging conflicts between this repo and miho's soon, so that we can have the most recent version here as well.

andrasfuchs, 2020-03-24:
We have a new release here: https://github.com/freerouting/freerouting/releases

Could someone with MacOS confirm please, that this bug got fixed?

focuzar, 2020-07-24:
[image]

I am getting this error when try to launch.

andrasfuchs, 2020-07-27:
Could someone with MacOS fix this and create a pull request?

luke-biel, 2021-01-15:
I've managed to go into `Freerouting.app/Contents/MacOS` and run `java -jar Freerouting` and it worked, no exceptions. It seems that there's issue in how the application is bundled for Osx.

rothn, 2021-09-05:
I am still seeing this issue in 1.4.4:

```
nrothlaptop-8:~ nroth$ java -version
openjdk version "11.0.11" 2021-04-20
OpenJDK Runtime Environment AdoptOpenJDK-11.0.11+9 (build 11.0.11+9)
OpenJDK 64-Bit Server VM AdoptOpenJDK-11.0.11+9 (build 11.0.11+9, mixed mode)
nrothlaptop-8:~ nroth$ java -jar /Applications/layout.app/Contents/MacOS/freeRouting.jar
Exception in thread "AWT-EventQueue-0" java.lang.ClassCastException: class java.awt.geom.Rectangle2D$Double cannot be cast to class java.awt.Rectangle (java.awt.geom.Rectangle2D$Double and java.awt.Rectangle are in module java.desktop of loader 'bootstrap')
	at boardgraphics.GraphicsContext.draw(GraphicsContext.java:131)
	at boardgraphics.GraphicsContext.draw_boundary(GraphicsContext.java:240)
	at boardgraphics.GraphicsContext.draw_boundary(GraphicsContext.java:255)
	at board.ComponentOutline.draw(ComponentOutline.java:146)
	at board.Item.draw(Item.java:849)
	at board.BasicBoard.draw(BasicBoard.java:1072)
	at interactive.BoardHandling.draw(BoardHandling.java:724)
	at gui.BoardPanel.paintComponent(BoardPanel.java:220)
	at java.desktop/javax.swing.JComponent.paint(JComponent.java:1074)
	at java.desktop/javax.swing.JComponent.paintChildren(JComponent.java:907)
	at java.desktop/javax.swing.JComponent.paint(JComponent.java:1083)
	at java.desktop/javax.swing.JViewport.paint(JViewport.java:737)
	at java.desktop/javax.swing.JComponent.paintChildren(JComponent.java:907)
	at java.desktop/javax.swing.JComponent.paint(JComponent.java:1083)
	at java.desktop/javax.swing.JComponent.paintChildren(JComponent.java:907)
	at java.desktop/javax.swing.JComponent.paint(JComponent.java:1083)
	at java.desktop/javax.swing.JComponent.paintChildren(JComponent.java:907)
	at java.desktop/javax.swing.JComponent.paint(JComponent.java:1083)
	at java.desktop/javax.swing.JLayeredPane.paint(JLayeredPane.java:590)
	at java.desktop/javax.swing.JComponent.paintChildren(JComponent.java:907)
	at java.desktop/javax.swing.JComponent.paintToOffscreen(JComponent.java:5262)
	at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBufferedImpl(RepaintManager.java:1643)
	at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(RepaintManager.java:1618)
	at java.desktop/javax.swing.RepaintManager$PaintManager.paint(RepaintManager.java:1556)
	at java.desktop/javax.swing.RepaintManager.paint(RepaintManager.java:1323)
	at java.desktop/javax.swing.JComponent.paint(JComponent.java:1060)
	at java.desktop/java.awt.GraphicsCallback$PaintCallback.run(GraphicsCallback.java:39)
	at java.desktop/sun.awt.SunGraphicsCallback.runOneComponent(SunGraphicsCallback.java:78)
	at java.desktop/sun.awt.SunGraphicsCallback.runComponents(SunGraphicsCallback.java:115)
	at java.desktop/java.awt.Container.paint(Container.java:2002)
	at java.desktop/java.awt.Window.paint(Window.java:3940)
	at java.desktop/javax.swing.RepaintManager$4.run(RepaintManager.java:876)
	at java.desktop/javax.swing.RepaintManager$4.run(RepaintManager.java:848)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:848)
	at java.desktop/javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:823)
	at java.desktop/javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:772)
	at java.desktop/javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1890)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Exception in thread "AWT-EventQueue-0" java.lang.ClassCastException: class java.awt.geom.Rectangle2D$Double cannot be cast to class java.awt.Rectangle (java.awt.geom.Rectangle2D$Double and java.awt.Rectangle are in module java.desktop of loader 'bootstrap')
	at boardgraphics.GraphicsContext.draw(GraphicsContext.java:131)
	at boardgraphics.GraphicsContext.draw_boundary(GraphicsContext.java:240)
	at boardgraphics.GraphicsContext.draw_boundary(GraphicsContext.java:255)
	at board.ComponentOutline.draw(ComponentOutline.java:146)
	at board.Item.draw(Item.java:849)
	at board.BasicBoard.draw(BasicBoard.java:1072)
	at interactive.BoardHandling.draw(BoardHandling.java:724)
	at gui.BoardPanel.paintComponent(BoardPanel.java:220)
[... 15 lines cut ...]
	at java.base/java.security.AccessController.doPrivileged(Native Method)
[... cut ...]
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)```

... Along with a very strange-looking window:

![image](https://user-images.githubusercontent.com/4665783/132144915-c247a867-20d1-44cd-b68b-561761c2cbb8.png)

Is it possible this got re-introduced at some point?

closed 2022-06-06 by andrasfuchs

phil123456, 2022-08-11:
just downloaded and installed layout-20220423-win-64bit-installer.msi

>java -jar freeRouting.jar
"class java.awt.geom.Rectangle2D$Double cannot be cast to class java.awt.Rectangle"

on Win10, used a dsn file created with Kicad

java version "17.0.2" 2022-01-18 LTS
Java(TM) SE Runtime Environment (build 17.0.2+8-LTS-86)
Java HotSpot(TM) 64-Bit Server VM (build 17.0.2+8-LTS-86, mixed mode, sharing)

phil123456, 2022-08-11:
ok I used jre 1.8, it works