freerouting/freerouting #13: Is this repo ready to be archived?

andrasfuchs, 2020-02-10:
As far as I can see this repo wasn't updated in a while.

Could we reach out to the owners and ask them to archive this repo as described [in this blog post](https://github.blog/2017-11-08-archiving-repositories/)? 

It would be also great to update the README, and let users know where the currently most active fork is:
https://github.com/miho/freerouting

rusefi, 2020-03-14:
I am sorry I have not noticed this until now. Will do!

rusefi, 2020-03-14:
Well, I might have lost access... Let me see if I can recover it.

abelom, 2020-03-14:
Great news - I have access apparently, I was able to edit the readme.md

I will now email you and maybe we can catch-up and I should transfer ownership to someone more involved?

rusefi, 2020-03-15:
Speaking about array of relevant repos - there is also the one by the original auther.

This repo was not forked from https://github.com/nikropht/FreeRouting/ because it seemed like SouceForge copy had newer version of the source code or something like that.

andrasfuchs, 2020-03-21:
It's great that we are aiming to merge all freerouting repos!

In which repo should I submit my latest PRs? This one or the one [miho has](https://github.com/miho/freerouting)?

rusefi, 2020-03-21:
I do not know

I think it really depends on who plans to be actively developing. I do not plan to be actively developing

It looks like mine insist on java 11 which I believe is a mistake. On the other hand since I do not plan to develop and there is a self contained installer it's probably fine.

You now have access to merge into this repo directly. I am totally fine if my most recent changes would be dropped - I guess unified fork is more important than java 8 support

andrasfuchs, 2020-03-21:
I'm planning to be involved with the project as a developer, if we can confirm that people actually use it. If switching to Java 11 would mean that we loose users, then I would prefer the older version, otherwise I would vote for the most recent version.

Btw, could we solve the Java 8 vs 11 dilemma with the multi-jar release feature?

https://blog.idrsolutions.com/2019/05/java-8-vs-java-11-what-are-the-key-changes/
> 3. Multi-jar releases: One of the features we tested out at the start of the year was the use of multi-jar releases. This feature provides a way for you to still code with the latest features of Java 11 while maintaining support and functionality for Java 8 clients. It’s certainly worth checking out.

I'm not sure how much complication would it mean for @miho though. 
@miho, what do you think about this solution?

miho, 2020-03-21:
I think the discussion is purely theoretical. There is no real dilemma. Users don't (and should not) care which Java version is used internally. They should not be forced to install anything in addition to Freerouting. Neither Java 8 nor 11. Native installers do exactly that. Nobody is excluded or left behind.

rusefi, 2020-03-21:
Just for the sake of being factually correct, 32 bit users are left behind :) But we can agree that we leave 32 bit users behind since those are only a few percent at this point :)

closed 2020-03-21 by rusefi

miho, 2020-03-21:
FYI we build 32 bit snapshot binaries for Windows ;)

rusefi, 2020-03-21:
Weird, I thought there is no 32 bit java 11