freerouting/freerouting #14: Making code compatible with java 8

rusefi, 2020-03-14:
Java 8 vs java 11 is a bit controversial - the fact that the industry choice to back-port fixed into 8 and provide support of java 8 speaks volume, also there is some licencing drama.

As long as it's trivial to do so I believe that it would be nicer to stay on java 8.

This could absolutely be revisited once/if there are additional arguments for java 9+.

miho, 2020-03-15:
Personally, I see no point in supporting Java 8 anymore. So you are definitely right about it being controversial ;) We plan to ship an application with a "minimal" runtime (in Java-terms, which means probably 30MB compressed). So there's no real Java 8 vs. Java 11 issue for regular end-users. 

Since I maintain a lot of different repositories and cannot provide support for issues with multiple JVM versions. And although they are rare, they still exist and are sometimes hard to find and fix. I will keep my code compatible with the latest LTS release of the Java platform if possible. Sticking to an older release would only be an option for me if there's a large group of end-users that I'd prevent from using the application at all.

rusefi, 2020-03-15:
Once this application is self-contained it can use any version of java and not care for the users, but as of today I did not have java 11 installed nor did my PCB contributer - we both only had java 8.

https://www.baeldung.com/java-in-2019 is probably not good user statistics but 80% usage of version 8 is a very serious which should not be ignored in my opinion.

miho, 2020-03-15:
I am well aware of the statistics but since we agree that a self-contained app is the best solution anyway, that's not an issue. In the meantime people can use older binaries if they are unable to update their installation of Java. As of now, changes between older and newer versions of freerouting are not that huge.

rusefi, 2020-03-15:
It is my experience that the target audience is very confused by "incompatible class version 53" message (from memory) - some of the target audience would miss the fact that JRE 11 is needed.

I would actually make an argument that current release title "JDK 11" instead of "Java 11" would confuse at least someone :)

Speaking of usability let me create a usability issue :)

rusefi, 2020-03-15:
"self-contained app is the best solution anyway" I could be missing something.

Is self-contained app available as of today? Does freerouting.exe require external JRE as of today?

miho, 2020-03-15:
I just uploaded an MSI-installer with an integrated runtime. The .EXE is not self-contained. I renamed it on the GitHub release page to indicate that. I will configure GitHub actions to automatically create release builds for Windows, macOS and Linux (including installers and/or app-bundles). This should simplify things for end-users.

rusefi, 2020-03-15:
Sweet stuff :)

When you say macOS I think you really want to look at https://github.com/freerouting/freerouting/issues/9

rusefi, 2020-03-15:
I've installed the amazing self-contained .MSI and JFrame title shows what seems to be wrong version number?

[image]

miho, 2020-03-15:
That's a bug. The version should automatically update. But I forgot to add that to the build script.

andrasfuchs, 2020-03-21:
Is my understanding correct that I can use Java 9+ constructs in the code, because you agreed to go with Java 11 and self-contained release files?

rusefi, 2020-03-21:
what specific constructs of java 9 are you interested to use? at the moment only "var" was used about 8 times.

i guess the question is: do we believe in this jar used as a library in a near future? if this jar would be used as a library maybe that's a reason NOT to force java 11 on consumers?

if this is always a standalone tool and now we have self-contained release I guess there are little reasons to keep java 8 compatibility.

andrasfuchs, 2020-03-21:
I can't say what I would actually use from the newer features, I worked mostly on C# projects in the past. 

So, I'm not really against Java 8, I just suppose that many improvements happened to Java (just like to C#) in the last 6 years, and it would be nice to benefit from them, that's all.

rusefi, 2020-03-21:
I am a professional java developer and things are a bit more complicated with java 8 to 11 migration. java 11 has removed a few things (which were probably supposed to be removed) and this has thrown many legacy projects into issues

also Oracle licensing has changed. it's a complicated matter and I would prefer not to debate it. Please make your own research on the java 11 controversy if you are interested to have your own opinion.

as a professional java developer I have not yet seen benefits which justify the drama. "var" syntax sugar cannot be considered worth all the negative consequences in my opinion.

miho, 2020-03-21:
The licensing issue is complex and a huge problem if users have to install their own JDK. This is unfortunately true for JDK 8 and 11 alike. So the real solution is to not force users to install Java at all.

rusefi, 2020-03-21:
https://github.com/miho/freerouting/issues/19 created

I think the focus should be on how to bring two forks together and if https://github.com/miho/freerouting plans to merge into https://github.com/freerouting/freerouting

My preference would be for @miho and @andrasfuchs to make this decision without me. Any decision you guys make work for me, just make a decision and implement it.

closed 2020-03-21 by rusefi

andrasfuchs, 2020-03-23:
@miho, I merged my last PR from my own repo today, and I'm not planning to keep my repo anymore.
Would you also consider completely moving the our upstream repo (this), since we are both collaborators here?

I'm going to resolve the merging conflicts soon, so when yours and this get into sync, we could theoretically stop commiting to miho, and move here. What do you think?

miho, 2020-03-23:
Sounds good.