freerouting/freerouting #15: StackOverflowException with attached DSN

rusefi, 2020-03-14:
@miho @andrasfuchs do you recognize this issue by any chance?

[stackOverflow.zip](https://github.com/freerouting/freerouting/files/4333415/stackOverflow.zip)

[image]

[image]

abelom, 2020-03-15:
And the way we got this .DSN is exactly matching #4 problem statement!

rusefi, 2020-03-21:
@andrasfuchs @miho any interest to look into this bug? My PCB engineer is affected :(

andrasfuchs, 2020-03-21:
I'm already working on it ;) 
I have a working solution at the moment that I'm planning to create a PR of in an hour or so.

andrasfuchs, 2020-03-21:
Here is my branch with the fix for this error, but it needs some testing. Could you try it?
https://github.com/andrasfuchs/freerouting/tree/issue-15-fix

I run it with the file you provided and as you will also see in the logs, there is something strange going on with a few nets. Namely "Net-(P1-Pad36)", "Net-(P1-Pad39)", "Net-(P1-Pad40)" and "Net-(P1-Pad44)".

Normally the method that caused the StackOverflowException normalizes traces, breaks them up and combines them in a way. 
I'm not 100% sure why it must run when we load a DSN file, but it was there, so I left it there.

Since I detect endless recursive loops now and I break the loop during this normalization, I'd like to ask you to double-check if the nets in question look good to you after loading your file. If they are ok, please confirm, and then we can accept this fix. If something looks fishy, I will need to come up with another solution.

rusefi, 2020-03-21:
Can you please attach a .jar file here? I am just the messenger. I can forward the .jar file to my PCB engineer. and since that would not be a self-contained archive you know... that little thing.

andrasfuchs, 2020-03-21:
I think it's a not a good practice to publish executables as attachments in issues, so I would like to ask you (or your college) to download the branch I mentioned above, and follow [the instructions](https://github.com/freerouting/freerouting#how-to-build-it) and use Gradle to compile the code into a .jar.

Let me know if you have any troubles with it, so we can find a workaround next week.

rusefi, 2020-03-22:
That's fair, let me cook the jar and report the results!

rusefi, 2020-03-22:
Proposed behavior abort the process, not really solves the root cause of why it fails:

```2020-03-21 20:36:03.110 [AWT-EventQueue-0] ERROR eu.mihosoft.freerouting.FreeRouting - The normalization of net 'Net-(P1-Pad36)' failed.
java.lang.Exception: We reached the maximum normalization depth (16).
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:804) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:835) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:835) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:835) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:835) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:835) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:835) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:835) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:835) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:835) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:835) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:835) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:835) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:835) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:835) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:835) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:835) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:835) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.PolylineTrace.normalize(PolylineTrace.java:799) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.board.BasicBoard.normalize_traces(BasicBoard.java:873) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.designforms.specctra.Wiring.read_scope(Wiring.java:116) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.designforms.specctra.ScopeKeyword.read_scope(ScopeKeyword.java:113) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.designforms.specctra.DsnFile.read(DsnFile.java:88) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.interactive.BoardHandling.import_design(BoardHandling.java:1053) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.gui.BoardFrame.read(BoardFrame.java:180) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.gui.MainApplication.create_board_frame(MainApplication.java:359) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.gui.MainApplication.open_board_design_action(MainApplication.java:305) ~[freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.gui.MainApplication.lambda$new$2(MainApplication.java:244) ~[freerouting-executable.jar:unspecified]
	at javax.swing.AbstractButton.fireActionPerformed(Unknown Source) [?:1.8.0_211]
	at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source) [?:1.8.0_211]
	at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source) [?:1.8.0_211]
	at javax.swing.DefaultButtonModel.setPressed(Unknown Source) [?:1.8.0_211]
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source) [?:1.8.0_211]
	at java.awt.AWTEventMulticaster.mouseReleased(Unknown Source) [?:1.8.0_211]
	at java.awt.Component.processMouseEvent(Unknown Source) [?:1.8.0_211]
	at javax.swing.JComponent.processMouseEvent(Unknown Source) [?:1.8.0_211]
	at java.awt.Component.processEvent(Unknown Source) [?:1.8.0_211]
	at java.awt.Container.processEvent(Unknown Source) [?:1.8.0_211]
	at java.awt.Component.dispatchEventImpl(Unknown Source) [?:1.8.0_211]
	at java.awt.Container.dispatchEventImpl(Unknown Source) [?:1.8.0_211]
	at java.awt.Component.dispatchEvent(Unknown Source) [?:1.8.0_211]
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source) [?:1.8.0_211]
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source) [?:1.8.0_211]
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source) [?:1.8.0_211]
	at java.awt.Container.dispatchEventImpl(Unknown Source) [?:1.8.0_211]
	at java.awt.Window.dispatchEventImpl(Unknown Source) [?:1.8.0_211]
	at java.awt.Component.dispatchEvent(Unknown Source) [?:1.8.0_211]
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source) [?:1.8.0_211]
	at java.awt.EventQueue.access$500(Unknown Source) [?:1.8.0_211]
	at java.awt.EventQueue$3.run(Unknown Source) [?:1.8.0_211]
[... cut ...]
	at java.awt.EventQueue.dispatchEvent(Unknown Source) [?:1.8.0_211]
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source) [?:1.8.0_211]
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source) [?:1.8.0_211]
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source) [?:1.8.0_211]
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source) [?:1.8.0_211]
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source) [?:1.8.0_211]
	at java.awt.EventDispatchThread.run(Unknown Source) [?:1.8.0_211]
```

andrasfuchs, 2020-03-22:
Don't worry too much about the exception details in the log, it only indicates that the normalization process got interrupted because of the potential endless loop condition. 

What was the actual result from the user's perspective? Did you still get the StackOverflowException? Did you manage to open the file that you couldn't before? Were you able to do the routing on the file? Is there anything else that blocks your/your engineer's work now?

If you really want to go deeper, you can do your own investigation: what are the differences between the problamatic nets I listed above and the ones that can be loaded without problems?

rusefi, 2020-03-22:
@jharvey says "autorouter has worked with this newer JAR file and the original DNS file." - i.e. the fix has worked and output is still produced as expected

is there a way to hide scary stacktrace from strout? it should probably be a less scary one like INFO line

is this worth a unit test?

andrasfuchs, 2020-03-22:
Alright, great, thank you for the test!

Sure, sorry, I forgot to change the logging level for the stdout, I thought that that exception is only visible in the more detailed log file. I'll fix that in the next release.

The whole project would need a good unit testing framework, and many unit tests should be created, I'm just not sure I'm up to the task. 
Nonetheless I think it's worth creating an issue here on GitHub about unit tests, so someone else can grab it and start working on it if they feel like it.

rusefi, 2020-03-22:
from my experience it's impossible to just create all unit tests in a legacy project

on the other hand creating them one by one at least every time an issues is witnessed gives a ray of hope

andrasfuchs, 2020-03-22:
Sounds good to me, feel free to create a PR with the unit test you have in mind. 
I already added the DSN file into the [tests folder](https://github.com/miho/freerouting/tree/master/tests) so you can use it for the unit test if you wish.

rusefi, 2020-03-22:
Паяльная станция.dsn and Pajalnaja_stancija.dsn is this the same file? the name is the same in Russian

andrasfuchs, 2020-03-22:
Yes, both would make a good unit test, one for the cyrillic file names, the other for the cyrillic characters inside the file.

rusefi, 2020-04-04:
Let's keep this issue open until public binaries have the fix. It looks like latest 1.4.2 has the same (or different? stack over flow with the same input file.

rusefi, 2020-04-04:
@andrasfuchs poke

andrasfuchs, 2020-04-06:
@rusefi I think you're right, the fix is only included in the snapshot build if I remember correctly.

andrasfuchs, 2020-04-08:
@miho Could you create a new, 1.4.3 release for us in this repo please?

miho, 2020-04-18:
Just go to the release page and create a new one. Installers are built automatically.

closed 2020-05-14 by andrasfuchs