freerouting/freerouting #16: is there anything to merge from FreeRoutingNew

rusefi, 2020-03-15:
https://github.com/Engidea/FreeRoutingNew has a lot of commits but I am not sure if it's even possible now to bring all versions of freerouting together

andrasfuchs, 2020-03-23:
It's a shame that it was technically not forked from this repo, looking at their commits, it seems like @Engidea put in a lot of good work into the project. 
He kept the source structure, so it might be still possible to at least try to merge his work, I'll look into this later this week.

Is there a chance that we can invite @Engidea to this repo too?

rusefi, 2020-03-23:
there must be a way to grab commits as patches and to rebase on this repo as long as it's more or less the same files originally?

Engidea, 2020-03-23:
There is a fundamental issue with freerouting that is bound to the use of rational numbers to describe the "lines". 

I have managed to reduce as much as possible the use of the rationals but completing the step is really difficult due to some weird relationship between the classes on shapes.

I have been thinking of using another logic to describe the layout, a logic where you describe "tensions" between nodes, but it is a major rewrite.

The reason I did start the "adjust" was to be able to export/reimport freecand from Kicad. Something that it is now actually reasonably possible. What happens is that when you export to Kicad then the rational points get "rounded" and that is kind of fine with Kicad but a mess when you import them back into freeroute.

I can share with you some more insight if you wish.

andrasfuchs, 2020-03-24:
To be honest, I don't think anyone would start a bug rewrite of freerouting. We have to deal with what we got.

What do you think, are your improvements still relevant, should we spend some time with trying to include your code into the current codebase?
Do I understand it correctly that the export/reimport problem is kind of solved now with the newer version of KiCad and FreeCad?

If not, and if we wanted to reproduce the problem what would be the steps? I understand that you imported and exported files among KiCad, FreeCad and freerouter, but what was the order and when did you first see the problem?

Thanks!

Engidea, 2020-03-25:
The issue is only relevant if you keep going back - forth using Kicad - freeroute

If you stay in Kicad or stay in freeroute, you are fine.

Also, I do not really use autoroute since most of my work is using just two sides PCB. One side is groundplane and the top side is "connections". With this logic, routin is manual since there is no autoroute that is smart enough to avoid chopping the groundplane.

If you do not have this requirement, then, autoroute is fine.

Also, another issue that I had to deal with is freerouting crashing or better, avoid crashing.
The origianl version had quite some unlimited recursion issues that I had to stop in some way. 
Not that what I did is optimal, but the way the original code is assumes that there is always a solution to some calculation and that is not always the case, even with big numbers involved and rationals. (close to infinity is a difficult concept to code with, I said "close")

So,  freeroute is magic, when it works, but with Kicad having some sort of push and shove routing...
I do not know.... difficult.

Also I did LOT of cleaning up on the gui and classnames, basically the main problem is that core classes had the same name but different package and it is really confusing to deal with.

I am willing to chat about the code, if you wish...

rusefi, 2020-03-25:
We cannot apply changed from https://github.com/Engidea/FreeRoutingNew directly since folder structures are different between these two repositories, on the other hand the changes are DEFINITELY worth applying! For example, here is the first change in FreeRoutingNew - https://github.com/Engidea/FreeRoutingNew/commit/46ccc61a3c73c55562292c6019ce5941a808ebe8 (gui help extracted into separate class, resource files for help)

I think we need some scripts to help bring those two forks together and then see what to cherry-pick.

here's an idea:
1) for each FreeRoutingNew commit make it a patch file
2) automatically edit paths in those patch files
3) apply those patches to a branch of freerouting
4) start picking changes in chronological order, looks like some are DEFINITELY worth picking!!!

@Engidea would you by any chance be interested to create these scripts to bring your changes to this repo?

Engidea, 2020-03-28:
The logic of the application is quite complex, ALfons is surely quite smart.

There are quite a few "caching" classes that make the actual working obscure

The above to say that if there is something that does not work.... finding why is REALLY complicated. (this was part of my effort, reduce the number of "shapes" and reduce the complexity of the logic, beside what I wrote on my previous post.

There is no actual consistent way to do regression testing since the behaviour really depends on the board you have loaded.

I really doubt that you can "merge" my changeset back into Freeroute, you would end up with Freeroutingnew, you may as well just pick up Freeroutingnew, I clearly have no problem into either giving you commit privilege or you could just grab the source.

Before doing anything, you should state your goals. 

My end goal was to have the "points" of Freecad lines always fall into a 1micron grid, and thus remove the need for rationals. This should increase the reliability since it should avoid having lines that intersect "close to infinity ahead", it should also allow for reliable bach-forth with Kicad.

Doing it means having a firm grasp of the logic, something that for some reason I did not have time to fully complete.

I can share what I found out, if you wish.

closed 2022-06-06 by andrasfuchs