freerouting/freerouting #17: Release files usability

rusefi, 2020-03-15:
Typically an end-users of any tool has to download one most important for end user file.

In our case we have freerouting.exe and what else? Which jar(s)? I would suggest packaging all the files needed to the end user into one .zip for a release.

To be honest I am a java developer yet I had to google the difference between freerouting.jar and freerouting-executable.jar myself - that's a gradle naming convention for "-with-dependencies" vs "-no-dependencies".

What is the workflow to create freerouting.exe since we no longer have it in repository?

miho, 2020-03-20:
**Update:** there are native packages for all major platforms. They are built via GitHub Actions and uploaded automatically whenever a release is created on GitHub. The freerouting-executable.jar is now merely a fallback for platforms/architectures we don't support officially.

miho, 2020-03-20:
snapshots are provided in a "SNAPSHOT" release: https://github.com/miho/freerouting/releases/tag/SNAPSHOT

rusefi, 2020-03-21:
https://github.com/miho/freerouting has great usability, this issue is closed

closed 2020-03-21 by rusefi