freerouting/freerouting #20: Windows dpi scaling goes wrong

markuckermann, 2020-03-24:
I hope this is the correct repo for issues!

After starting freerouter it looks normal but as soon as I click on "Open your own desgin", everything is twice the size it should be. Possibly something to do with having a 4k high dpi screen but the same thing happens on my second screen (1080p). 

[image]

After clicking: 
[image]

This leaves a lot of text invisble or hard to read: 

[image]

Cheers to all who keep working on this!

markuckermann, 2020-03-24:
**Workaround** 

If you specify the .dsn file via the command line it works well.
`\AppData\Local\Freerouting\Freerouting.exe -de '.\path_to\file.dsn'` 

This opens the dsn file immediately and the scaling issue doesn't appear.

andrasfuchs, 2020-03-24:
Thank you @markuckermann for reporting this!

Could you share your Windows _Scale and layout_ settings with us?

[image]

Do you have anything set in the _Advanced scaling settings_?

[image]

markuckermann, 2020-03-26:
@andrasfuchs yes I can:
[image]

closed 2022-08-01 by andrasfuchs