freerouting/freerouting #22: Auto-router Interrupted error and failed to auto-route.

jharvey, 2020-04-05:
When I click the auto-router button, I get the below, and the auto-router simply fails to try and work.
[image]

This DSN was created by inserting several footprints on a layout. Then auto-route, then desire to make changes. The changes needed to delete several traces and moved components. Then this DSN was created and used to attempt a re-auto-route. 

I'm doing this by launching freerouting-executable.jar which notes 1.4.0 in the top left of the dialog box. However this is a special build which was provided to me via Andrey (rusEFI). This version allows opening the file. The official builds fail with a stack overflow error. 

I tried the official builds posted at the below link including 1.4.0, 1.4.1 and 1.4.2. All of the official builds simply say stack overflow. This only opens with the .jar that Andrey provided to me. 
https://dl.bintray.com/miho/Freerouting/eu/mihosoft/freerouting/freerouting/  

This is Win10 64bit, 
C:\>java -version
java version "13.0.2" 2020-01-14
Java(TM) SE Runtime Environment (build 13.0.2+8)
Java HotSpot(TM) 64-Bit Server VM (build 13.0.2+8, mixed mode, sharing)

[Issue_AutoRouter_interupted.zip](https://github.com/freerouting/freerouting/files/4432863/Issue_AutoRouter_interupted.zip)

rusefi, 2020-04-05:
@andrasfuchs FYI

andrasfuchs, 2020-04-06:
Thank you for your report!

I can confirm that AutoRouter has some difficulties with the DSN file provided, but the good news is that the latest version could do the routing anyway for me for ~10 minutes without the "autoroute interrupted" error. 

I used the following command line arguments to start the autorouting:
`freerouting-executable.jar -mp 16 -de .\tests\Issue22-AutoRouter_interrupted.dsn -do .\tests\Issue22-AutoRouter_interrupted.ses`

I recommend you to try the snapshot version here: 
https://github.com/freerouting/freerouting/releases

We should keep this issue open until we have a proper fix though.

jharvey, 2020-04-06:
When I try the below it very quickly displays the last line, I see a very brief blink of what is perhaps the java launching. I do not get a ses file. See attached log. 
[freerouter.zip](https://github.com/freerouting/freerouting/files/4437692/freerouter.zip)
 
This is from the cmd. 

```
C:\blah\freerouting\Specail_build>freerouting-SNAPSHOT-2020-04-06_09-06-11-executable.jar -mp 16 -de .\tests\Issue22-AutoRouter_interrupted.dsn -do .\tests\Issue22-AutoRouter_interrupted.ses

C:\blah\freerouting\Specail_build>
```

I got my snap shot jar from the below link. 
https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT

jharvey, 2020-04-06:
I see the last line of the log notes failed to find location, so I tried the below and it seems to be working. AKA ./ vs ../

`C:\Users\Owner\Desktop\freerouting\Specail_build>freerouting-SNAPSHOT-2020-04-06_09-06-11-executable.jar -mp 16 -de ..\tests\Issue22-AutoRouter_interrupted.dsn -do ..\tests\Issue22-AutoRouter_interrupted.ses`

jharvey, 2020-04-06:
I can confirm this has worked, it successfully auto-routed the dsn and has imported into KC and seems to be working as expected. Thanks for the solution to this issue. Let me know if there is something I can do to help with a permanent solution.

closed 2020-05-09 by andrasfuchs