freerouting/freerouting #23: The future of freerouting with deep learning

andrasfuchs, 2020-04-07:
Hi guys,

I just wanted to start a discussion about the vision of freerouting.

Yesterday I watched [Lex Fridman's interview with David Silver](https://www.youtube.com/watch?v=uPUEq8d73JI). It's absolutely mind blowing what potential deep learning and AI has.

They got me inspired, and I would be interested in your feedback.

Very simply put, we could use state of the art, open-source AI to do the routing in freerouting. 
In order to do that we would need many steps (meaning many developer-hours of work), but it's not a totally unrealistic goal, and I think it would significantly better the end result.

We would need to design a very good reward function (board evaluation scoring system), define the rules of the "game of routing", and build at least some prediction model. 
The first two of these are somewhat present already in freerouting, but we would need to separate these modules from each other.

If we could pull this off, we could train our AI on GPUs by letting it play against the current algorithm, and learn from it in a natural way and then compete with itself to further improve its results. 
If we could build a good model this way, the end user wouldn't need too much computing power to get good results.

I'm not sure how much time I can put into this project in the future, because one of my project just won a sponsorship a few days ago, and it needs my full attention now, but I would be glad if someone with more experience with deep learning would join our discussion and efforts here, so please don't hesitate to give some feedback, if you feel exited like I do!

JustinEldracher, 2020-04-18:
Hello, unrelated to the current post, but I couldn't see a better place to put this:

THIS SOFTWARE IS AWESOME!!! 👍 👍 👍 👍 👍

I use KiCad for occasional PCB projects, and for my latest ones I really didn't want to route all the traces by hand.  So I did a quick search for a automatic routing software, found yours, installed it, and presto!!  I had fully routed PCBs in no time!!  The note in the Readme about how to use it with KiCad was very helpful!  I'm hooked, I don't think I'll ever route traces by hand again!!  This package has saved me hours of time!!  Great work everyone involved, keep it up!! 👍 👍 👍 👍

andrasfuchs, 2020-04-18:
Thank you @JustinEldracher for your feedback!

I joined to the project only recently, but to be honest I'm hesitant to invest more time into this since I don't know how many people use it at the moment and how important is it for their projects.

The feedback like yours inspires me to spare more time for it in the future ;)

oektomo, 2020-05-09:
Hi There,
I have an EE background but doing a lot in soft dev and still learning in statistic.
Just today I browse about AI autorouter and found out about:
deeppcb.ai
There is discussion in:
https://news.ycombinator.com/item?id=21660365
https://forum.kicad.info/t/tensorflow-based-autorouter/15844
some story here:
https://spectrum.ieee.org/tech-talk/computing/hardware/startup-jitx-uses-ai-to-automate-complex-circuit-board-design

I really surprise how much resources deeppcb have. With open sources mind If you want to take AI to do routing in PCB it will be long road ahead. For a long time, I do want to contribute in some open-source project but with my knowledge and experience don't really know which suit me and my mediocre level. I want to help but can't decide where to start. Can you help me getting started?
Thanks.

andrasfuchs, 2020-05-09:
@oektomo , thanks for the links! 
I think it's a great that we have DeepPCB as an option and I'm really rooting for them.

Most of the comments by professionals on the ycombinator forum were a little arrogant for my taste, but we'll see, I hope AI will prove them wrong and it will help us make better PCB boards quicker, even if you are just an open-source hobbyist (like many of us).

Anyway, I watched [DeepPCB's 1-hour-long introduction](https://www.youtube.com/watch?v=Ea5i-l8YKQo), registered, and I will test their service soon. They are aiming to solve both the routing and the placing problem, I'm impressed!

@oektomo , I'm really glad you're entering the open-source world, I was in your shoes a few years ago. I can recommend you [a video about Max, an open-source developer](https://www.youtube.com/watch?v=ifq3xhik8tE) as an inspiration. 

Feel free to reach out to me privately if you think I can help with your efforts, I've learned a few things about open-source collaborations in the recent years ;)

oektomo, 2020-05-11:
Couldn't agree more about the comment on ycombinator. I was a bit down to read such comment. But after reading your response, I think it's worth to make freerouting, AI capable.

It's a good video about Max since he quit college twice and keep working and pursuing what is good for him. It's not only inspire for entering open-source developer but also for keep working consistently.

For start, I already forking the freerouting repos. I think I'll start with small tweak in the UI. Since that's what itching me when I use freerouting since a couple years ago. Maybe when I used with the coding style and understand the data structure I'll get started with some AI tinkering. What do you think?

I wonder how can I reach out to you privately, perhaps email?

andrasfuchs, 2020-05-11:
I think your plan sounds great, work on what you are passionate about, that's the most important :) 
You can also see the planned to-dos listed here: https://github.com/freerouting/freerouting/projects/1

Have you worked with GitHub pull requests before, or do you need a quick introduction how to propose your changes for freerouting?

Sure, you can see my e-mail on my GitHub profile.

oektomo, 2020-05-11:
It's a good to-dos listed there, I was skimming it before you point it out. But after read it carefully, I think the multi thread things really interest me. I'll start it when I'm done trying out with my plan, the UI. 

No, I haven't worked with GitHub pull requests. it's good if you can point out any tutor or link.

I'll e-mail you when the time's come.
Thanks.

andrasfuchs, 2020-05-11:
I run tests in the last 24 hours to compare the current [DeepPCB](https://deeppcb.ai/) and [freerouting v1.4.4](https://github.com/freerouting/freerouting/releases/tag/v1.4.4).

I used one of my projects, the [Bio Balance Detector](http://www.biobalancedetector.com/) Mars-64 rev. E PCB for the test. I chose this because I knew that it is not an easy routing, but it can be completed on 2 layers.

[image]

Freerouting completed the task in 38 minutes on my PC, all 106 connections connected, no DRC errors.

DeepPCB generated the first 3 partial solutions in 11 minutes, with 77 connected of 106 connections. 
It stopped after nearly 24 hours with the final result, 82 connected of 106, no DRC errors.

andrasfuchs, 2020-05-11:
@oektomo the process isn't too difficult, but it wasn't obvious for me when I first started doing open-source development, so don't get discouraged if something goes sideways.

It's actually kind of a coincidence that I just wrote a small summary of how to create a pull request on GitHub in another repo, [please check out my comment there](https://github.com/restsharp/RestSharp/issues/1452#issuecomment-626584425), and let me know if something isn't working for you.

lynchaj, 2021-08-06:
Hi
Years ago, I recall corresponding with Anthony Blake who wrote the Topological Autorouter

https://geda-dev.seul.narkive.com/gWNGZKO5/gsoc-project-topological-autorouter

This is an extremely interesting project and could produce some truly amazing results.  I think remnants of the program are still on the internet although probably difficult to find these days.

https://github.com/russdill/pcb/blob/master/src/toporouter.h

I recall seeing the Topological Autorouter using what looked like spline curves for traces and for the ability to minimize vias and overall trace length on PCBs that would confuse FreeRouting

https://github-wiki-see.page/m/bert/pcb/wiki/Autorouters:-gEDA-pcb-Toporouter

Unfortunately I can no longer find the pictures which go with the 2009 webpage.  They were truly amazing

https://web.archive.org/web/20100914113959/http://anthonix.resnet.scms.waikato.ac.nz/toporouter/detour.html

I think that if these algorithms could be incorporated in FreeRouting that it would improve its ability to route difficult boards.  Also smoothly curving traces would have electrical benefits and generally have a more pleasing appearance

closed 2022-05-12 by andrasfuchs