freerouting/freerouting #24: Binaries are not generated for the release

andrasfuchs, 2020-04-19:
Binaries are not generated for the release after it was created here, in this repo.

I got the following error:
[image]

andrasfuchs, 2020-04-19:
I don't have the error now, but the binaries are not generated, they are not listed by the release.

[image]

miho, 2020-04-19:
Looks like you didn't trigger the create-release workflow this time. Usually, this happens if the release wasn't fully deleted (including associated tags). See the test release [v0.0.0](https://github.com/freerouting/freerouting/releases/tag/v0.0.0) for which installers are generated. To see which workflows are/were running, see https://github.com/freerouting/freerouting/actions. BTW. documentation for GH Actions can be found here: https://help.github.com/en/actions

andrasfuchs, 2020-04-19:
Great, thank you for the details. I needed to delete the tag as well and recreate the release. 
It looks good now.

closed 2020-04-19 by andrasfuchs