freerouting/freerouting #25: Release v1.4.4 still has snapshot in version string

miho, 2020-04-19:
BTW it seems like version 1.4.3 has been skipped accidentally.

andrasfuchs, 2020-04-20:
Alright, I will not fix this for this release (v1.4.4), only for the next.

miho, 2020-04-20:
@andrasfuchs that's fine. Just keep in mind that the word SNAPSHOT is added as suffix to the version number which is set to the version currently under development. Simply put, `snapshot-version > current version` (see [maven docs](http://maven.apache.org/guides/getting-started/index.html#What_is_a_SNAPSHOT_version)). When you release a Java library/application you remove the word snapshot from the version string and tag/create the release. Then you start the release process. After you tagged/released the version you'd increase the version number and add the "-SNAPSHOT" suffix to the version string.

closed 2020-05-09 by andrasfuchs