freerouting/freerouting #26: Autoroute interrupted, connections not found

FXRer, 2020-05-07:
Hello,

I'm using freerouting 1.4.4 and I had successfully autoroute my small design exported from Kicad 5 with freerouting. However, after rotating one of the component in the same design, freerouting does not continue to autoroute and says: "Autoroute interrupted, xx connections not found" :

[image]

I wonder where is the problem, please ?
Thanks and best regards,
Pascal.

andrasfuchs, 2020-05-07:
Hi @FXRer,

Thank you for using freerouting. 
Could you share you DSN with us to reproduce the problem?

When I had a similar problem, I forgot to remove the already wired tracks in KiCad after I rotated the component. Did you remove all wired tracks and vias in Pcbnew before exporting the Speccra DSN with the rotated component?

You can do that easily with the Edit / Global Deletions command
[image]

If you still have the problem, please include the last few lines from your log file as well.

FXRer, 2020-05-07:
Dear @andrasfuchs 
Honestly, I had successfully autorouted the 1st version of .kicad_pcb of my design. I then recognized that a component needed to be rotate. I rotated it, updated DSN without removing the already routed tracks. I had the problem in this case. 
Then, I wen back to the .kicad_pcb file, unrouted (deleted) all the routed tracks, updated the DSN and had freerouting to autoroute again. But I still had the same problem.   

My last tentative was removing the .kicad_pcb file, re-created a new one, rotated the component in question and updated the DSN. But, the problem still persist.

As your request, I upload the DSN of my design for debugging. 

[e-con_STEEReoCAM_Nano_FFC_MiniSAS_Adapter.dsn.zip](https://github.com/freerouting/freerouting/files/4594535/e-con_STEEReoCAM_Nano_FFC_MiniSAS_Adapter.dsn.zip)

Best regards,
Pascal.

andrasfuchs, 2020-05-07:
Hi @FXRer,

I have just tested your file with the following command line arguments:
`-mp 16 -de Issue26-e-con_STEEReoCAM_Nano_FFC_MiniSAS_Adapter.dsn -do Issue26-e-con_STEEReoCAM_Nano_FFC_MiniSAS_Adapter.ses`

The process was finished, and I got this in my log file:
> 2020-05-07 23:33:33.069 [Thread-0] TRACE eu.mihosoft.freerouting.FreeRouting - Method 'BatchAutorouterThread.thread_action()' was performed in 24.16 seconds.
> 2020-05-07 23:33:33.070 [Thread-0] INFO  eu.mihosoft.freerouting.FreeRouting - Saving '.\tests\Issue26-e-con_STEEReoCAM_Nano_FFC_MiniSAS_Adapter.ses'...

It looked just fine, but I couldn't import the .ses file into KiCad, I got the following error:
[image]

Then I tried to open the .dsn file on freerouting, and saved the .ses file without doing the autorouting, and I got the same error.

It looks like a bug, I need to investigate a little further, but I'm not sure when I'll have some spare time for it. 

Meanwhile, do you have any guess, why there is a problem with J2 (and not the other parts)? 
Does it look okay to you in your design?
Is this the component that you rotated?

FXRer, 2020-05-08:
Hi @andrasfuchs,
I think I resolved the problem. It did not come from the component rotation but from unused pads of J2. I updated the Kicad schematic by adding "no connection flag" for unused pads of J2 then re-ran the DRC of the schematic to make sure there's no significant error. I then re-created the .kicad_pcb file, exported the DSN and freerouting autorouted my design without any problem.

For this ticket, I think it can be closed but if you need more information for debugging, I will find a time slot to re-produce the problem.

Regards,
P.

andrasfuchs, 2020-05-08:
Alright then, thank you for your feedback. I'm glad it worked out for you in the end.

I'll close this ticket now, but I uploaded your DSN file into our test-case folder, so that I can get back to it, if time allows it. 

I'll let you know if I need more assistance with this, thank you for your offer.

closed 2020-05-08 by andrasfuchs

Pontifex42, 2021-04-07:
Got the same issue. And found a solution: there is a separated ".rule" file beside the dsn file. This file has been created at the "Export Spectra Session" time at a previous autorouting. Deleting this file solves the issue.
Obviously, the rule-file is used without further notice during opening a dsn file. If it is corrupt or does not fit to dsn file .... thats it.

shenjicai, 2022-07-07:
Anyone who encounters this problem should double check the DRC to see if the rules are set incorrectly causing the software to fail to automatically wire.