freerouting/freerouting #27: AutoRouter Interrupted does not route.

jharvey, 2020-05-14:
This file was routed, then changed, then traces deleted, and now again auto-routed. However this second route is failing to route. 

* [ ] What version of freerouting are you using?
freerouting-1.4.4-executable.jar
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
Win10 64 bit prof. 
* [ ] What type of file are you working with? Can you attach the actual (DSN) file?
[zMRETestFixture.zip](https://github.com/freerouting/freerouting/files/4625685/zMRETestFixture.zip)
* [ ] Did you use any command line arguments?
No I did not. 
* [ ] Can you post a screenshot of the problem?
[image]
* [ ] Can you include the latest, related log entries from the .log file?
[freerouter.log](https://github.com/freerouting/freerouting/files/4625679/freerouter.log)
* [ ] Please list the detailed steps to reproduce the problem
Open DSN, click OK on warning about rules, click autoroute button. 
* [ ] What would you expect freerouting to do in this situation?
I would hope it routes the traces like it normally does on a new file. 
* [ ] Is fixing this issue critical for your work?
I can hand route, it would be really helpful to autoroute. I have deadlines, this will require lots of work for me to obtain the dead lines.

andrasfuchs, 2020-05-14:
Thank you @jharvey for the detailed report.

I looked into it, and the routing seems to be working on my machine with v1.4.4. Looking at your log file and the actual source code of freerouting, my guess is that something went wrong in your .rules file.

Do you have custom rules defined for this board? Could you share the zMRETestFixture.rules file too?

Could you try to rename your zMRETestFixture.rules file next to the zMRETestFixture.dsn file to something else, and rerun the autorouter? Do you still have the same problem?

jharvey, 2020-05-14:
The freerouting log gives warning that apparently are from the .rules file. In KiCAD I deleted a bunch of unused net classes, then re-exported and now its routing.

closed 2020-05-14 by andrasfuchs