freerouting/freerouting #29: AutoRouter Interrupted does not route. (perhaps continued from #27)

jharvey, 2020-05-21:
I thought issue #27 was good, but I'm having trouble again. 

This file was routed, then changed, then traces deleted, and now again auto-routed. However this second route is failing to route. This appears to be a .rules file issue. 

* [ ] What version of freerouting are you using?
freerouting-1.4.4-executable.jar
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
Win10 64 bit prof. 
* [ ] What type of file are you working with? Can you attach the actual (DSN) file?
See attached DSN and .rules file. Source Kicad 5.1.5 files found here. https://github.com/rusefi/hw_48_na
[hw_48_na.zip](https://github.com/freerouting/freerouting/files/4659926/hw_48_na.zip)
* [ ] Did you use any command line arguments?
No I did not. 
* [ ] Can you post a screenshot of the problem?
[image]
* [ ] Can you include the latest, related log entries from the .log file?
[freerouter.log](https://github.com/freerouting/freerouting/files/4659937/freerouter.log)
* [ ] Please list the detailed steps to reproduce the problem
Open DSN, click OK on warning about rules, click autoroute button. 
* [ ] What would you expect freerouting to do in this situation?
I would hope it routes the traces like it normally does on a new file. 
* [ ] Is fixing this issue critical for your work?
I can manually route, so not exactly critical, but really helpful.

andrasfuchs, 2020-05-21:
I think this happens because you have space characters in the name of your clearance rules. 
Could you replace the spaces with underscores (_), and try again?

I confirm that this is a bug, and we should not allow entering spaces in the first place (or at least not to save them into the .rules files), but for now I added a new, more detailed error message to the log file, so that we can at least fix the problem manually.

jharvey, 2020-05-21:
This is and has been my board setup in KiCad for quite a while. 
[image]

It appears the .rules file is created by Freerouting. I was not sure how that file is generated, so I deleted it and then when I opened the dsn with freerouting it was created. It now routes as expected. I had previously removed the spaces, but I did not know how to update the .rules file. I was getting a dialog about using rules, but it was not clear to me what it was saying. Perhaps that dialog should ask if I should use the most recent rules or the old rules. 

On a side note, it would be cool if it would remember the last folder, so I don't have to keep drilling up and down  in the folders.

andrasfuchs, 2020-05-21:
I think both of your ideas are good, so I'll keep this issue open so that anyone who feels like it could provide a fix as a pull request.

For the folder thing there is a workaround I think: if you start freerouting with a "-di [your folder]" argument, it will use that as default.

closed 2022-08-01 by andrasfuchs