freerouting/freerouting #30: Cannot Open FreeRouter 1.1.4 on MacOS

pierremolinaro, 2020-06-04:
Feerouting Version: 1.4.4
Platform: OSX Mojave, OSX El Capitan
Did you use any command line arguments? -> No
Can you post a screenshot of the problem? -> Yes
[image]

Please list the detailed steps to reproduce the problem
  Just trying to open FreeRouter

Running the following command from the terminal solves the issue:
`sudo xattr -cr <<path>>/Freerouting.app`

Best Regards,

Pierre Molinaro

andrasfuchs, 2020-06-04:
Thanks @pierremolinaro !

@miho , could you take a look? Something fishy is going on with the Mac build, I had [some errors](https://github.com/freerouting/freerouting/runs/695852630) as well.

miho, 2020-06-04:
@pierremolinaro, great that you mentioned the `xattr` command. That's a good starting point.

Here's my guess what's going on:

The `xattr -cr` command recursively removes file attributes. If the app bundle runs after issuing the command then this indicates we are not facing a technical issue with our executable. It seems that Apple rejects unsigned apps if the file attributes indicate that the app has been downloaded from the internet. They seem to classify apps without proper signing as being "damaged".

Since file attributes are added after downloading the app bundle I doubt we can fix this unless we have proper signing in place. If we have many macOS users we should definitely fix this.

@andrasfuchs the build error you got and the error described here seem to be unrelated.

pierremolinaro, 2020-06-04:
Thank you for message.

I agree, it is related to application's signature.

I don't know if it can help you, I have developed an OSX application (https://github.com/pierremolinaro/ElCanari-distribution <https://github.com/pierremolinaro/ElCanari-distribution>, https://github.com/pierremolinaro/ElCanari-dev <https://github.com/pierremolinaro/ElCanari-dev>), and I have had  many many issues before providing an application that can be download and installed. I am not sure it is fully fixed.

I am currently using productbuild et hdutil for building the dmg distribution:

+ /usr/bin/productbuild --component-compression auto --component build/Release/ElCanari.app /Applications ElCanari-1.1.11.pkg
productbuild: Adding component at /Volumes/dev-svn/DISTRIBUTION_EL_CANARI_1.1.11/ElCanari-dev-master/build/Release/ElCanari.app
productbuild: Wrote product to ElCanari-1.1.11.pkg
productbuild: Supported OS versions: [10.10, )
+ /bin/cp ElCanari-1.1.11.pkg /Volumes/dev-svn/ElCanari-dev/../DISTRIBUTION_EL_CANARI_1.1.11
+ /bin/mkdir ElCanari-1.1.11
+ /bin/cp ElCanari-1.1.11.pkg ElCanari-1.1.11
+ /usr/bin/hdiutil create -srcfolder ElCanari-1.1.11 ElCanari-1.1.11.dmg -fs HFS+
....................................................................................................................................
created: /Volumes/dev-svn/DISTRIBUTION_EL_CANARI_1.1.11/ElCanari-dev-master/ElCanari-1.1.11.dmg

I am very interested by your work on FreeRouter, as ElCanari embeds a FreeRouter application, but without any fix from Alfons Wirtz's last release. My FreeRouter release has a size of 200 Mo, yours is 40 Mo.

Best regards,

Pierre Molinaro