freerouting/freerouting #32: Exclude pins from autoroute.

adamkrol93, 2020-06-12:
First of all, thank you for taking the time to report an issue.

To make our work easier, please provide as many details as you can:
* [ ] What version of freerouting are you using? **1.4.5-SNAPSHOT**
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit? **MacOS**
* [ ] What type of file are you working with? Can you attach the actual (DSN) file? **Kicad files**
* [ ] Did you use any command line arguments? **No**
* [ ] Can you post a screenshot of the problem? **No**
* [ ] Can you include the latest, related log entries from the .log file? **Not important**
* [ ] Please list the detailed steps to reproduce the problem
* [ ] What would you expect freerouting to do in this situation?
* [ ] Is fixing this issue critical for your work?

I would like to have feature to exclude some pins from autorouter.
I have a lot of pins on my board and I want few of them to be not routed automatically.

andrasfuchs, 2020-06-13:
Thank you @adamkrol93 for this feature request.

Just to clarify: by pins you mean unconnected pads, right?

In an ideal situation how would you prefer to define which pads you don't want to let the autorouter connect? In Kicad? In freerouting's graphical editor? Defined as a command line argument?

Could you try this as a workaround:
1, Open KiCad with your PCB
2, Right click on the pad that you want autorouter to ignore
3, Properties...
4, Change the copper to "None"
[image]
5, Export the file
6, Run the autorouter with freerouting
7, Import

adamkrol93, 2020-06-13:
By pins I mean pads, right.
I would like to select pad in freerouting graphical editor to exclude it from autoroute or route at all.
The problem is that I have input to my module and output. I would like to connect input pins directly to output pins, so that it isn't connected with any other pads.
[image]
By default autorouting (due to modules positioning) will connect SDA od INPUT to one module then to another and then to OUTPUT.
I hope this is clear.

adamkrol93, 2020-07-01:
You are very close.
> You also have an SDA and SCL on the OUTPUT, but those two should connect only to another IC, not to the INPUT.
SDA and SCL from OUTPUT should be directly connected to INPUT. Because I can have some capacitor between SDA(INPUT) and SDA(IC1) and I want to skip this while connecting to OUTPUT.

closed 2022-06-06 by andrasfuchs