freerouting/freerouting #33: Can projects add i18n（internationalization） support?

CNLiuGuang, 2020-06-17:
Dear Developers：
     I am from a non-English speaking country, and I am using Freerouter on KiCad. I hope Freerouter can add i18N support, thank you!

andrasfuchs, 2020-06-17:
Thanks for the suggestion. I'm not familiar with i18n, but I can see that there is a good Java documentation about it here: https://docs.oracle.com/javase/tutorial/i18n/

If you feel like adding a new language to freerouting, you can do so by adding new files [to the resources folder](https://github.com/freerouting/freerouting/tree/master/src/main/resources/eu/mihosoft/freerouting). Only English and German are supported at the moment.

I would be glad to accept any pull-requests that extends our target audience, so let me know if you need more information to proceed!

andrasfuchs, 2022-01-23:
@CNLiuGuang Chinese translation just got merged, so it will be available in a few weeks with freerouting version 1.5.

closed 2022-01-23 by andrasfuchs