freerouting/freerouting #34: freerouting 1.4.4 does not display anything from KiCad 5.1.6 DSN file

timg11, 2020-07-02:
* [ ] What version of freerouting are you using?  1.4.4 - I have tested with freerouting-1.4.4-executable.jar and freerouting-1.4.4-windows-x64.msi   

* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit? Windows 10 x64 1909
```
c:\Data\DEV>C:\Data\DEV\OpenJDK11\bin\java.exe --version
openjdk 11 2018-09-25
OpenJDK Runtime Environment 18.9 (build 11+28)
OpenJDK 64-Bit Server VM 18.9 (build 11+28, mixed mode)

```
* [ ] What type of file are you working with? Can you attach the actual (DSN) file?  Specctra DSN export from KiCad 5.1.6  (tried to attach, but attachment is unsupported type) 
* [ ] Did you use any command line arguments?  
c:\Data\DEV\OpenJDK11\bin>java -jar c:\data\dev\freeRouting\freerouting-1.4.4-executable.jar

* [ ] Can you post a screenshot of the problem?   
Empty gray screen with blue rectangle. 
[image]

Unrouted design in KiCad PCBnew:
[image]

* [ ] Can you include the latest, related log entries from the .log file?
```
2020-07-02 07:59:10.397 [main] INFO  eu.mihosoft.freerouting.FreeRouting - Freerouting application is started.
2020-07-02 07:59:34.942 [AWT-EventQueue-0] INFO  eu.mihosoft.freerouting.FreeRouting - Opening 'Green14SegLED.dsn'...
2020-07-02 07:59:35.542 [AWT-EventQueue-0] INFO  eu.mihosoft.freerouting.FreeRouting - Opening 'Green14SegLED.rules'...
2020-07-02 07:59:35.544 [AWT-EventQueue-0] ERROR eu.mihosoft.freerouting.FreeRouting - File 'Green14SegLED.rules' was not found.
```
* [ ] Please list the detailed steps to reproduce the problem
1) Export Specctra DSN from KiCad
2) Open freerouting with commandline above
3) Open the DSN file in freeRouting dialog "Open Your Own Design"

* [ ] What would you expect freerouting to do in this situation?
Display the PCB design
* [ ] Is fixing this issue critical for your work?
Yes

Please see [this thread for further information: 
](https://forum.kicad.info/t/first-try-with-freerouting-nothing-but-a-blue-rectangle/23651/15)

andrasfuchs, 2020-07-02:
@timg11 thank you for the bug report!

It looks strange indeed.

Could you try to run the following:
`java -jar freerouting-executable.jar -de Green14SegLED.dsn -do Green14SegLED.ses -mp 100`

If it doesn't start the autorouting, can you zip your DNS file first and than attach it to your reply so that I can test it too?

Thanks!

timg11, 2020-07-02:
@andrasfuchs thanks for the response. Here is the result of your command line suggestion.  When run, the UI opens the same way with the empty window with blue rectangle. 

```
c:\Data\DEV\OpenJDK11\bin>java -jar c:\data\dev\freeRouting\freerouting-1.4.4-executable.jar -de "C:\Data\DEV\KiCad\Green14segLED\Green14SegLED\Green14SegLED.dsn" -do "C:\Data\DEV\KiCad\Green14segLED\Green14SegLED\Green14SegLED.ses" -mp 100
2020-07-02 13:49:26.352 [main] INFO  eu.mihosoft.freerouting.FreeRouting - Freerouting application is started.
2020-07-02 13:49:26.357 [main] INFO  eu.mihosoft.freerouting.FreeRouting - Opening 'C:\Data\DEV\KiCad\Green14segLED\Green14SegLED\Green14SegLED.dsn'...
2020-07-02 13:49:27.287 [main] INFO  eu.mihosoft.freerouting.FreeRouting - Opening 'Green14SegLED.rules'...
2020-07-02 13:49:27.288 [main] ERROR eu.mihosoft.freerouting.FreeRouting - File 'Green14SegLED.rules' was not found.
2020-07-02 13:49:27.305 [main] INFO  eu.mihosoft.freerouting.FreeRouting - Generating snapshot
2020-07-02 13:49:27.367 [Thread-0] INFO  eu.mihosoft.freerouting.FreeRouting - Saving 'C:\Data\DEV\KiCad\Green14segLED\Green14SegLED\Green14SegLED.ses'...
WARNING: sun.reflect.Reflection.getCallerClass is not supported. This will impact performance.
2020-07-02 13:49:27.369 [Thread-0] ERROR eu.mihosoft.freerouting.FreeRouting - Couldn't export board to file
java.lang.NullPointerException: null
        at eu.mihosoft.freerouting.designforms.specctra.SessionFile.write_placement(SessionFile.java:115) ~[freerouting-1.4.4-executable.jar:unspecified]
        at eu.mihosoft.freerouting.designforms.specctra.SessionFile.write_session_scope(SessionFile.java:102) ~[freerouting-1.4.4-executable.jar:unspecified]
        at eu.mihosoft.freerouting.designforms.specctra.SessionFile.write(SessionFile.java:71) ~[freerouting-1.4.4-executable.jar:unspecified]
        at eu.mihosoft.freerouting.interactive.BoardHandling.export_specctra_session_file(BoardHandling.java:1119) ~[freerouting-1.4.4-executable.jar:unspecified]
        at eu.mihosoft.freerouting.gui.MainApplication$1.ExportBoardToFile(MainApplication.java:144) [freerouting-1.4.4-executable.jar:unspecified]
        at eu.mihosoft.freerouting.gui.MainApplication$1.autorouterFinished(MainApplication.java:125) [freerouting-1.4.4-executable.jar:unspecified]
        at eu.mihosoft.freerouting.interactive.BatchAutorouterThread.thread_action(BatchAutorouterThread.java:146) [freerouting-1.4.4-executable.jar:unspecified]
        at eu.mihosoft.freerouting.interactive.InteractiveActionThread.run(InteractiveActionThread.java:80) [freerouting-1.4.4-executable.jar:unspecified]

```

timg11, 2020-07-03:
I have figured out the issue. The design is divided into two boards, and there are two separate board outlines on the edge.cuts layer. 
I modified the edge cuts layer to have a boundary line around the perimeter of both boards, and drew the board separate lines on the silkscreen layer.  With that change, freerouter works as expected. 

Apparently there is an issue with multiple board outlines that causes a silent failure.

andrasfuchs, 2020-07-06:
Excellent, thank you again for the report!

andrasfuchs, 2022-05-12:
I added a handler for this case, so now freerouting displays the following error message if the boundary definition in the DSN file is not as simple as expected.

`There are multiple shapes defined in the boundary section of the DSN file. This scenario is not currently supported. If you have more than one board outlines defined, try to merge them into one.`

closed 2022-06-06 by andrasfuchs

sneakin, 2025-03-21:
Fyi, just had this problem and had a less useful error message about "this.board is null" in both the latest release from Github and from the Kicad plugin.