freerouting/freerouting #35: Can't open file. Stuck at "Loading desing..." in Windows 10 64 bits.

spiwocoal, 2020-07-22:
Whenever I try to open a .dsn file, it gets stuck at `Loading design F:\[path to design]\Freerouting Issue.dsn`.
The logs say `eu.mihosoft.freerouting.FreeRouting - Component.read_place_scope: ) expected`.

* [ ] What version of freerouting are you using?
Freerouting version 1.4.4
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
I am running Windows 10 64 Bit.
```
$ D:\Java\jdk-11.0.8\bin\java.exe --version
java 11.0.8 2020-07-14 LTS
Java(TM) SE Runtime Environment 18.9 (build 11.0.8+10-LTS)
Java HotSpot(TM) 64-Bit Server VM 18.9 (build 11.0.8+10-LTS, mixed mode)
```
* [ ] What type of file are you working with? Can you attach the actual (DSN) file?
I am working with a KiCad generated file:
[FreeRouting Issue.zip](https://github.com/freerouting/freerouting/files/4957280/FreeRouting.Issue.zip)
* [ ] Did you use any command line arguments?
I tried using the GUI and passing the file to the executable using `-de`.
* [ ] Can you post a screenshot of the problem?
[image]

* [ ] Can you include the latest, related log entries from the .log file?
```
2020-07-21 20:45:23.244 [main] INFO  eu.mihosoft.freerouting.FreeRouting - Freerouting application is started.
2020-07-21 20:45:23.678 [main] TRACE eu.mihosoft.freerouting.FreeRouting - Method 'MainApplication.main()' was performed in 0,55 seconds.
2020-07-21 20:45:45.819 [AWT-EventQueue-0] INFO  eu.mihosoft.freerouting.FreeRouting - Opening 'Freerouting Issue.dsn'...
2020-07-21 20:45:46.111 [AWT-EventQueue-0] WARN  eu.mihosoft.freerouting.FreeRouting - Component.read_place_scope: ) expected
```
* [ ] Please list the detailed steps to reproduce the problem
Just try to open a .dsn file.
* [ ] What would you expect freerouting to do in this situation?
Show the tracer GUI
* [ ] Is fixing this issue critical for your work?
Yes as I can't use the software at all.

andrasfuchs, 2020-07-22:
@spiwocoal Thanks for the report.

Freerouting has a problem with the line 90 and 96 in your DSN file.
`(place SW41 242888 -97631.2 front 0 (PN '))`
`(place SW2 19050 -59531.2 front 0 (PN `))`

I spent an hour with the bug, but the DSN reader is such a [spaghetti code](https://en.wikipedia.org/wiki/Spaghetti_code) that I can't figure it out how to fix it properly.

Sorry for the inconvenience. 
If you want to make it work, change the `PN '` expressions and avoid the ' characters where you can.

If you (or anyone else) feels like investigating this issue further, you could start in the [SpacctraFileScanner.java@1448](https://github.com/freerouting/freerouting/blob/3793976de6d5976b1c2f521e8742cd3f026fc306/src/main/java/eu/mihosoft/freerouting/designforms/specctra/SpecctraFileScanner.java#L1448).

spiwocoal, 2020-07-22:
Thank you! I didn't even think quotes could be the problem but now that I think about it it makes sense...
I think I'll take this opportunity to see if I can contribute to the project in some way

qoq, 2020-08-01:
This problem is caused by unquoted non-ascii characters. 
For example, this works fine:
(place XX 111 222 front 90 (PN ASCII123))
This gets stuck at "Loading":
(place XX 111 222 front 90 (PN NONASCII))
And this also works fine:
(place XX 111 222 front 90 (PN "NONASCII"))

Spectra parser should learn how to parse unquoted UTF-8

undermark5, 2020-09-05:
> This problem is caused by unquoted non-ascii characters.
> For example, this works fine:
> (place XX 111 222 front 90 (PN ASCII123))
> This gets stuck at "Loading":
> (place XX 111 222 front 90 (PN [...]

While the spectra parser should probably learn how to parse unquoted UTF-8, it is still failing to properly parse unquoted ASCII apparently the issue here was with the `'` character which is plain old ASCII and I just ran into this issue today with both `'` and `` ` `` characters (keyboard layout) it also will fail with `|` as well. though I think it fails differently. I think part of the issue is that `'` and `` ` `` are both characters that are used to designate strings in some languages which results in the parser treating it as an unterminated string (or strings are assumed to terminate either with closing character or newline character) which results in the parser failing to find a matching `)`

Kicad exported a file containing these errors, so I'm not sure if it is Kicad's fault for outputting malformed files, or the Spectra parser's fault for failing to interpret the file correctly. I'm leaning towards it being Kicad's fault in my cause and not the Spectra parser.