freerouting/freerouting #36: Autorouter refuses to use 45 degree traces on pins

jacobtohahn, 2020-08-07:
First of all, thank you for taking the time to report an issue.

To make our work easier, please provide as many details as you can:
* [x] What version of freerouting are you using?
v1.4.4
* [x] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
Windows 10, 64-bit
* [x] What type of file are you working with? Can you attach the actual (DSN) file?
DSN, 
[SBC.zip](https://github.com/freerouting/freerouting/files/5038608/SBC.zip)
* [ ] Did you use any command line arguments?
No, I used the GUI
* [x] Can you post a screenshot of the problem?
[image]
The issue can be seen on all of the red traces and most blue traces. It occurs directly next to the pins.
* [ ] Can you include the latest, related log entries from the .log file?
* [x] Please list the detailed steps to reproduce the problem
Open a DSN file and run the autorouter
* [x] What would you expect freerouting to do in this situation?
Make a single, 45-degree trace instead of a 90-degree angle leading into a 45-degree trace.
* [x] Is fixing this issue critical for your work?
I can manually fix it afterwards, but it takes a very long time. Therefore, it's not critical, but important.

Hi,
I've just started using FreeRouting, and while it works well, if the autorouter tries to route a trace and hits a pin next to it, it will insert a 90-degree angle. As you can see in the photo, the 90-degree angle then leads to a 45-degree trace. This is particularly bad with traces coming from pins on two-row pin headers. It would be more efficient and better looking if it could come directly out of the pin at 45 degrees, and would improve signal integrity on high-speed boards.
Is there an option to allow this? If not, is it possible that such an option could be added to a future release?
Thanks.

pierremolinaro, 2020-08-07:
Hello,

I think the "restrict pin exit directions" checkbox is checked in the "Router Parameter" window (in the menu bar : Parameter -> Route).

Just uncheck it, I think it solves the issue.

Best regards,

Pierre Molinaro

jacobtohahn, 2020-08-07:
> Hello, I think the "restrict pin exit directions" checkbox is checked in the "Router Parameter" window (in the menu bar : Parameter -> Route). Just uncheck it, I think it solves the issue. Best regards, Pierre Molinaro
> […](#)

Pierre,
Thanks for the suggestion - it solved it! I had a feeling there was some option somewhere, but didn't realize that the "route" options also affected the autorouter, seeing as there was a separate menu for that.
Thanks!

closed 2020-08-07 by jacobtohahn