freerouting/freerouting #42: freerouting without graphical interface

simonbeylat, 2020-12-24:
First of all, thank you for taking the time to report an issue.

To make our work easier, please provide as many details as you can:
* [ ] What version of freerouting are you using?
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
* [ ] What type of file are you working with? Can you attach the actual (DSN) file?
* [ ] Did you use any command line arguments?
* [ ] Can you post a screenshot of the problem?
* [ ] Can you include the latest, related log entries from the .log file?
* [ ] Please list the detailed steps to reproduce the problem
* [ ] What would you expect freerouting to do in this situation?
* [ ] Is fixing this issue critical for your work?

simonbeylat, 2020-12-24:
I want to run freerouting on a dokcer container, I only need the autorouter and i run it with commande line. is it possible to run it without using graphical interface ? 
 I'll send it into a computer server without screen

devinpooja, 2021-08-09:
I too want to run auto-router in background without open the pop up of auto-routing.
Is it possible to run it without using graphical interface ?
Your efforts will be highly appreciated.

andrasfuchs, 2021-08-09:
Unfortunately we don't have the option to run it without the GUI at the moment. 
It would need great effort to separate the GUI and the business logic. I agree 100% that it would be a great step forward.

It would be a big contribution if someone dived into the source code and made the needed modification as a pull-request.

erichVK5, 2021-08-17:
I have removed the GUI swing and AWT code, locale logic and have simplified the logging to simply print to stdout.

My colleague has also simplified the build process with a Makefile, allowing gradle to be removed as well.

The end result is a functional command line implementation of Freerouting that can be run headless in automated workflows.

I have endeavored to minimise the required dependencies as well; the freerouting executable jar file now weighs in at around 800k.

We have done this to simplify automated, headless export and import of DSN files from the FOSS PCB editor pcb-rnd, but of course others should find it useful too.

I will follow bugfixes and improvements here and incorporate them as they arise in freerouting_cli

The CLI implementation is available on svn:

svn://svn.repo.hu/freerouting_cli/trunk 

see also

http://repo.hu/projects/freerouting_cli/

Cheers,

erichVK5

devinpooja, 2021-08-18:
Hello erichVK5 Sir,

I have tried the freerouting_cli but it is throwing me error shown below.
[image]

devinpooja, 2021-08-18:
Thank you so much for your efforts.
It exactly is doing what I wished for.

Your efforts are great.
[image]

erichVK5, 2022-04-01:
I have now merged

1) recent multithreading enhancements and
2) early postroute optimisation cessation with incremental improvement below a set threshold

into freerouting-cli

FJ-BoardArchitect, 2023-02-16:
Hi, I have done some testing using freerouting-cli, and it is very useful, thanks! Looking at erickVK5 comments from April 2022 it seems there have been some updates in 2022. However, the files on http://repo.hu/projects/freerouting_cli/ are still from 2021. Do anybody know if the newer version available somewhere?

erichVK5, 2023-02-21:
I have not merged newer updates beyond the multilthreaded post route optimisations (which can break in some situations it seems - see other github issues for this), but hope to do so in the near future.

Was there anything specifically you wanted merged?

closed 2023-10-13 by github-actions[bot]