freerouting/freerouting #49: Error Message When Installing on OSX Mojave 10.14.6

hhaudio, 2021-05-07:
Hi, I am trying to install Freerouting on OSX Mojave 10.14.6 for use with KiCad (5.1.2). I downloaded the installer from the following source: https://github.com/freerouting/freerouting/releases/tag/v1.4.4

When I open the downloaded dmg I get a folder with the freerouting program inside. When I try to open this program I get an error message saying that it's "damaged and can't be opened" (see attached image). Same thing happens when I try moving the application elsewhere on my computer.  

[image]

Anything you can think of that might be causing this? I am currently working on a board with several hundred low hz signals to route and this option would be much preferable to doing it by hand. Thanks!

andrasfuchs, 2021-05-07:
This might be a similar to the issue #30.

I'm not familiar with OSX, but @pierremolinaro's solution might work for you too:
`sudo xattr -cr <<path>>/Freerouting.app`

hhaudio, 2021-05-07:
@andrasfuchs This worked for me, thanks! Perhaps worth making a note of on the downloads page.

andrasfuchs, 2021-05-07:
I'm glad it worked!

Is your project publicly accessible? Judging by the "several hundred low hz signals" expression it must be a really interesting piece of hardware ;)

closed 2021-05-07 by andrasfuchs

hhaudio, 2021-05-07:
@andrasfuchs Not publicly accessible as of yet, but will be online once it's finished. It's for a piece of PCB artwork in this vein: https://www.hammerheadaudio.com/art/great-wurm https://www.hammerheadaudio.com/art/knave-death