freerouting/freerouting #52: Why old version (1.2.43 Extra) works better than the latest (1.4.4)?

AlexIII, 2021-07-19:
I've tried to run a few boards without messing with any parameters, just hitting "autorouter". 
`1.2.43 Extra` always seem to produce much more optimal (in terms of via count and length) and overall nicer looking boards.
It would be nice to have an option to use the old algorithm.

Old version:
[image]

New version:
[image]

andrasfuchs, 2021-07-23:
I don't think we published the `1.2.43 Extra` version, it's most likely a fork of this repo with possible improvements.  
Could you provide a link to its source code or if that's not available then the location where you downloaded it from?

AlexIII, 2021-07-23:
I'm sorry, I cannot recall where exactly I got this version. Here's the exe file, if it's of any help. It seems to be packed with launch4j.
[FreeRouting.zip](https://github.com/freerouting/freerouting/files/6868244/FreeRouting.zip)

andrasfuchs, 2021-07-23:
Alright, I found a reference to this "extra" version [on a Japanese site](https://kanpapa.com/today/2020/05/kicad-freerouter-openjdk.html), but I couldn't find its author or its source code, so unfortunately it's very likely that we are not able to integrate this routing algorithm into our version.

Thank you for the information though, it's good to know that someone had the time to improve on it, it's a shame we don't know who.

If you feel like it and you have some time to investigate this further, please do so, and let me know, if you find the original author(s) and I'll contact them directly.

AlexIII, 2021-07-23:
I investigated the situation a little. I didn't find direct mentions of the "Extra" version and where I get it, so I decided to de-compile it and compare with v1.2.43 that I've found [here](https://centos.pkgs.org/7/epel-x86_64/freerouting-1.2.43-2.20140513git265f040.el7.1.noarch.rpm.html) (sources available). I manually looked through all the files and have found no significant differences. Also it seems like the "Extra" version wasn't stripped of debug information and de-compilation produced particularly good results (names of all local variables are sane, etc.). Here are de-compilation results in case I've missed something.
[FreeRouting_1.2.43_Extra_source_from_Fernflower.zip](https://github.com/freerouting/freerouting/files/6869539/FreeRouting_1.2.43_Extra_source_from_Fernflower.zip)

After running v1.2.43 with the "reference" board above it didn't produce exactly same result, but in my opinion, it's still better then the current version. It has less vias (granted, for the price of total trace length) and it looks more aesthetically pleasing.

[image]

So I think it would be great if you could look what went missing started from version 1.2.43, or maybe some default parameters got changed. I realize that my complains may very well be unjustified considering it's just one example. But I traced around 10 boards over the last 2 years (I know it's not much!) of different complexity with 1.2.43 and the result always seemed better compared to any recent version I've tried at that time.

andrasfuchs, 2021-07-23:
Thank you for your efforts.

I don't have much free time nowadays, but I'll look into it as soon as I can.

Could we somehow formulate what do you mean by "better" boards? Via count and total track length sounds OK, but how do we define nicer looking? Does it really matter in the end?

It would be also great to have some sample boards that you are familiar with for testing. Could you share some with me so that I can run them?

I obviously don't know how experienced you are with Java, but let me know if you are open to the idea of looking at the source and improve it. I would very much welcome a pull request if you made one.

AlexIII, 2021-07-24:
> It would be also great to have some sample boards that you are familiar with for testing. Could you share some with me so that I can run them?

I started to pick the boards for testing and running them multiple times with 3 different versions of the program. I immediately confirmed something that I initially didn't want to believe: the algorithm is unstable (given the exact same input it may still produce different results). I tried to run the board in `1.2.43` and in 10 re-runs I was getting via count anywhere from 21 to 13. After trying ~20 times I was able to get 14 vias on `1.4.4`. 

**Now it seems to me, it's not a question of finding "lost improvements" in `1.2.43`, but rather, is there a way to automate running autorouter several times on the same board and selecting the best result (based on via count and possibly other criteria).**

Here's the dsn file for the board (do whatever you like with it)
[case3.zip](https://github.com/freerouting/freerouting/files/6871941/case3.zip)

> Could we somehow formulate what do you mean by "better" boards? ... how do we define nicer looking? Does it really matter in the end?

I was afraid of this question. =) 
[image]
Left to right is the same 3 images as above in this thread from top to bottom.
It's mostly subjective, but there's a few things that could be formalized.
1. Notice how blue traces on the first image tend to go around the IC's pads and not between them (when it's possible), I think this behavior should be preferred.
2.  Zigzag-like traces (short distance between two consecutive trace turns (in green circles)) should have higher penalty.

> I obviously don't know how experienced you are with Java, but let me know if you are open to the idea of looking at the source and improve it. I would very much welcome a pull request if you made one.

I can write in java (though I don't particularly enjoy it =) ). That is, the language is not a problem, but the program architecture is. I don't think I have enough enthusiasm to get to know it...

>  don't have much free time nowadays, but I'll look into it as soon as I can.

Sure, great! I hope I didn't sound needy or pushy. I fully understand that no one owes anything. You're doing terrific job maintaining the project, that, in my opinion, has no comparable open-source alternatives. Thank you for that. I was really glad to find that the project is not abandoned.

andrasfuchs, 2021-07-24:
First of all thank you for the kind words, and the work you already put into this issue.

I see what you mean by improving the routing by introducing penalties for certain patterns, and I agree that it would be beneficial. 
My hesitation to modify the routing code roots in my previous attempt to refactor the code. It's reasonably complicated and it's written in an old school way, which makes it difficult for me to make significant changes without breaking something else.

We already had some discussions about the (lack of) architecture of the sourcecode here (#37), and I think it would take weeks to upgrade the code to a level that (1) separates the GUI from business logic, (2) clearly decouples the routing algorithm and (3) adds a separate board evaluation function that would weight the different properties of the board (including the patterns you mentioned) to make a judgement if that particular (partially) routed board is a good design. 

I don't have enough free time to get more involved at the moment, but I still appreciate your feedback a lot, I think it will be very useful in the future when me or someone else starts actively working on this project again. 

I'll keep this issue open for visibility.

lynchaj, 2022-03-01:
Hi
I have noticed on my large and complex PCBs that there is an appreciable difference between FreeRouting 1.5.0 and 1.2.43 Extra.  I am not sure what the difference is exactly but they behave differently or at least appear to be different.  

Since 1.5.0 quits at 0.1% it ends quite prematurely so it is not useful for me with large boards.  1.2.43 Extra will keep going much longer and squeeze out vias and trace length.  However, when 1.2.43 Extra starts to bog down and not find additional optimization I will let it run for a day or two and save the PCB as a DSN.  

Then I'll reopen the PCB in 1.5.0 and let it run to completion a couple of times.  It is amazing how it finds additional optimizations and eliminates vias and trace length.  Save and reopen in 1.2.43 Extra and the optimization is given new life and it will now find even more optimizations.  I think this is because the differences between 1.2.43 Extra and 1.5.0 are complimentary is some fashion.

In my current board, if I would have stopped with 1.5.0 when it was done I would have like 230 vias but now with the switching back and forth approach I am down to 190 vias and the shortest overall trace length so far.  Significant improvements!

Thanks, Andrew Lynch

PS, I've also noticed that occasionally "selecting all" and unfixing tends to free up the PCB too.  As does "select all" and pull-tight once or twice.  This seems to tweak the trace paths enough to find additional optimizations.  Not sure why it works but it certainly makes a difference.

closed 2022-06-06 by andrasfuchs