freerouting/freerouting #53: Request Windows 32 bit version

lynchaj, 2021-08-04:
First of all, thank you for taking the time to report an issue.

To make our work easier, please provide as many details as you can:
* [ ] What version of freerouting are you using?  very old Java based I think 1.2.x something
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?  Windows10, 32 bit with Java
* [ ] What type of file are you working with? Can you attach the actual (DSN) file?  Multiple DSN files
* [ ] Did you use any command line arguments? No
* [ ] Can you post a screenshot of the problem? N/A
* [ ] Can you include the latest, related log entries from the .log file? N/A
* [ ] Please list the detailed steps to reproduce the problem N/A
* [ ] What would you expect freerouting to do in this situation? Please post a Windows 32 bit build along side the 64 bit build
* [ ] Is fixing this issue critical for your work? No, but would be nice to use updated version of FreeRouting

lynchaj, 2021-08-04:
Thank you for your consideration.  Much appreciated!

andrasfuchs, 2021-09-19:
I spent a few hours on our build scripts and the Windows x86 (32-bit) version are now available, you can download it from here:
https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT

Could you test it and confirm that it works as expected on a 32-bit system?

lynchaj, 2021-09-19:
Hi Andras,
Thank you so much for making a 32 bit Windows version of FreeRouting. Unfortunately I get an error message (see attached image) and it does not install.

Thanks, Andrew Lynch

 
 

https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT

lynchaj, 2021-09-19:
Hi Andras,
Well, the attachment got stripped off my email. It was a PNG but it said "This installation package is not supported by this processor type. Contact your product vendor."

Hopefully this is easily fixable. Thank you! Andrew Lynch

 
 

 
 

https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT

andrasfuchs, 2021-09-19:
Sure, no problem. 
Thanks for the feedback! I will probably need your help again if I have a newer version to test, because I don't have a 32-bit system. I'll let you know here when the new file is ready.

lynchaj, 2021-09-19:
Hi Andras
OK great! Looking forward to it

Thanks, Andrew Lynch

andrasfuchs, 2021-09-19:
I know it's not ideal, but you can try to use the .jar release. If you [download that JAR file](https://github.com/freerouting/freerouting/releases/download/SNAPSHOT/freerouting-SNAPSHOT-2021-09-19_14-45-42.jar), and install [the 32-bit Java 11 runtime](https://adoptopenjdk.net/releases.html?variant=openjdk11&jvmVariant=hotspot#x32_win) on your machine, you should be able to run it.

For example, the following should work:
`java.exe -jar freerouting-executable.jar`

Could you get back to me if it works this way?

lynchaj, 2021-09-19:
Hi Andras,
OK, that worked. I downloaded the 32 bit Windows version of the OpenJDK JRE (29MB) and installed it. Then downloaded the FreeRouting snapshot. It ran right away in the new OpenJDK JRE like expected. Also my previous version of Java 8 and FreeRouting 1.2.43 Extra were not effected so they still run.

The new FreeRouting starts right up and loads designs no problem. However most of the time when I press the "autoroute" button on the top of the screen I get a message in the status bar on the bottom saying "autoroute interrupted, x connections incomplete" or something to that effect. Occasionally it seems to work but there must be some kind of bug still present. Other buttons on the top of the screen work fine like "zoom all" and "incompletes" so not sure what's happening with "autoroute"

On my computer the UI is plainly been updated to more recent standards and looks really nice. The FreeRouting application is basically the same. Possibly the autorouting and optimization seems to be going somewhat quicker but my computer is old and slow, Still any improvement, perceived or otherwise, is greatly appreciated!

Looks good! Thanks for making this work, I'll send more reports as I go.

Thanks, Andrew Lynch

andrasfuchs, 2021-09-20:
Hi Andrew,

Yes, you are right about the looks, it hasn't been changed in ages. :)

The 32-bit MSI installer would be a lot trickier than I thought, and I already spent more time on it than I wanted, so if the workaround works then I will leave it like that. I added a new section about the problem to [our readme page here on GitHub](https://github.com/freerouting/freerouting#running-freerouting-on-32-bit-systems).

The interruption bug you mentioned is interesting. If you could reproduce it in a consistent way, that would be great. If you can, please open a new issue and attach the .DSN file that caused the trouble. Thank you!

I'll close this issue now.

closed 2021-09-20 by andrasfuchs