freerouting/freerouting #54: Autorouter aborts with "This board was already evaluated, so we stop autorouter to avoid the endless loop."

strayer, 2021-08-10:
I've been trying to run freerouting on the keyboard I'm working on, but it always fails at pass 5 (or somewhere around that). I can't say that there are no issues with the PCB, but Kicad DRC reports nothing except unconnected elements. Due to being a mirrored PCB there are a lot of traces to be made, but I don't think it should be impossible.

When I import the session file in Kicad, only these two pads are not connected:

[image]

Looks like it fails here for some reason? When I route those pads manually in Kicad the DRC reports no errors left, so I'm a bit confused why freerouting simply stops there.

* What version of freerouting are you using?
https://github.com/freerouting/freerouting/releases/tag/v1.4.4

* On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
macOS Big Sur 11.5.1 (20G80)

* What type of file are you working with? Can you attach the actual (DSN) file?

[tairakb.dsn.txt](https://github.com/freerouting/freerouting/files/6959553/tairakb.dsn.txt)

* Did you use any command line arguments?

`java -jar .../freerouting-executable.jar -mp 100 -de ~/tmp/tairakb.dsn -do ~/tmp/tairakb.ses`

* Can you post a screenshot of the problem?

[image]

* Can you include the latest, related log entries from the .log file?

```
2021-08-10 08:45:40.383 [main] INFO  eu.mihosoft.freerouting.FreeRouting - Freerouting application is started.
2021-08-10 08:45:40.392 [main] INFO  eu.mihosoft.freerouting.FreeRouting - Opening '/Users/strayer/tmp/tairakb.dsn'...
2021-08-10 08:45:44.180 [main] INFO  eu.mihosoft.freerouting.FreeRouting - Opening 'tairakb.rules'...
2021-08-10 08:45:44.184 [main] ERROR eu.mihosoft.freerouting.FreeRouting - File 'tairakb.rules' was not found.
2021-08-10 08:45:45.031 [main] INFO  eu.mihosoft.freerouting.FreeRouting - Generating snapshot
2021-08-10 08:46:17.860 [Thread-0] WARN  eu.mihosoft.freerouting.FreeRouting - This board was already evaluated, so we stop autorouter to avoid the endless loop.
2021-08-10 08:46:17.888 [Thread-0] INFO  eu.mihosoft.freerouting.FreeRouting - Saving '/Users/strayer/tmp/tairakb.ses'...
```

*  Please list the detailed steps to reproduce the problem

Run the above command line or open the dsn in UI and click on autoroute.

strayer, 2021-08-10:
I rotated the first row of switches at the bottom right, that allowed the autorouter to at least route the top switch that didn't work before. After manually routing the bottom switches pads that don't work, the autorouter now seems to be able to continue:

[image]

I will have to do some modifications to the PCB anyway, so maybe this will fix itself after moving some other stuff around, but I still wonder why the autorouter is simply giving up.

Updated dsn (please excuse the shoddy ground planes, I was digging in the dark): 
[tairakb.dsn.txt](https://github.com/freerouting/freerouting/files/6959767/tairakb.dsn.txt)

andrasfuchs, 2021-09-20:
Thanks for the report and dsn file, and sorry for the late reply. I got some time to test it just today with the latest snapshot version of Freerouting.

I used this command line:
`freerouting.jar -de .\tests\Issue54-tairakb.dsn -do .\tests\Issue54-tairakb.ses`

The log I have looks suspicious, something is definitely not ideal, but the auto-router and the routing optimizer completes on my machine in 5 minutes.

Here is the stripped log I have (just FYI):

> 2021-09-20 17:27:50.401 [main] INFO  Freerouting v1.4.5-SNAPSHOT (build-date: 2021-09-20)
> 2021-09-20 17:27:50.412 [main] INFO  Opening '.\tests\Issue54-tairakb.dsn'...
> 2021-09-20 17:27:52.014 [main] INFO  Opening 'Issue54-tairakb.rules'...
> 2021-09-20 17:27:52.016 [main] ERROR File 'Issue54-tairakb.rules' was not found.
> 2021-09-20 17:27:52.471 [Thread-0] INFO  Starting auto-routing...
> 2021-09-20 17:27:52.479 [main] TRACE Method 'MainApplication.main()' was performed in 2.17 seconds.
> 2021-09-20 17:28:00.226 [Thread-0] TRACE Method 'BatchAutorouter.autoroute_pass #1 on board 'd55be0908a041fb621e10078dd1d549a' making 485 changes' was performed in 7.53 seconds.
> 2021-09-20 17:28:02.879 [Thread-0] TRACE Method 'BatchAutorouter.autoroute_pass #2 on board 'b72b9f9488b76ef692e93196fca57467' making 180 changes' was performed in 2.58 seconds.
> 2021-09-20 17:28:04.306 [Thread-0] TRACE Method 'BatchAutorouter.autoroute_pass #3 on board '7c47369a083d5249543733d8efbc53ec' making 83 changes' was performed in 1.36 seconds.
> 2021-09-20 17:28:04.664 [Thread-0] TRACE Method 'BatchAutorouter.autoroute_pass #4 on board '8025c3b31e4357bc5ba7966c7369c0dc' making 9 changes' was performed in 0.29 seconds.
> 2021-09-20 17:28:04.760 [Thread-0] TRACE Method 'BatchAutorouter.autoroute_pass #5 on board 'aa78fe1dc878fd569d693da31d49634b' making 0 changes' was performed in 0.05 seconds.
> 2021-09-20 17:28:04.761 [Thread-0] TRACE Method 'BatchAutorouterThread.thread_action()-autorouting' was performed in 12.29 seconds.
> 2021-09-20 17:28:04.762 [Thread-0] INFO  Auto-routing was completed in 12.29 seconds.
> 2021-09-20 17:28:04.763 [Thread-0] INFO  Starting routing optimization...
> 2021-09-20 17:28:05.849 [Thread-19] ERROR Exception in thread pool worker thread: Thread[Thread-19,5,main]
> java.lang.NullPointerException: null
> 2021-09-20 17:28:18.843 [Thread-0] TRACE Method 'BatchOptRouteMT.opt_route_pass #1 with 0 items, 95 vias and 3,815,094.35 trace length running on 32 threads.' was performed in 14.07 seconds.
> 2021-09-20 17:28:18.844 [Thread-0] INFO  Saving '.\tests\Issue54-tairakb_op1.bin'...
> 2021-09-20 17:28:20.161 [Thread-71] ERROR Exception in thread pool worker thread: Thread[Thread-71,5,main]
> java.lang.NullPointerException: null
> 2021-09-20 17:28:32.052 [Thread-0] TRACE Method 'BatchOptRouteMT.opt_route_pass #2 with 276 items, 91 vias and 3,831,677.64 trace length running on 32 threads.' was performed in 12.91 seconds.
> 2021-09-20 17:28:32.053 [Thread-0] INFO  Saving '.\tests\Issue54-tairakb_op2.bin'...
> 2021-09-20 17:28:32.763 [Thread-104] ERROR Exception in thread pool worker thread: Thread[Thread-104,5,main]
> java.lang.NullPointerException: null
> 2021-09-20 17:28:43.322 [Thread-0] TRACE Method 'BatchOptRouteMT.opt_route_pass #3 with 259 items, 85 vias and 3,948,574.36 trace length running on 32 threads.' was performed in 11.00 seconds.
> 2021-09-20 17:28:43.323 [Thread-0] INFO  Saving '.\tests\Issue54-tairakb_op3.bin'...
> 2021-09-20 17:28:47.011 [Thread-162] ERROR Exception in thread pool worker thread: Thread[Thread-162,5,main]
> java.lang.NullPointerException: null
> ...
> 2021-09-20 17:31:36.139 [Thread-0] TRACE Method 'BatchOptRouteMT.opt_route_pass #18 with 246 items, 72 vias and 4,160,598.13 trace length running on 32 threads.' was performed in 11.56 seconds.
> 2021-09-20 17:31:36.139 [Thread-0] INFO  Saving '.\tests\Issue54-tairakb_op18.bin'...
> 2021-09-20 17:31:47.759 [Thread-0] TRACE Method 'BatchOptRouteMT.opt_route_pass #19 with 246 items, 72 vias and 4,160,167.55 trace length running on 32 threads.' was performed in 11.35 seconds.
> 2021-09-20 17:31:47.760 [Thread-0] INFO  Saving '.\tests\Issue54-tairakb_op19.bin'...
> 2021-09-20 17:31:54.828 [Thread-778] ERROR Exception in thread pool worker thread: Thread[Thread-778,5,main]
> java.lang.NullPointerException: null
> 2021-09-20 17:31:59.599 [Thread-0] TRACE Method 'BatchOptRouteMT.opt_route_pass #20 with 244 items, 72 vias and 4,160,165.48 trace length running on 32 threads.' was performed in 11.58 seconds.
> 2021-09-20 17:31:59.600 [Thread-0] INFO  Saving '.\tests\Issue54-tairakb_op20.bin'...
> 2021-09-20 17:32:04.865 [Thread-819] ERROR Exception in thread pool worker thread: Thread[Thread-819,5,main]
> java.lang.NullPointerException: null
> 2021-09-20 17:32:04.905 [Thread-820] ERROR Exception in thread pool worker thread: Thread[Thread-820,5,main]
> java.lang.NullPointerException: null
> 2021-09-20 17:32:06.813 [Thread-816] ERROR Exception in thread pool worker thread: Thread[Thread-816,5,main]
> java.lang.NullPointerException: null
> 2021-09-20 17:32:06.856 [Thread-825] ERROR Exception in thread pool worker thread: Thread[Thread-825,5,main]
> java.lang.NullPointerException: null
> 2021-09-20 17:32:11.691 [Thread-0] TRACE Method 'BatchOptRouteMT.opt_route_pass #21 with 244 items, 72 vias and 4,160,366.12 trace length running on 32 threads.' was performed in 11.83 seconds.
> 2021-09-20 17:32:11.691 [Thread-0] INFO  Saving '.\tests\Issue54-tairakb_op21.bin'...
[... cut ...]
> 2021-09-20 17:32:34.473 [Thread-0] TRACE Method 'BatchAutorouterThread.thread_action()-routeoptimization' was performed in 4 minute(s) 29.71 seconds.
> 2021-09-20 17:32:34.476 [Thread-0] INFO  Routing optimization was completed in 4 minute(s) 29.71 seconds and it improved the design by ~45.47%.
> 2021-09-20 17:32:34.515 [Thread-0] TRACE Method 'BatchAutorouterThread.thread_action()' was performed in 4 minute(s) 42.04 seconds.
> 2021-09-20 17:32:34.516 [Thread-0] INFO  Saving '.\tests\Issue54-tairakb.ses'...
> 

Did you manage to get a usable .ses file in the end?

strayer, 2021-09-20:
Thanks for looking into it! My PCB layout changed a bit since creating my original issue and currently I can't get it to work at all, it always fails. I'll try the latest snapshot if I have time, but for now I simply manually routed everything (with a shocking amount of Vias though...)

While trying the autorouter on the (hopefully) final design I had a feeling that it may have issues with all of the mirrored pad components, but wasn't able to pinpoint it. E.g. in my first screenshot in the original issue there is a mirrored diode next to the row4 pad. The autorouter always disconnects the diodes in that area, connects them again, disconnects them, ... until it fails. Really weird.

If something comes up with the latest snapshot I'll upload an updated DSN.