freerouting/freerouting #56: Add "Discussions" tab to freerouting/freerouting

lynchaj, 2021-08-14:
First of all, thank you for taking the time to report an issue.

To make our work easier, please provide as many details as you can:
* [ ] What version of freerouting are you using? 1.2.43 Extra
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit? Windows 10, 32 bit
* [ ] What type of file are you working with? Can you attach the actual (DSN) file? N/A
* [ ] Did you use any command line arguments? N/A
* [ ] Can you post a screenshot of the problem? N/A
* [ ] Can you include the latest, related log entries from the .log file? N/A
* [ ] Please list the detailed steps to reproduce the problem N/A
* [ ] What would you expect freerouting to do in this situation? Add "Discussions" tab to freerouting/freerouting
* [ ] Is fixing this issue critical for your work? No

Hi

Please add a "Discussions" tab to the github.com/freerouting/freerouting page.  I am a huge fan and regular user of FreeRouting since 2007 and very interested in its development.  Would like to discuss with other FreeRouting users in forum without bothering the devs with my questions and thoughts.  

Some other github projects have a discussions forum and I think it would be helpful here as well.  At least I could speak with other people like myself without bugging devs with silly, repetitive questions and bounce ideas around in the appropriate forum.  If there is a FreeRouting forum somewhere else please let me know because I am not aware of any.

Please consider, thanks!  Andrew Lynch

andrasfuchs, 2021-08-14:
Hi Andrew, thanks for the tip, I like it.

I don't have the option in the settings where it suppose to be, probably because I have limited access rights to the repo.

@abelom, could you take a look?
Here is a quick tutorial how to enable discussions: https://docs.github.com/en/discussions/quickstart

lynchaj, 2021-08-14:
Thank you!

Andrew Lynch

abelom, 2021-08-15:
We now have https://github.com/freerouting/freerouting/discussions/57

abelom, 2021-08-15:
@andrasfuchs I've elevated your account on freerouting as well

andrasfuchs, 2021-08-15:
That's great, thank you @abelom!

closed 2021-08-15 by andrasfuchs