freerouting/freerouting #65: Sign our executables

andrasfuchs, 2021-09-18:
Not signing our executables might cause problems on Mac, so it would be great to include signing in our builds scripts.

[Jsign](https://github.com/ebourg/jsign) might be a good candidate for this task.

andrasfuchs, 2021-09-20:
This would theoretically solve the DMG related issues (#63, #31, #30).

pierremolinaro, 2021-09-21:
Sorry, but the application in freerouting-SNAPSHOT-2021-09-20_18-02-16-macos-x64.dmg also fails to execute.

You can check it is invalid by running :
 /usr/bin/codesign --verify --deep --strict --verbose=2 /Volumes/Freerouting/Freerouting.app

The response is:
/Volumes/Freerouting/Freerouting.app: code has no resources but signature indicates they must be present

Submitting a valid application (as /System/Applications/Mail.app) returns:
/System/Applications/Mail.app: valid on disk
/System/Applications/Mail.app: satisfies its Designated Requirement

Pierre