freerouting/freerouting #66: Spam of thread error exception

JonathSpirit, 2021-09-21:
Freerouting v1.4.5-SNAPSHOT (build-date: 2021-09-21)
windows 10 21H1 64bit

Command line used :
java -jar freerouting-executable.jar -Xmx32g -mt 24 -us greedy -is prioritized -de Project_GP8B.dsn -do test.ses

This error message appears pretty often when optimizer start : 
`[Thread-1] ERROR Exception in thread pool worker thread: Thread[Thread-1,5,main]
java.lang.NullPointerException: null
        at eu.mihosoft.freerouting.autoroute.OptimizeRouteTask.run(OptimizeRouteTask.java:72) ~[freerouting-executable.jar:unspecified]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) ~[?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) ~[?:?]
        at java.lang.Thread.run(Thread.java:834) [?:?]`

[image]

Step : 
- Use the command line cited above and wait for the optimizer to start.

Expected : 
- Not spamming thread error exception and correctly catch them.

Critical : 
- Not really but it can become one

DSN file : 
[Project_GP8B.dsn.txt](https://github.com/freerouting/freerouting/files/7205134/Project_GP8B.dsn.txt)
(dsn file from my project https://github.com/JonathSpirit/GP8B/tree/V5.1)

--------------------------------------------------------------
First of all, thank you for taking the time to report an issue.

To make our work easier, please provide as many details as you can:
* [ ] What version of freerouting are you using?
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
* [ ] What type of file are you working with? Can you attach the actual (DSN) file?
* [ ] Did you use any command line arguments?
* [ ] Can you post a screenshot of the problem?
* [ ] Can you include the latest, related log entries from the .log file?
* [ ] Please list the detailed steps to reproduce the problem
* [ ] What would you expect freerouting to do in this situation?
* [ ] Is fixing this issue critical for your work?

andrasfuchs, 2021-09-21:
Thanks for the report, that's a cool project!

I had the same NullPointerException thing yesterday and hopefully I fixed it. I can't tell which exact build you had, could you make sure that you use the currently latest `freerouting-SNAPSHOT-2021-09-20_21-26-39.jar`?

I'm running it with your provided DSN file right now, it's doing the first route optimization pass and it looks okay at the moment.

andrasfuchs, 2021-09-22:
I let the route optimizer to run for pass 16, and it still looked good, so I close this issue.

You can always get the latest binaries from the SNAPSHOT section of [the releases page](https://github.com/freerouting/freerouting/releases).

closed 2021-09-22 by andrasfuchs

JonathSpirit, 2021-09-22:
Thanks,

I build freerouting using Netbeans yesterday (on master, with latest commit 39281fb993e671ce17df0ef6b6b201be40a687c9), I can send you more information on what I do when I'm home if you want.

JonathSpirit, 2021-09-22:
I test with freerouting-SNAPSHOT-2021-09-22_13-16-03.jar and there is no issue.

Here is the build output from NetBeans : 
[image]

And here is the about page (using jdk-11.0.10) : 

Product Version: Apache NetBeans IDE 12.0
Java: 11.0.10; Java HotSpot(TM) 64-Bit Server VM 11.0.10+8-LTS-162
Runtime: Java(TM) SE Runtime Environment 11.0.10+8-LTS-162
System: Windows 10 version 10.0 running on amd64; Cp1252; fr_CH (nb)

I don't know if I make a mistake or just forget something (I'm not experienced with java) but it's a little bit weird. Maybe the problem is from NetBeans ?