freerouting/freerouting #67: Round pads transformed in octogonal pins that cause 90° tracks

Tomaaso, 2021-09-27:
To make our work easier, please provide as many details as you can:
* [X] What version of freerouting are you using?
   https://github.com/freerouting/freerouting/releases/tag/v1.4.4
* [X] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
   Linux Ubuntu 20.04.2 LTS 64-bit
* [X] What type of file are you working with? Can you attach the actual (DSN) file?
  I work with Kicad 5.99 with this DSN file.
  [Carte mère.dsn.zip](https://github.com/freerouting/freerouting/files/7235150/Carte.mere.dsn.zip)

Hello,
I started to using freerouting to autoroute my boards, thats is a powerfull tool, but I encounter a problem. Somes footprints with round pad are transfomed in octogonal pads. 

[image]

That cause all tracks are connected to those octogonal pads with 90° corner. It's not very clean and can cause EMC problems.

I saw some pictures on the web with this similar issue, but nobody report it.

So I would like to know if someone has a solution for it.

Regards.

Thomas.

andrasfuchs, 2021-09-27:
Hi Thomas, 

Thanks for the report! I can't get into this issue in the next 3 weeks, but I'll look into it after I'm back.

Until then, could you try [our latest snapshot release](https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT) if that makes any difference?

Also, if you or anyone else have other DSN files that demonstrate this problem, please share it here. The more samples we have the more likely I can find the root of the problem.

Tomaaso, 2021-09-27:
Hi @andrasfuchs,

Thanks for your answer. I tried with the snapshot release, issue still here. If thats help you, I joined all my dsn files of my project (3 cards).

[DSN files.zip](https://github.com/freerouting/freerouting/files/7235328/DSN.files.zip)

lynchaj, 2021-09-27:
Hi,
Yes, I've noticed this phenomenon as well. In KiCAD the pad looks like a circular pad but in FreeRouting its an octagon. This impacts routing because sometimes it places tracks from pins between pins so other tracks cannot flow between. Certainly not a show stopper but not what I hoped for or intended. Where did come from? Is it inherent in the KiCAD footprints library? It is present in both 1.2.43 Extra and the most recent snapshot so its either part of KiCAD or been in FreeRouting a long time.

Thanks, Andrew Lynch

 
 

   
https://github.com/freerouting/freerouting/releases/tag/v1.4.4

janding, 2021-12-27:
I attached a smaller DSN with two resistors to illustrate this issue: [octagon_pads.dsn.txt](https://github.com/freerouting/freerouting/files/7781364/octagon_pads.dsn.txt)

Note that the right pads are octagons and the left pads are circles. The autorouter creates a trace that comes out of the right pads horizontally instead of diagonally like on the left pads.

[image]

I believe this is related to how some pads are defined in KiCad. Many footprints have (circular) ovals instead of circles as pads. These are exported as path shapes instead of circles. From the attached DSN:

    (image Resistor_THT:R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal

      [...]

      (pin Round[A]Pad_1600_um 1 0 0)
      (pin Oval[A]Pad_1600x1600_um 2 10160 0)
    )
    (padstack Round[A]Pad_1600_um
      (shape (circle F.Cu 1600))
      (shape (circle B.Cu 1600))
      (attach off)
    )
    (padstack Oval[A]Pad_1600x1600_um
      (shape (path F.Cu 1600  0 0  0 0))
      (shape (path B.Cu 1600  0 0  0 0))
      (attach off)
    )

This is generated from the footprint R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal from the Resistor_THT footprint library (KiCad 6.0.0).

(freerouting 1.4.5.1, Windows 10, 64-Bit)

lynchaj, 2021-12-27:
Hi
I see this phenomenon is most of my PCBs. I suspect is seriously degrades trace routing performance of DIP and PLCC packages because it unnecessarily blocks pathways between pins. Would love to see this fixed but I suspect it is a KiCAD problem not FreeRouting

Thanks, Andrew Lynch

lynchaj, 2021-12-27:
Hi
I can provide some samples without using attachments. 

Please go to the nhyodyne repo and pick nearly any board with DIP chips (all of them) and you'll see it wedged in between DIP pins. 

I don't know what is causing this issue but it is seriously degrading PCB trace route quality in my opinion.

https://github.com/lynchaj/nhyodyne

Thanks, Andrew Lynch

wc7086, 2023-05-28:
Octagonal pad is a circle defined as oval pad (X and Y have the same size). I think this problem is caused by freerouting defining oval pad to be octagonal, and changing freerouting's definition of oval should help solve this problem.

wc7086, 2023-05-28:
Although the 90 ° wiring has no effect on electrical properties <sup>[[1]](https://resources.altium.com/p/pcb-routing-angle-myths-45-degree-angle-versus-90-degree-angle)</sup><sup>[[2]](https://bbs.elecfans.com/jishu_1899594_1_1.html)</sup>, But I don't think the 90 degree route looks as elegant as the 45 degree route because the 45 degree route has a shorter distance.  Does KiCAD have a plug-in that can check whether there is a 90° or acute Angle wiring in the line?