freerouting/freerouting #68: ARM64 binary support

tinyboxxx, 2021-09-30:
Hello, first of all thanks for maintaining such a good software. The results presented by the software help me save a lot of effort.

I have been using this software for a month or two now, but in the last few days I switched from a desktop with a 3700x to a MacBook Air with an m1 chip (16G RAM).
From my experience I think the speed of this software has been greatly reduced.

I hope to be using the ARM64 version of freerouting in the near future!

I didn't find an issue mentioning the ARM version, so I'm raising it here.
---

* [ ] What version of freerouting are you using?
1.44, which is latest in GitHub release.
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
MacOS, 64-bit, ARM.
* [ ] Can you post a screenshot of the problem?
[image]

* [ ] What would you expect freerouting to do in this situation?
release a ARM64 version.

* [ ] Is fixing this issue critical for your work?
not critical but performance should be better.

lynchaj, 2021-09-30:
Hi
Try downloading the OpenJRE and running the FreeRouting JAR file. I am using Win32 (non-supported platform) and it worked for me. I think difference between using the JAR and a platform specific versions is just packaging -- its the same basic Java software FreeRouting has always been. Any way, using the JAR directly with the OpenJRE works fine for me and maybe you too

Good luck, Thanks, Andrew Lynch

tinyboxxx, 2021-09-30:
Hi, thank you for your quick reply.

I found the right version of JRE here:
https://www.azul.com/downloads/?version=java-17-lts&os=macos&architecture=arm-64-bit&package=jre

I didn't find the jar file in the release, but I did find it in the Mac version release.
[image]

The jar file currently works fine, I haven't benchmarked it to compare the two versions, but I think it does seem a bit faster. Thank you very much for your suggestions!
But I still wish there was an easy to use version of ARM64 in the release.

thanks again!

andrasfuchs, 2021-10-13:
We use this Java JDK at the moment: https://adoptium.net/releases.html 
According to their website, M1 could be supported only if we moved from JDK 11 to JDK 17.

My understanding is that freerouting still can be run on M1 processors as it is, so this issue has a low priority for now.

If you need more speed, you could try our snapshot JAR release, that has multi-threaded route optimization ;)
https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT

koendv, 2022-03-01:
FYI. I am running freerouting on a raspberry pi4, running raspberry os, an aarch64 debian.
```
$ uname -a
Linux raspberrypi 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux
```

Build notes:

Download adoptium Temurin 11, linux aarch64, JDK.

```
wget https://github.com/adoptium/temurin11-binaries/releases/download/jdk-11.0.14.1%2B1/OpenJDK11U-jdk_aarch64_linux_hotspot_11.0.14.1_1.tar.gz
cd /opt
sudo tar xvf ~/Downloads/OpenJDK11U-jdk_aarch64_linux_hotspot_11.0.14.1_1.tar.gz
cd /tmp
git clone https://github.com/freerouting/freerouting
cd freerouting/
export JAVA_HOME=/opt/jdk-11.0.14.1+1/
bash gradlew assemble
```

I've attached a tar of the jar files in build/libs. 
[freerouting.tar.gz](https://github.com/freerouting/freerouting/files/8164181/freerouting.tar.gz)

closed 2022-05-12 by andrasfuchs