freerouting/freerouting #70: Skipping Nets absent

savio-code, 2021-11-01:
I cant find anyway to skip a particular net when routing.. The absent feature is really frustrating

szundi, 2022-03-16:
I would second this. I am using the Inner 1 layer in a 4 layer stack as a ground layer. I connect several component's ground pins to this ground layer with vias. How can I tell Freerouting not to bother connecting these islands of ground pins together wasting precious PCB wires/area and adding complexity? These are already routed via the ground layer. 

I think this use case is pretty common with like all 4 layer boards out there basically.

It would be really really nice to be able to just skip a net or a net class from the autoroute.

andrasfuchs, 2022-05-12:
Hey guys, sorry for getting back to you so late. I think I understand the problem, I'll look into it in the next few weeks.

Would it be acceptable solution for you if you had to pass the net/net class names manually as a command line argument? 
It would be a quicker addition if you didn't need this option on the GUI.

As always pull-requests welcome, if you have the time to play around with the source code ;)

szundi, 2022-05-13:
@andrasfuchs of course, I'll prepare a DSN file that I create on EasyEDA just for this. 

I didn't even know there is command line interface :) If I start the program with these arguments, and it does not show up on the GUI, can I use the GUI? There are some settings I always click in before routing.

szundi, 2022-05-13:
[Autorouter_FQ101_PCB_2022-05-13.zip](https://github.com/freerouting/freerouting/files/8685167/Autorouter_FQ101_PCB_2022-05-13.zip)

Here it is. The layer I would like to skip is GND. There are vias next to the antenna holes that are connected by the GND copper area there (not visible in the DSN file)

andrasfuchs, 2022-05-13:
Great stuff @szundi, thank you! I can't promise I can get to it this week, but I'll drop you a line here as soon as I have something.

Just curious: you mentioned that you always click around on the GUI before routing. What are those clicks? Are there some settings that would be better saved globally and set them by default?

The settings set by the command line arguments should be reflected on the GUI, and you can interact with the GUI the same way whether you had command line arguments or not.

szundi, 2022-05-13:
@andrasfuchs of course I can tell you:

When there is the design that the autorouter cannot finish with default parameters, but only 1-2 wires are left, i start tinkering with the via costs, ripup start costs and the preferred direction cost setting after I carefully (just visually) analyzed the end result.  Then I make notes of the settings what worked well for THAT project, and next time I set up the same. I think this is just being lazy about doing some work by hand and I enjoy tinkering better than doing manual routing.

andrasfuchs, 2022-05-13:
I opened your file and check the net classes in the `Rules` / `Net Classes` menu.

[image]

Would it make sense to add a new `none` dropdown item for all net classes into the `on layer` column that would indicate that all the pins in that net class should be ignored by the router?

szundi, 2022-05-14:
Hmm... I think yes, it would make perfect sense to have an "ignore" option here.

andrasfuchs, 2022-05-24:
@szundi @savio-code Could you guys give it a go?

szundi, 2022-05-27:
@andrasfuchs this is awesome, will test it soon on an other project too.

andrasfuchs, 2022-06-01:
Sorry for pushing this again, but I'd like to make a new release early next week, and this feature will be in it unless something is broken. I would need a quick confirmation from you guys that it is at least something along the lines you imagined.

Thank you!

savio-code, 2022-06-05:
Awesome!! .. ill test this once the the fix is released

szundi, 2022-06-06:
Hello, sorry but having a small baby does not help, I'll do this testing once I have an hour to do it!

andrasfuchs, 2022-06-07:
No worries, and congratulations! :)

It's already in the latest release, so please report me back if something is wrong with it and I'll try to fix it for the next release (in ~6 months).

szundi, 2022-06-07:
Hello, made some time because I just ran into some prepared file from the past. It works! It was only one test obviously, much more complex than what I uploaded previously. Thanks! 🚀

andrasfuchs, 2022-06-07:
That's great to hear! :+1: