freerouting/freerouting #71: Error in importing specctra session in KiCAD PCBNEW (Msg: Board may be corrupted)

Varat7v2, 2021-11-01:
I have followed the following steps:
1. Create the schematic in Eschema.
2. Annotated the components
3. Associated the components with their corresponding footprints.
4. Generated a netlist.
5. Opened the netlist in PCBNEW and then exported as Specctra DSN format
6. Opened the DSN file in FREEROUTING
7. Ran AUTOROUTER (successfully got the routing traces as expected)
8. Exported auto-routed design as Specctra Session file

**PROBLEM**: The exported specctra file is of 0 bytes. I don't what is going wrong here. Then once, I import the .sess file back to PCBNEW, it is throwing the following error.

    Board may be corrupted, do not save it. Fix problem and try again
    Expanded Error: Expecting " '(' " in input/source "/home/projectfolder/myPCB.sess" line 1, offset 1

**NOTE**: I am running this entire project on Ubuntu 20.04. And I am using Freerouting version v1.4.5-SNAPSHOT(build-date:2021-10-25)

cmburg, 2021-11-02:
I had this message once after I canceled the optimisation by left klick and then killing the terminal I was running freerouting off of.

You aren't saying anything about layout. Maybe you need to specify a frame on the "cuts" layer?!

Varat7v2, 2021-11-02:
@cmburg Thank you for your reply. I even tried with the `edge-cut` layer. But I got the same error. However, I tried with lower version of `Freerouting` i.e., 1.4.4, and it worked. I think there is some problem on the upgraded version.

closed 2021-11-02 by Varat7v2

noblepepper, 2021-11-02:
I can confirm this behavior I am getting an empty .ses file using Kicad 5.99 or 5.0.2 and Freerouting 1.4.5 executable downloaded here or master branch compiled locally.

If I save the file in freerouting native format and reopen it and export the ses file then it works fine.

/Hacking/freerouting/build/libs$ java -jar freerouting-executable.jar 
2021-11-01 19:35:30.653 [main] INFO  Freerouting v1.4.6-SNAPSHOT (build-date: 2021-10-31)
2021-11-01 19:35:45.616 [AWT-EventQueue-0] INFO  Opening 'test2.dsn'...
2021-11-01 19:35:46.051 [AWT-EventQueue-0] INFO  Opening 'test2.rules'...
2021-11-01 19:35:53.595 [Thread-0] INFO  Starting auto-routing...
2021-11-01 19:35:53.724 [Thread-0] INFO  Auto-routing was completed in 0.13 seconds.
2021-11-01 19:35:53.725 [Thread-0] INFO  Starting routing optimization...
2021-11-01 19:35:53.800 [Thread-0] INFO  Saving '/home/pep/Hacking/KicadProjects/test2/test2_op1.bin'...
WARNING: sun.reflect.Reflection.getCallerClass is not supported. This will impact performance.
2021-11-01 19:35:54.060 [Thread-0] ERROR / by zero
java.lang.ArithmeticException: / by zero
	at eu.mihosoft.freerouting.interactive.BatchAutorouterThread.thread_action(BatchAutorouterThread.java:146) [freerouting-executable.jar:unspecified]
	at eu.mihosoft.freerouting.interactive.InteractiveActionThread.run(InteractiveActionThread.java:80) [freerouting-executable.jar:unspecified]
2021-11-01 19:36:07.959 [AWT-EventQueue-0] INFO  Saving 'test2.ses'...
2021-11-01 19:36:17.054 [AWT-EventQueue-0] INFO  Saving '/home/pep/Hacking/KicadProjects/test2/test2.bin'...

/Hacking/freerouting/build/libs$ cat ../../../KicadProjects/test2/test2.ses 

/Hacking/freerouting/build/libs$ java -jar freerouting-executable.jar 
2021-11-01 19:36:57.945 [main] INFO  Freerouting v1.4.6-SNAPSHOT (build-date: 2021-10-31)
2021-11-01 19:37:13.999 [AWT-EventQueue-0] INFO  Opening 'test2.bin'...
2021-11-01 19:37:18.633 [AWT-EventQueue-0] INFO  Saving 'test2.ses'...
2021-11-01 19:37:20.722 [AWT-EventQueue-0] INFO  Saving 'test2.rules'...

/Hacking/freerouting/build/libs$ cat ../../../KicadProjects/test2/test2.ses 

(session test2.bin
  (base_design test2.bin)
  (placement
    (resolution um 10)
    (component Resistor_THT:R_Axial_DIN0414_L11.9mm_D4.5mm_P5.08mm_Vertical
      (place R1 1397000 -952500 front 0)
    )
    (component Resistor_THT:R_Axial_DIN0414_L11.9mm_D4.5mm_P15.24mm_Horizontal
      (place R2 1460500 -889000 front 0)
    )
  )
  (was_is
  )
  (routes 
    (resolution um 10)
    (parser
      (host_cad "KiCad's Pcbnew")
      (host_version "(5.99.0-12984-g1dbe8442b1)")
    )
    (library_out 
      (padstack "Via[0-1]_800:400_um"
        (shape
          (circle F.Cu 8000 0 0)
        )
        (shape
          (circle B.Cu 8000 0 0)
        )
        (attach off)
      )
...

StormTrooper, 2021-11-24:
I am experiencing the same issue. Using KiCAD 6.0 RC1 and a empty .ses file is saved. 
Going back to 1.4.4 fixes the issue.

andrasfuchs, 2021-11-30:
Could you please confirm that it is working as expected with the current v1.4.5.1?

StormTrooper, 2021-11-30:
I've tested with 1.4.5.1 and KiCad Version: 6.0.0-rc1-unknown-84cfa052a1~144~ubuntu20.04.1 and can confirm the issue is fixed for me.

andrasfuchs, 2021-11-30:
Thank you @StormTrooper!