freerouting/freerouting #72: Freerouting produces empty spectra session file

noblepepper, 2021-11-02:
* What version of freerouting are you using? Problem exists in 1.4.4 1.4.5 and master if compiled with gradle, precompiled version 1.4.4 works, 1.4.5 does not.
* On what platform did you run it? Linux Debian 10
* Can you attach the actual (DSN) file? See below
* Please list the detailed steps to reproduce the problem : open dsn from kicad, run autorouter, export spectra session file
* What would you expect freerouting to do in this situation? Produce non empty session file
* Is fixing this issue critical for your work? No, workaround below

After running the autorouter exporting a spectra session produces a empty file. If I save the file in freerouting native format, close routing window, open the saved file and then export a spectra session it gives a good file.

DSN file:
``` cattest2.dsn 
(pcb test2.dsn
  (parser
    (string_quote ")
    (space_in_quoted_tokens on)
    (host_cad "KiCad's Pcbnew")
    (host_version "(5.99.0-12984-g1dbe8442b1)")
  )
  (resolution um 10)
  (unit um)
  (structure
    (layer F.Cu
      (type signal)
      (property
        (index 0)
      )
    )
    (layer B.Cu
      (type signal)
      (property
        (index 1)
      )
    )
    (boundary
      (path pcb 0  165100 -101600  127000 -101600  127000 -82550  165100 -82550
            165100 -101600)
    )
    (via "Via[0-1]_800:400_um")
    (rule
      (width 250)
      (clearance 200.1)
      (clearance 200.1 (type default_smd))
      (clearance 50 (type smd_smd))
    )
  )
  (placement
    (component Resistor_THT:R_Axial_DIN0414_L11.9mm_D4.5mm_P5.08mm_Vertical
      (place R1 139700.000000 -95250.000000 front 0.000000 (PN R))
    )
    (component Resistor_THT:R_Axial_DIN0414_L11.9mm_D4.5mm_P15.24mm_Horizontal
      (place R2 146050.000000 -88900.000000 front 0.000000 (PN R))
    )
  )
  (library
    (image Resistor_THT:R_Axial_DIN0414_L11.9mm_D4.5mm_P5.08mm_Vertical
      (outline (path signal 120  2370 0  3580 0))
      (outline (path signal 120  2370 0  2350.54 -303.069  2292.49 -601.161  2196.79 -889.383
            2065.03 -1163  1899.35 -1417.52  1702.49 -1648.77  1477.67 -1852.94
            1228.59 -2026.69  959.337 -2167.16  674.33 -2272.04  378.252 -2339.62
            75.962 -2368.78  -227.575 -2359.05  -527.375 -2310.58  -818.515 -2224.17
            -1096.22 -2101.24  -1355.92 -1943.81  -1593.35 -1754.46  -1804.63 -1536.3
            -1986.27 -1292.92  -2135.3 -1028.3  -2249.26 -746.806  -2326.3 -453.046
            -2365.13 -151.846  -2365.13 151.846  -2326.3 453.046  -2249.26 746.806
            -2135.3 1028.3  -1986.27 1292.92  -1804.63 1536.3  -1593.35 1754.46
            -1355.92 1943.81  -1096.22 2101.24  -818.515 2224.17  -527.375 2310.58
            -227.575 2359.05  75.962 2368.78  378.252 2339.62  674.33 2272.04
            959.337 2167.16  1228.59 2026.69  1477.67 1852.94  1702.49 1648.77
            1899.35 1417.52  2065.03 1163  2196.79 889.383  2292.49 601.161
            2350.54 303.069  2370 0))
      (outline (path signal 50  6530 2500  -2500 2500))
      (outline (path signal 50  6530 -2500  6530 2500))
      (outline (path signal 50  -2500 -2500  6530 -2500))
      (outline (path signal 50  -2500 2500  -2500 -2500))
      (outline (path signal 100  0 0  5080 0))
      (outline (path signal 100  2250 0  2230.75 -293.684  2173.33 -582.343  2078.73 -861.038
            1948.56 -1125  1785.05 -1369.71  1590.99 -1590.99  1369.71 -1785.05
            1125 -1948.56  861.038 -2078.73  582.343 -2173.33  293.684 -2230.75
            0 -2250  -293.684 -2230.75  -582.343 -2173.33  -861.038 -2078.73
            -1125 -1948.56  -1369.71 -1785.05  -1590.99 -1590.99  -1785.05 -1369.71
            -1948.56 -1125  -2078.73 -861.038  -2173.33 -582.343  -2230.75 -293.684
            -2250 0  -2230.75 293.684  -2173.33 582.343  -2078.73 861.038
            -1948.56 1125  -1785.05 1369.71  -1590.99 1590.99  -1369.71 1785.05
            -1125 1948.56  -861.038 2078.73  -582.343 2173.33  -293.684 2230.75
            0 2250  293.684 2230.75  582.343 2173.33  861.038 2078.73  1125 1948.56
            1369.71 1785.05  1590.99 1590.99  1785.05 1369.71  1948.56 1125
            2078.73 861.038  2173.33 582.343  2230.75 293.684  2250 0))
      (pin Round[A]Pad_2400_um 1 0 0)
      (pin Oval[A]Pad_2400x2400_um 2 5080 0)
    )
    (image Resistor_THT:R_Axial_DIN0414_L11.9mm_D4.5mm_P15.24mm_Horizontal
      (outline (path signal 120  1440 0  1550 0))
      (outline (path signal 120  1550 2370  1550 -2370))
      (outline (path signal 120  1550 -2370  13690 -2370))
      (outline (path signal 120  13800 0  13690 0))
      (outline (path signal 120  13690 2370  1550 2370))
      (outline (path signal 120  13690 -2370  13690 2370))
      (outline (path signal 50  -1450 -2500  16690 -2500))
      (outline (path signal 50  16690 -2500  16690 2500))
      (outline (path signal 50  -1450 2500  -1450 -2500))
      (outline (path signal 50  16690 2500  -1450 2500))
      (outline (path signal 100  13570 -2250  13570 2250))
      (outline (path signal 100  0 0  1670 0))
      (outline (path signal 100  13570 2250  1670 2250))
      (outline (path signal 100  1670 2250  1670 -2250))
      (outline (path signal 100  1670 -2250  13570 -2250))
      (outline (path signal 100  15240 0  13570 0))
      (pin Round[A]Pad_2400_um 1 0 0)
      (pin Oval[A]Pad_2400x2400_um 2 15240 0)
    )
    (padstack Round[A]Pad_2400_um
      (shape (circle F.Cu 2400))
      (shape (circle B.Cu 2400))
      (attach off)
    )
    (padstack Oval[A]Pad_2400x2400_um
      (shape (path F.Cu 2400  0 0  0 0))
      (shape (path B.Cu 2400  0 0  0 0))
      (attach off)
    )
    (padstack "Via[0-1]_800:400_um"
      (shape (circle F.Cu 800))
      (shape (circle B.Cu 800))
      (attach off)
    )
  )
  (network
    (net "Net-(R1-Pad1)"
      (pins R1-1 R2-1)
    )
    (net "Net-(R1-Pad2)"
      (pins R1-2 R2-2)
    )
    (class kicad_default
      (circuit
        (use_via Via[0-1]_800:400_um)
      )
      (rule
        (width 250)
        (clearance 200.1)
      )
    )
  )
  (wiring
  )
)
```

ahtih, 2021-11-06:
Same here, I can reproduce that with another simple PCB.

This seems to be **caused by commit a4ec13293** from 20-Sep-2021, which produces "ArithmeticException: / by zero" in BatchAutorouterThread.java:146. This is dividing by via_count_before and trace_length_before; if either of these is 0, an error occurs.

andrasfuchs, 2021-11-14:
Fixed by @VicSanRoPe, I just merged the pull request. Thank you!

andrasfuchs, 2021-11-25:
The fix is now included in our latest release, v1.4.5.1.

closed 2021-11-25 by andrasfuchs