freerouting/freerouting #74: Unable to view any files within my Desktop, Documents, or Downloads Directory on MacOS (Permissions Issue?)

Gibbaaay76, 2021-11-06:
First of all, thank you for taking the time to report an issue.

To make our work easier, please provide as many details as you can:
* [ ] What version of freerouting are you using?
Using v1.4.5
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
MacOS 64-bit
* [ ] What type of file are you working with? Can you attach the actual (DSN) file?
* [ ] Did you use any command line arguments?
No
* [ ] Can you post a screenshot of the problem?
Yes, see images below.
* [ ] Can you include the latest, related log entries from the .log file?
* [ ] Please list the detailed steps to reproduce the problem
-Double click freerouting-1.4.5.jar file
-Attempt to browse Desktop, Documents, or Downloads directories.
-Results in a blank screen.
* [ ] What would you expect freerouting to do in this situation?
Provide access to these directories or provide documentation on how to provide permissions to the jar file?
* [ ] Is fixing this issue critical for your work?
An alternative I found is to place the the dsn file in the home directory, which freerouting has access to.
<img width="548" alt="Screen Shot 2021-11-06 at 7 33 58 PM" src="https://user-
[image]
[image]
images.githubusercontent.com/77290645/140626979-cfb9e15d-cf40-41d9-9994-c867c3371ea8.png">
.

tinyboxxx, 2021-11-14:
same here, I have to move file to my user folder

rhinodavid, 2022-01-08:
It is indeed a permissions issue.

There's a fix, but it exposes disk access for all JARs you run, so up to you to decide if it's worth it.

MacOS ships with an app called "JavaLauncher". You need to add this to the "Full Disk Access" panel under System Preferences > Security & Privacy > "Privacy Tab".

1. Launch the Terminal App
2. At the prompt, run ` open /System/Library/CoreServices` 
3. In the Finder window that opens, scroll down until you find the **JavaLauncher** app
4. Drag that icon into the list of apps in System Preferences > Security & Privacy > "Privacy Tab" (you may have to click the unlock button in the bottom left and enter your password first
[image]
[image]
)