freerouting/freerouting #76: Saving Session File Failed after routing.

casiobearing, 2021-11-16:
First of all, thank you for taking the time to report an issue.

To make our work easier, please provide as many details as you can:
* [ ] What version of freerouting are you using? 1.4.5
* [ ]  On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit? Win 10 x64, JDK 11.0.13
* [ ] it was a fresh install of Java as freeroute didn't run with Java 8.
* [ ] What type of file are you working with? Can you attach the actual (DSN) file? KiCad 5.1.12-1, Attached
* [ ] Did you use any command line arguments? java -jar freerouting-1.4.5.jar
* [ ] Can you post a screenshot of the problem? attached cmd line output and GUI screen
* [ ] Can you include the latest, related log entries from the .log file? I assume this is the cmd line output
* [ ] Please list the detailed steps to reproduce the problem, select DSN file, click autoroute, and when complete click export SES
* [ ] What would you expect freerouting to do in this situation? Save the SES file. It does create a SES file but it is 0 bytes in size.
* [ ] Is fixing this issue critical for your work? yes, it negates the purpose of using the app!
[VtoF.zip](https://github.com/freerouting/freerouting/files/7546993/VtoF.zip)
[Output.zip](https://github.com/freerouting/freerouting/files/7547056/Output.zip)

andrasfuchs, 2021-11-16:
Thank you for your bug report!

This is supposedly fixed in our snapshot: https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT

Could you try it and give us a feedback here if it worked?

casiobearing, 2021-11-16:
Hi, Went for a walk and came back have another look. I saw the label "Avoid division by cero (fixes issue #72)" against src\main in the source tree window and seeing my log file contains divide zero exceptions as well I have confirmed my issue is the same problem.... i.e. bug #72 is still active (by following the workaround of saving in native format, closing the app, reopening, loading and then exporting SES which worked).

andrasfuchs, 2021-11-16:
Thank you @casiobearing, I appreciate it. I'll take a look as soon as I got some time.

andrasfuchs, 2021-11-25:
You are right, we needed to include this fix in our latest release. I now got the time to create the new v1.4.5.1 hotfix release which includes that update from the snapshots.

closed 2021-11-25 by andrasfuchs