freerouting/freerouting #82: Missing help and spectra session file automaticly saved on exit command line options

ReportX, 2021-12-31:
Hello, thank you for keeping alive this really useful piece of software.

I'm just switching from freerouting-1.3.1 (running on Ubuntu 20.04) and I miss the command line help (-h) and -s command line option. I mainly running freerouting from a plugin installed in a CAD application, the command line set for this plugin is: 

java -jar /path/freerouting-1.4.5.1.jar -de /path/freerouting.dsn -s

as the "-s" command option is present, I've expected the spectra session file was automatic saved on exit but it looks like this command line option isn't implemented yet.

```
$ java -jar freerouting-1.3.1.jar -h
FreeRouting version 1.3.1
command line options are:
-de  provide design file
-di  design folder used in file dialog
-l   provide locale
-s   spectra session file is automatic saved on exit
-t   debug option
-white   white background
-h   this help

$ java -jar freerouting-1.4.5.1.jar -h
{date} {time} [main] INFO  Freerouting v1.4.5.1 (build-date: 2021-11-25)
```

andrasfuchs, 2021-12-31:
Hello @ReportX, thank you for your report!

Do you know which CAD plugin is used?

The automatic spectra file saving should work with the `-do` parameter like this:
`java -jar /path/freerouting-1.4.5.1.jar -de /path/freerouting.dsn -do /path/freerouting.ses`

Could you confirm that this works for you?

Do you have any contact details to the authors of the CAD plugin so that I can contact them to update the syntax on their end?

I agree with your remark about the `-h`, we should list the command line argument options just like we do in our [README](https://github.com/freerouting/freerouting#using-the-command-line-arguments) here.

ReportX, 2021-12-31:
Hello @andrasfuchs, thank you for your quick reply.

> Do you know which CAD plugin is used?

The plugin is kicad_freerouting-plugin and I think I downloaded it from https://github.com/random-builder/kicad_freerouting-plugin.git  

Hope it helps to contact the author.

> The automatic spectra file saving should work with the `-do` parameter like this: `java -jar /path/freerouting-1.4.5.1.jar -de /path/freerouting.dsn -do /path/freerouting.ses`
> Could you confirm that this works for you?

I have just modified the plugin in order to run the command with the -do parameter and yes, it works but I've noticed some changes in the previous behaviour. First, freerouting started to autoroute the board as soon I clicked in the plugin button; second, when the route is complete or aborted, freerouting is automatically terminated.

Without -do I had to start manually the autorouter and close the freerouting window to terminate it. I'd rather to do it in this way but I'd get used to the changes.

I'd like to add that I guess this plugin was probably downloaded massively so I wonder if every current user would be able to modify the python code in order to use the -do parameter.

> I agree with your remark about the `-h`, we should list the command line argument options just like we do in our [README](https://github.com/freerouting/freerouting#using-the-command-line-arguments) here.

Thank you. JFYI, I've just downloaded the jar file using a link so I just assumed the -h option would be available and the computer running it is an air-gapped computer, no Internet there.

Happy New Year!

andrasfuchs, 2022-01-04:
I have just created [a new issue](https://github.com/random-builder/kicad_freerouting-plugin/issues/11) in the repo of the plugin to include the `-do` parameter.

If you already have your modified script, would you consider creating a pull request there with your modifications? It would be great to help others as you said. ;)

ReportX, 2022-01-10:
I wish but I know a little about github, I'm just currently using it as if it was an Internet board. Moreover, my production server is in an air-gap network so I think it would be not easy for me to create a proper pull request but anyway I may share the modifications in an attached file here.

[freerouting_plugin_mod.zip](https://github.com/freerouting/freerouting/files/7837319/freerouting_plugin_mod.zip)

The attached .ZIP file includes two modified files which should be copied to the plugin directory (where has been previously installed) then edit the .INI file to customize extensions (probably there isn't need to) and paths (the most important is the path to freerouting.jar in the last lines).

andrasfuchs, 2022-01-10:
Thank you! I will create a pull request then in the repo of the plugin, mentioning you as the contributor.

closed 2022-01-18 by andrasfuchs