freerouting/freerouting #83: Question about log4j version vulnerability

techn0de, 2022-01-17:
All,

Hey, I am a sys admin and we whitelist our apps and we have an engineer that has always used your application and would like me to create a rule for this so he can continue to use it in our environment but I am hesitant due to the version of log4j this is using. From my understanding, the current version apache is asking people to use is 2.17.1. Will you be upgrading to a newer version or does anyone have a way for us to update this version so my engineers can continue to use this? Thanks in advance!!!!

andrasfuchs, 2022-01-17:
Thank you @techn0de for letting me know about this, I didn't know this vulnerability. 
I planned to make a new release in a few weeks, so I'll look into this later this week.

If you need it sooner, you can always fork this repo, make the changes you need and compile a new executable.

techn0de, 2022-01-17:
Andras,

Thank you for your quick reply, we really appreciate it! I am not in any hurry and am happy to wait until you have time. Also, I am not a developer, just a sys admin who specializes in security and can write some scripts. I have never complied any code before lol. Thanks again and I wish you the best! Our engineers love this application! You all did a great job.

Regards,

Adam Comstock

andrasfuchs, 2022-01-18:
Hey Adam, thank you for the kind words! 

I updated the log4j package, so you can test it with the [snapshot release](https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT), if you like.

Alternatively if you preferred a more stable version, please wait until I test it myself and make an official release in a few weeks.

closed 2022-01-18 by andrasfuchs