freerouting/freerouting #93: FreeRouting does not handle CJK character paths: llegal character '涓'

taotieren, 2022-01-24:
First of all, thank you for taking the time to report an issue.

To make our work easier, please provide as many details as you can:
* [ x ] What version of freerouting are you using?
- master
* [ x ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
- Windows 10 21H2 x64
* [ x ] What type of file are you working with? Can you attach the actual (DSN) file?
- [interf_u.zip](https://github.com/freerouting/freerouting/files/7922399/interf_u.zip)

* [ ] Did you use any command line arguments?
GUI
* [ ] Can you post a screenshot of the problem?
[image]

* [ x] Can you include the latest, related log entries from the .log file?
```bash
2022-01-24 11:19:57.424 [main] INFO  Freerouting v1.5.0-SNAPSHOT(汉化版)(build-date: 2022-01-21)
2022-01-24 11:19:58.697 [main] TRACE Method 'MainApplication.main()' was performed in 1.53 seconds.
2022-01-24 11:20:12.786 [AWT-EventQueue-0] INFO  Opening 'interf_u.dsn'...
2022-01-24 11:21:00.224 [AWT-EventQueue-0] ERROR Illegal character '涓' was found at position 27
java.lang.Error: Illegal character '涓' was found at position 27
	at app.freerouting.designforms.specctra.SpecctraFileScanner.next_token(SpecctraFileScanner.java:1404) ~[freerouting-executable.jar:unspecified]
	at app.freerouting.designforms.specctra.ScopeKeyword.read_scope(ScopeKeyword.java:60) ~[freerouting-executable.jar:unspecified]
	at app.freerouting.designforms.specctra.DsnFile.read(DsnFile.java:64) ~[freerouting-executable.jar:unspecified]
	at app.freerouting.interactive.BoardHandling.import_design(BoardHandling.java:1030) ~[freerouting-executable.jar:unspecified]
	at app.freerouting.gui.BoardFrame.read(BoardFrame.java:145) ~[freerouting-executable.jar:unspecified]
	at app.freerouting.gui.MainApplication.create_board_frame(MainApplication.java:348) ~[freerouting-executable.jar:unspecified]
	at app.freerouting.gui.MainApplication.open_board_design_action(MainApplication.java:292) ~[freerouting-executable.jar:unspecified]
	at app.freerouting.gui.MainApplication.lambda$new$2(MainApplication.java:237) ~[freerouting-executable.jar:unspecified]
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972) ~[?:?]
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313) ~[?:?]
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405) ~[?:?]
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262) ~[?:?]
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279) ~[?:?]
	at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:297) ~[?:?]
	at java.awt.Component.processMouseEvent(Component.java:6626) ~[?:?]
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3389) ~[?:?]
	at java.awt.Component.processEvent(Component.java:6391) ~[?:?]
	at java.awt.Container.processEvent(Container.java:2266) ~[?:?]
	at java.awt.Component.dispatchEventImpl(Component.java:5001) ~[?:?]
	at java.awt.Container.dispatchEventImpl(Container.java:2324) ~[?:?]
	at java.awt.Component.dispatchEvent(Component.java:4833) ~[?:?]
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4948) ~[?:?]
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4575) ~[?:?]
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4516) ~[?:?]
	at java.awt.Container.dispatchEventImpl(Container.java:2310) ~[?:?]
	at java.awt.Window.dispatchEventImpl(Window.java:2780) ~[?:?]
	at java.awt.Component.dispatchEvent(Component.java:4833) ~[?:?]
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773) ~[?:?]
	at java.awt.EventQueue$4.run(EventQueue.java:722) ~[?:?]
	at java.awt.EventQueue$4.run(EventQueue.java:716) ~[?:?]
	at java.security.AccessController.doPrivileged(AccessController.java:399) ~[?:?]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) ~[?:?]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97) ~[?:?]
	at java.awt.EventQueue$5.run(EventQueue.java:746) ~[?:?]
	at java.awt.EventQueue$5.run(EventQueue.java:744) ~[?:?]
	at java.security.AccessController.doPrivileged(AccessController.java:399) ~[?:?]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) ~[?:?]
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:743) ~[?:?]
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) [?:?]
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) [?:?]
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) [?:?]
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) [?:?]
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) [?:?]
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:90) [?:?]
2022-01-24 11:24:52.900 [AWT-EventQueue-0] INFO  Opening 'interf_u.dsn'...

```
* [x ] Please list the detailed steps to reproduce the problem
1. Demonstrate the project interf_u using kicad v6.0.1. 
2. Pcbnew opens the `.kicad_pcb` file. 
3. Export the DSN file to the `中文测试目录`. 
4. The text editor opens the DSN file in the `中文测试目录`. 
On the first line, you can see the *\`中文测试目录`\ interf_u.dsn. 
5. An error was reported when freerouting opened the DSN file.
* [ ] What would you expect freerouting to do in this situation?
* [ ] Is fixing this issue critical for your work?

interf_u.dsn

```bash
(pcb C:\Users\YI\Downloads\中文测试目录\interf_u.dsn
  (parser
    (string_quote ")
    (space_in_quoted_tokens on)
    (host_cad "KiCad's Pcbnew")
    (host_version "(6.0.1)")
  )
...
```

andrasfuchs, 2022-01-25:
I might be wrong, but as far as I can tell only the first line of the .dsn file contains a few Chinese characters, so if you change the `C:\Users\YI\Downloads\中文测试目录\interf_u.dsn` value to just `interf_u.dsn` it should work.

It is still a bug though, so I leave this issue open.

erichVK5, 2022-03-26:
I have made a pull request for modifications to SpecctraFileScanner.java that substitute 'X' for illegal characters found while parsing in an attempt to allow DSN parsing to continue successfully, and logs a warning while doing so.