freerouting/freerouting #94: Latest FreeRouting 1.5.0 (installed via .msi) doesn't launch on Windows 11

chf-6309, 2022-01-29:
First of all, thank you for taking the time to report an issue.

To make our work easier, please provide as many details as you can:
* [ X ] What version of freerouting are you using?                                                      1.5.0
* [ X ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?    Windows 11 (64bits, what else?)
* [ X ] What type of file are you working with? Can you attach the actual (DSN) file?   small 6809 kit project (route on 1.4.1...)
* [ X ] Did you use any command line arguments?                                                      no, launch through the Windows shortcut                                        
* [ X ] Can you post a screenshot of the problem?                                                      yes, see attached screenshot
* [ O ] Can you include the latest, related log entries from the .log file?                      no logs found :o(
* [ X ] Please list the detailed steps to reproduce the problem                                    install the 1.5.0.msi over the previous 1.4.1.msi installation, and lauch the router with the Windows shortcut...
* [ X ] What would you expect freerouting to do in this situation?                              freerouting should lauch and shox the "select your dsn file" panel...
* [ X ] Is fixing this issue critical for your work?                                                           no, the freerouting_1.5.0.jar file launch (after latest java runtime installation, separately with ninite.com)

I think that there is a bad link (hard coded directory?) that avoid the installed version to reach the "local" java runtime.

Nothing like this with the previous 1.4.1 installation (through the .msi, on the same HP 820G4 computer, running Windows 11 up to date).

Also, with the .jar file, I test on a previous (6809 scb card) dsn file, the freerouter leave more than 60 vias, compared to the less than 25, when running 1.4.1...

Anyway, thanks for your great piece of software.
-=chf=-

[image]
-

andrasfuchs, 2022-01-30:
I can confirm that this is an issue on Windows 11. Thank you @chf-6309 for reporting it!

Anyone who has the same problem, please try to run [freerouting-v1.5.0.jar](https://github.com/freerouting/freerouting/releases/download/v1.5.0/freerouting-1.5.0.jar) using an installed Java runtime for the time being. 
You can find more information about the usage [here](https://github.com/freerouting/freerouting#running-freerouting-on-32-bit-systems).

chf-6309, 2022-01-30:
Thank you Andras for your reply and confirmation!

Some more test: I finally uninstall the 1.5.0 and reinstall the 1.4.5.1 .msi version => no issue when launching.
It seems that the latest Windows 11 upgrade/updates are not the issue.

Anyway the .jar file is just fine!

chf-6309, 2022-01-31:
One more information: I tried to install the 1.5.0.msi version on Windows 10 on my work desktop: same error "JVM launch faill" :o(

Both installation, on french language Win11 and Win10: is the language the issue?

Reinstalling the previous .msi is OK and running.

andrasfuchs, 2022-02-01:
Thanks for the new info! I looked into it real quick, but I couldn't find the fix just yet.

I suspect that something goes wrong with the installer binary creation during one of the GitHub Actions, but I need to test it locally to get closer to the source of the problem. Unfortunately I couldn't find the Java log files either that could help us pin this one down, "JVM launch faill" is too general and I got sidetracked when I googled it. 
I don't think it's a language-related issue, but who knows :)

I don't think that I will be able to work on this in the next few weeks, so if anyone feels like doing some investigation, please go ahead! A pull-request would be ideal, but every little feedback helps. Thank you @chf-6309 again for your time working on this.

deltaford, 2022-02-01:
Same issue on Windows 10 Pro Version 21H2 Swedish language

raymondbernard, 2022-02-04:
Yes I am having the same issue :(

timo1235, 2022-02-28:
In my case it seems to be a problem with the java version:

```
Exception in thread "main" java.lang.UnsupportedClassVersionError: app/freerouting/gui/MainApplication has been compiled by a more recent version of the Java Runtime (class file version 55.0), this version of the Java Runtime only recognizes class file versions up to 52.0
```

bombela, 2022-03-18:
Install the latest openjdk instead of the java.com version. That's what solved it for me.

andrasfuchs, 2022-04-29:
There are new snapshot packages available for testing [on the releases page](https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT).

Could you guys give a short feedback if it works for you?

timo1235, 2022-04-29:
Thanks @andrasfuchs.

I have installed the latest version: v1.5.1-SNAPSHOT and can start it without problems on win11.

andrasfuchs, 2022-04-30:
Thank you @timo1235 for the feedback, that's great! 

It looks like we will have a working ,msi for the next release then.

chf-6309, 2022-04-30:
Hello Andras, 

I did uninstall the previous (working) 1.4.5.1 and install the latest v1.5.1 Snapshot (build 30apr22). 

My first test on an previous dsn file is successful! :o) 
The application launch from the Windows 11 64 bits 21H2 without any problem, the dsn file load as in the jar file: the previous issue sema to be definitely solved? Thanks for that. 

Notice that I did uninstall the "regular" Oracle Java runtime, and replace it with the "openjdk version "17.0.2" 2022-01-18 LTS" from Microsoft a few weeks ago, to use with the different freeRouting collection I keep (1.3 to 1.5.0). 
I will feed mode dsn file that I route the last weeks with your new 1.5.1 build. 

Again, thank you very much for your work. 

Kind regards. 
-=chf=- 

----- Mail original -----

andrasfuchs, 2022-05-02:
Thank you @chf-6309 for the detailed feedback and your continuous support, I appreciate it!

closed 2022-06-06 by andrasfuchs