freerouting/freerouting #96: App version OSX: corrupted

f1gwr, 2022-02-08:
Hi,
Just downloaded freerouting.dmg 1.5.0, opening it under OSX Mojave 10.14.6, application says "freerouting is corrupted, should be sent to the trash". The Java counterpart .jar doesn't open either.

_Originally posted by @f1gwr in https://github.com/freerouting/freerouting/discussions/57#discussioncomment-2132679_

f1gwr, 2022-02-08:
[image]

andrasfuchs, 2022-02-08:
Thanks for the report. I think it's related to #63 and #65.

You mentioned that the .jar doesn't work either. Could you share your error message when you tried to run it? 
That should still work, I think.

f1gwr, 2022-02-08:
Hi Andras,

Thanks for your answer. You might be right about the relation to message 63 and 65. But these were 4 and 5 months ago, and as it was a file corruption problem, I thought a few minutes recompiling and reposting could solve the problem meanwhile. Unless the problem is more serious?

As for the java version the message is enclosed.
[image]

Below is all I could find in console after jre-8u321-macosx-x64.dmg reinstall and then again freerouting launch with same error message:
par défaut	19:10:09.869083 +0100	MacJREInstaller	SecCodeCopySelf returned:0
par défaut	19:10:09.872875 +0100	MacJREInstaller	DR status:0
par défaut	19:10:10.703434 +0100	MacJREInstaller	Validate Status:0
par défaut	19:10:10.705602 +0100	MacJREInstaller	path: /Users/mp12c/Library/Application Support/JREInstaller
par défaut	19:10:10.731409 +0100	MacJREInstaller	---- INSTALL RUN |||| 02-08-2022 07:10:10 +0100
par défaut	19:10:10.731772 +0100	MacJREInstaller	Install Log: Type:Network Severity:Debug Message:Get Staged XML from :https://javadl-esd-secure.oracle.com/update/1.8.0/df5ad55fdd604472a86a45a217032c7d/mac-1.8.0_321-b07.xml
par défaut	19:10:10.876950 +0100	MacJREInstaller	Faulting in NSHTTPCookieStorage singleton
par défaut	19:10:10.876982 +0100	MacJREInstaller	Faulting in CFHTTPCookieStorage singleton
par défaut	19:10:10.877001 +0100	MacJREInstaller	Creating default cookie storage with process/bundle identifier
par défaut	19:10:10.878674 +0100	MacJREInstaller	Task <64AED7E3-4B12-4026-85BD-D6A924EB1F0F>.<0> {strength 1, tls 8, ct 0, sub 0, sig 0, ciphers 1, bundle 0, builtin 0}
par défaut	19:10:10.879425 +0100	MacJREInstaller	TIC Enabling TLS [1:0x600000204540]
par défaut	19:10:10.879647 +0100	MacJREInstaller	TIC TCP Conn Start [1:0x600000204540]
par défaut	19:10:10.879836 +0100	MacJREInstaller	[C1 Hostname#ea511be9:443 tcp, url hash: 17153c5a, tls] start
par défaut	19:10:10.911672 +0100	MacJREInstaller	nw_connection_report_state_with_handler_locked [C1] reporting state preparing
par défaut	19:10:10.912485 +0100	MacJREInstaller	Task <64AED7E3-4B12-4026-85BD-D6A924EB1F0F>.<0> setting up Connection 1
par défaut	19:10:10.967755 +0100	MacJREInstaller	nw_socket_handle_socket_event [C1.1:3] Socket received CONNECTED event
par défaut	19:10:10.967826 +0100	MacJREInstaller	nw_endpoint_flow_protocol_connected [C1.1 IPv4#babb7987:443 in_progress socket-flow (satisfied)] Transport protocol connected
par défaut	19:10:10.971066 +0100	MacJREInstaller	TIC TLS Event [1:0x600000204540]: 1, Pending(0)
par défaut	19:10:10.991082 +0100	MacJREInstaller	TIC TLS Event [1:0x600000204540]: 11, Pending(0)
par défaut	19:10:10.991155 +0100	MacJREInstaller	TIC TLS Event [1:0x600000204540]: 14, Pending(0)
par défaut	19:10:11.168767 +0100	MacJREInstaller	System Trust Evaluation yielded status(0)
par défaut	19:10:11.173613 +0100	MacJREInstaller	TIC TLS Trust Result [1:0x600000204540]: 0
par défaut	19:10:11.174729 +0100	MacJREInstaller	TIC TLS Event [1:0x600000204540]: 2, Pending(0)
par défaut	19:10:11.191458 +0100	MacJREInstaller	nw_endpoint_flow_protocol_connected [C1.1 IPv4#babb7987:443 in_progress socket-flow (satisfied)] Output protocol connected
par défaut	19:10:11.191581 +0100	MacJREInstaller	nw_connection_report_state_with_handler_locked [C1] reporting state ready
par défaut	19:10:11.191999 +0100	MacJREInstaller	TIC TLS Event [1:0x600000204540]: 20, Pending(0)
par défaut	19:10:11.192054 +0100	MacJREInstaller	TIC TCP Conn Connected [1:0x600000204540]: Err(16)
par défaut	19:10:11.192290 +0100	MacJREInstaller	TIC TCP Conn Event [1:0x600000204540]: 1
par défaut	19:10:11.192339 +0100	MacJREInstaller	TIC TCP Conn Event [1:0x600000204540]: 8
par défaut	19:10:11.192384 +0100	MacJREInstaller	TIC TLS Handshake Complete [1:0x600000204540]
par défaut	19:10:11.192613 +0100	MacJREInstaller	Task <64AED7E3-4B12-4026-85BD-D6A924EB1F0F>.<0> now using Connection 1
par défaut	19:10:11.194062 +0100	MacJREInstaller	nw_endpoint_flow_protocol_connected [C1.1 IPv4#babb7987:443 ready socket-flow (satisfied)] Output protocol connected
par défaut	19:10:11.194223 +0100	MacJREInstaller	Task <64AED7E3-4B12-4026-85BD-D6A924EB1F0F>.<0> sent request, body N
par défaut	19:10:11.210799 +0100	MacJREInstaller	Task <64AED7E3-4B12-4026-85BD-D6A924EB1F0F>.<0> received response, status 304 content U
par défaut	19:10:11.210922 +0100	MacJREInstaller	Task <64AED7E3-4B12-4026-85BD-D6A924EB1F0F>.<0> done using Connection 1
par défaut	19:10:11.255931 +0100	MacJREInstaller	Install Log: Type:Network Severity:Debug Message:Arguments:(
    "/var/folders/_b/2jn0mp5x4550xrkwmkqq23k80000gq/T/AppTranslocation/91B32098-F024-4CD2-8FFB-0B930B0373B6/d/Java 8 Update 321.app/Contents/MacOS/MacJREInstaller",
    "-psn_0_2511461",
    ""
)
par défaut	19:10:11.256171 +0100	MacJREInstaller	Install Log: Type:Other Severity:Debug Message:Dictionary:{
}
par défaut	19:10:11.256726 +0100	MacJREInstaller	Install Log: Type:Network Severity:Debug Message:Full JRE version: 1.8.321.07
par défaut	19:10:11.288273 +0100	lsd	Non-fatal error enumerating at <private>, continuing: Error Domain=NSCocoaErrorDomain Code=260 "Impossible d’ouvrir le fichier « PlugIns » car il n'existe pas." UserInfo={NSURL=PlugIns/ -- file:///private/var/folders/_b/2jn0mp5x4550xrkwmkqq23k80000gq/T/AppTranslocation/91B32098-F024-4CD2-8FFB-0B930B0373B6/d/Java%208%20Update%20321.app/Contents/, NSFilePath=/private/var/folders/_b/2jn0mp5x4550xrkwmkqq23k80000gq/T/AppTranslocation/91B32098-F024-4CD2-8FFB-0B930B0373B6/d/Java 8 Update 321.app/Contents/P [... cut ...]
par défaut	19:13:26.053727 +0100	com.apple.WebKit.WebContent	Current memory footprint: 64 MB
par défaut	19:13:29.829174 +0100	com.apple.WebKit.WebContent	Current memory footprint: 58 MB
par défaut	19:13:56.054677 +0100	com.apple.WebKit.WebContent	Current memory footprint: 64 MB
par défaut	19:13:59.829814 +0100	com.apple.WebKit.WebContent	Current memory footprint: 58 MB
par défaut	19:14:26.054673 +0100	com.apple.WebKit.WebContent	Current memory footprint: 64 MB
par défaut	19:14:29.830753 +0100	com.apple.WebKit.WebContent	Current memory footprint: 58 MB

[image]

andrasfuchs, 2022-04-29:
Unfortunately it's unlikely that it will solve our macOS-related issues, but there are new snapshot packages available for testing [on the releases page](https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT).

Could you guys give a short feedback if it works for you?

f1gwr, 2022-04-29:
Hi Andreas,

Thank you for returning. I appreciate.

Unfortunately, when trying to launch your prerelease version freerouting-SNAPSHOT-20220429_154300-macos-x64.dmg <https://github.com/freerouting/freerouting/releases/download/SNAPSHOT/freerouting-SNAPSHOT-20220429_154300-macos-x64.dmg>, the message remains the same:

Just noticed a detail: in the info window the Mac tries to compute the app file size ("Taille"), and the result is veeeery long to obtain (about 5 minutes, usually a second or so):

After finding its size I tried again, but with the same result (corrupted file)

Best,
Christian

andrasfuchs, 2022-04-30:
Hi Christian, thank you for the feedback! 

I'm sorry to hear that it still doesn't work. I can't see your screenshot here on GitHub, but I suppose that's the same as before. The issue with the size calculation is strange indeed.

I can't really go forward with this until we have someone who: 1, has a Mac and 2, is a member of the [Apple Developer Program](https://developer.apple.com/programs/), because I think that is needed to finally solve this issue for macOS users.
Please let me know if any of you guys have those and are willing to spend a few hours on this with me.

Until we do this I will disable the macOS package generation.

closed 2022-06-06 by andrasfuchs