freerouting/freerouting #100: Freeroute doesn't start on Linux 64bit

pcfreak1201, 2022-02-20:
Hi!
When I start the Linux binary, it prints out:
```
Exception in thread "main" java.lang.NoClassDefFoundError: javax/naming/Context
	at org.apache.logging.log4j.core.lookup.Interpolator.<init>(Interpolator.java:113)
	at org.apache.logging.log4j.core.config.AbstractConfiguration.<init>(AbstractConfiguration.java:133)
	at org.apache.logging.log4j.core.config.NullConfiguration.<init>(NullConfiguration.java:32)
	at org.apache.logging.log4j.core.LoggerContext.<clinit>(LoggerContext.java:85)
	at org.apache.logging.log4j.core.selector.ClassLoaderContextSelector.createContext(ClassLoaderContextSelector.java:254)
	at org.apache.logging.log4j.core.selector.ClassLoaderContextSelector.locateContext(ClassLoaderContextSelector.java:218)
	at org.apache.logging.log4j.core.selector.ClassLoaderContextSelector.getContext(ClassLoaderContextSelector.java:136)
	at org.apache.logging.log4j.core.selector.ClassLoaderContextSelector.getContext(ClassLoaderContextSelector.java:123)
	at org.apache.logging.log4j.core.selector.ClassLoaderContextSelector.getContext(ClassLoaderContextSelector.java:117)
	at org.apache.logging.log4j.core.impl.Log4jContextFactory.getContext(Log4jContextFactory.java:150)
	at org.apache.logging.log4j.core.impl.Log4jContextFactory.getContext(Log4jContextFactory.java:47)
	at org.apache.logging.log4j.LogManager.getContext(LogManager.java:196)
	at org.apache.logging.log4j.LogManager.getLogger(LogManager.java:599)
	at app.freerouting.logger.FRLogger.<clinit>(FRLogger.java:13)
	at app.freerouting.gui.MainApplication.main(MainApplication.java:33)
Caused by: java.lang.ClassNotFoundException: javax.naming.Context
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(Unknown Source)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(Unknown Source)
	at java.base/java.lang.ClassLoader.loadClass(Unknown Source)
```

* What version of freerouting are you using? - freerouting-1.5.0-linux-x64 (also tried the freerouting-1.5.0.jar)
* On what platform did you run it? - Linux 64-bit (Linux Mint 20 Ulyana base: Ubuntu 20.04 focal)
* What type of file are you working with? - none
* Did you use any command line arguments? - No
* Can you post a screenshot of the problem? - only the output above, no GUI comming up
* Can you include the latest, related log entries from the .log file? - "Freerouting Failed to launch JVM"
* Please list the detailed steps to reproduce the problem - just typing in the command
* What would you expect freerouting to do in this situation? - starting the GUI ;-)
* Is fixing this issue critical for your work? - well, yes ;-))

When I try to start the .jar-file, I get:
[image]

erichVK5, 2022-03-25:
I am successfully using the pre built .jar on lubuntu 20.04 with

$ java -version
openjdk version "11.0.11" 2021-04-20
OpenJDK Runtime Environment (build 11.0.11+9-Ubuntu-0ubuntu2.20.04)
OpenJDK 64-Bit Server VM (build 11.0.11+9-Ubuntu-0ubuntu2.20.04, mixed mode)

You could try building from source, which I have also managed successfully on the same system

git clone
cd freerouting
bash gradle assemble

this will pull down any dependencies needed, which, looking at the above, may be something to do with log4j on your system

Xeraster, 2022-04-27:
I'm having this same problem. The compiled release won't run and I can't compile. I've tried changing my java environment variable to just about every java version and trying with each one. I can neither compile or run the release. All I ever get when trying to compile is this error
```
* What went wrong:
Execution failed for task ':compileJava'.
> invalid source release: 11
```

I'm using Ubuntu 22.04 and all my packages are up to date.

andrasfuchs, 2022-04-28:
There must be a mismatch somewhere with the Java versions. I looked into this today for hours, and there is hope: finally I could make a working Windows installer, and I think the problem is similar for Linux as well. 

Don't get your hopes up just yet, I just wanted to drop by and let you guys know that it's not forgotten, I'll try to spare a few more hours to fix this in the near future.

andrasfuchs, 2022-04-29:
There are new snapshot packages available for testing [on the releases page](https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT).

Could you guys give a short feedback if it works for you?

TiagoBruno, 2022-05-01:
> There are new snapshot packages available for testing [on the releases page](https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT).
> 
> Could you guys give a short feedback if it works for you?

Snapshot 20220430_230000 fixed the issue on Fedora 35

andrasfuchs, 2022-05-02:
Great, thank you @TiagoBruno, good to know!

closed 2022-06-06 by andrasfuchs