freerouting/freerouting #103: Route optimization creates clearance violations

humannum14916, 2022-02-27:
I have a board made in KiCad that I am routing using Freerouting. While the actual routing works, the optimization creates clearance violations.

Freerouting version: v1.4.5.1 (build-date 2021-11-25)
Platform: Linux 64-bit
Files: [files.zip](https://github.com/freerouting/freerouting/files/8149230/files.zip)
  Unrouted board - Board-Unrouted.dsn
  Unoptimized board - Board-Routed.dsn
  After optimizer pass - Board-Opt.dsn
Command line arguments: none
Screenshot: [Screenshot from 2022-02-27 10-45-26](https://user-images.githubusercontent.com/87036457/155895500-6a7815df-2424-47a2-83b1-c8b4571a7c83.png)
Logs:
Recreation: Open the unoptimized board in Freerouting and run the route optimizer. After several passes, clearance violations will appear.
Expected behavior: Optimize the board without creating clearance violations

cmburg, 2022-04-11:
I do have clearance issues too but it only affects large holes (plastic pins for manual switches of a keyboard). I affects multiple versions (1.4.5, 1.4.5.1, 1.5) and occurs even when no optimisation is chosen.

`Error: Hole clearance violation (board setup constraints hole clearance 20 mils, actual *X* mils`  with *X* being 7.89/ 8.11/...

```
foo@bar:~/Projekte/Tastatur/Fritz$ java -jar freerouting-1.5.0.jar -de Fritz.dsn -do Fritz.ses -mp 20
2022-04-10 18:41:47.036 [main] INFO  Freerouting v1.5.0 (build-date: 2022-01-27)
2022-04-10 18:41:47.091 [main] INFO  Opening 'Fritz.dsn'...
2022-04-10 18:41:49.163 [main] INFO  Opening 'Fritz.rules'...
2022-04-10 18:41:49.165 [main] ERROR File 'Fritz.rules' was not found.
2022-04-10 18:41:50.007 [Thread-0] INFO  Starting auto-routing...
2022-04-10 18:43:59.928 [Thread-0] INFO  Auto-routing was completed in 2 minute(s) 9,90 seconds.
2022-04-10 18:43:59.930 [Thread-0] INFO  Starting route optimization on 3 threads...
2022-04-10 18:44:22.853 [Thread-3] WARN  CalcFromSide: start corner was not found
2022-04-10 18:44:22.887 [Thread-3] WARN  CalcFromSide: start corner was not found
2022-04-10 18:44:24.040 [Thread-3] WARN  CalcFromSide: start corner was not found
2022-04-10 18:44:24.209 [Thread-2] WARN  CalcFromSide: start corner was not found
2022-04-10 18:44:24.260 [Thread-2] WARN  CalcFromSide: start corner was not found
2022-04-10 18:44:25.296 [Thread-2] WARN  CalcFromSide: start corner was not found
2022-04-10 18:53:16.017 [Thread-0] INFO  Route optimization was completed in 9 minute(s) 16,08 seconds and it improved the design by ~51,35%.
2022-04-10 18:53:16.068 [Thread-0] INFO  Saving 'Fritz.ses'...
foo@bar:~/Projekte/Tastatur/Fritz$ 
```

This even appears when no optimisation or single threaded optimisation is chosen. So I think the problem lies somewhere in the router than the optimiser.

andrasfuchs, 2023-02-15:
It got fixed in version 1.7.0, so I close this now.

closed 2023-02-15 by andrasfuchs