freerouting/freerouting #106: my .dsn file is stuck on loading design

bradac1, 2022-03-10:
First of all, thank you for taking the time to report an issue.

To make our work easier, please provide as many details as you can:
* [ ] What version of freerouting are you using?
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
* [ ] What type of file are you working with? Can you attach the actual (DSN) file?
* [ ] Did you use any command line arguments?
* [ ] Can you post a screenshot of the problem?
* [ ] Can you include the latest, related log entries from the .log file?
* [ ] Please list the detailed steps to reproduce the problem
* [ ] What would you expect freerouting to do in this situation?
* [ ] Is fixing this issue critical for your work?

bradac1, 2022-03-10:
Im using th 64-bit version. I just downloaded it so its the latest version.
[image]

andrasfuchs, 2022-04-05:
Could you attach the DNS file you tried to load?

Do you have the log file by any chance?

closed 2022-05-02 by andrasfuchs