freerouting/freerouting #107: freerouting does not find obvious rutes?

githubaf, 2022-03-19:
First of all, thank you for taking the time to report an issue.

To make our work easier, please provide as many details as you can:
* [ ] What version of freerouting are you using?
* [ ] On what platform did you run it? Windows, Linux, MacOS? 32-bit or 64-bit?
* [ ] What type of file are you working with? Can you attach the actual (DSN) file?
* [ ] Did you use any command line arguments?
* [ ] Can you post a screenshot of the problem?
* [ ] Can you include the latest, related log entries from the .log file?
* [ ] Please list the detailed steps to reproduce the problem
* [ ] What would you expect freerouting to do in this situation?
* [ ] Is fixing this issue critical for your work?

erichVK5, 2022-03-28:
Is there a reason you are constraining the autorouting to one layer?

Some of the traces generated are quite tortuous with single layer routing attempts. Single layer routing is not always possible topologically.

The board routes quite well as a two layer board.

You may wish to try a simpler test case to see if there is an issue with KiCad exporting a partially autorouted layout, thereby ruling out an issue with the DSN being exported.

githubaf, 2022-03-28:
> Is there a reason you are constraining the autorouting to one layer?
Yes, the producer with the best price doesn't do vias and single side is cheaper than double sided.

> Single layer routing is not always possible topologically.
That is absolutely clear. My point was:
- the first dsn-file I showd can be roueted completely in freeroute.
- The second one only adds two connectors. If I take the result from the first routing and add the two connectors manually, I can very easily add the additional traces manually. (That is a have done actually) So it should be topographically possible to autoroute the complete board, shouldn't it?

One additional thing I observed:
During my work I found a routable layout (without the two connectors) Then I changed the placement of the components a tiny bit, such as shifting only one IC a centimeter to the right. There is still enough space between all pads the find the same traces again, but freerouting cannot solve the puzzle again...

andrasfuchs, 2022-04-05:
I haven't tested it yet, but I think you are right @githubaf ,this should not be a problem for freerouting and it's very likely a bug. Thank you for reporting and for the sample files.

andrasfuchs, 2022-05-12:
@githubaf , I have just tested your files with the newest snapshot build.

The first file (freq_teiler_200kHz_kicad.dsn) with its .rules was completed in a few minutes with one unroutable connection. 
The second file (freq_teiler_200kHz_kicad_bad.dsn) without the .rules completed fully as far as I can tell. If I applied the same rules to it, it also finished with one unroutable connection.

Could you also [download the snapshot build](https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT), and test it if it works as you would expect?

closed 2022-06-06 by andrasfuchs