freerouting/freerouting #109: msi installer is missing in v1.5

rezaneam, 2022-03-27:
If I am not mistaking MSI installer used to be available for v1.5 which is no longer available.

andrasfuchs, 2022-04-05:
The installers were removed because they were reported as invalid (see #94, #96, #100).

There must be something wrong with [our installer generation scripts](https://github.com/freerouting/freerouting/tree/master/distribution), but I don't have the capacity to fix them, so we are going to have only jar files in the releases until it's fixed. 

Pull requests are welcome!

andrasfuchs, 2022-04-29:
There are new snapshot packages available for testing [on the releases page](https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT).

Could you guys give a short feedback if it works for you?

andrew64k, 2022-05-01:
It's better? I guess... It installed and starts (loading board file) but then exits with "Child process exited with code 1" when I load the file on the command line (short cut).

When run Freerouting directly it starts and I select the file but it says it's unable to read design file.

Design file is from Kicad 5.12. The older 1.4.5.1 does not cause these errors.

It looks like you have solved the java loader problem but now it can't read my design file (no additional error).

I'm using Windows 7 64 bit and it's important to have Freerouting be a stand-alone installer with its own Java because I require older versions of Java installed to run management tools and it's great to have Freerouting be self contained.

Thanks for working on this!!!!

andrasfuchs, 2022-05-02:
Hi @andrew64k thank you for the feedback. Yes, there is some work in progress on the DSN reader module, and despite my efforts I don't expect it to work properly on all inputs with the current snapshot release. 

If you could send me here your problematic DSN file(s), that would be a great help. I need as much test data as possible in order to release a more robust version next time.

It's good that you mentioned the need for multiple Java versions on your system, and I'm glad that the latest installer works for you.

I'm not 100% because I run Windows 11 at the moment, but I think you can have multiple Java JREs installed in a system, and choose which one to be the default. So if you wanted to test this out, you could install more then version of Java JRE from [here](https://adoptium.net/temurin/releases/), and use version 17 only for freerouting while you keep using an older version of your other tools that need it.

andrew64k, 2022-05-12:
@andrasfuchs Great news! **freerouting-SNAPSHOT-20220511_234400-windows-x64.msi** works again for me!!!
It install on windows 7 (64 bit), runs, loads the DSN file (that loaded on the old version), and starts autorouting!

Thanks for spending the time to make it work again and improving freerouting.

Now for the bad news.... the autorouter now only got down to about 99 vias where the old version got down to less than 70.
Running the autorouter again seems to continue the progress. So it looks like it is finishing early now.

andrasfuchs, 2022-05-12:
@andrew64k Thank you for the kind words, I spent a few nights on this for sure :) I'm glad to hear that it works for you, we are getting closer to the next release.

About the via number: I'm still testing the different settings, but you can play around with the `-oit` parameter. `-oit 0.1` is the default, and that makes the route optimizer pretty quick, but the end result might not be optimized enough for your scenario. If you start decreasing that parameter, you let freerouting work more on the board. Please try `-oit 0.01`, `-oit 0.001`, `-oit 0.0001` etc, and see what you get. Using `-oit 0` will mean that it works on the PCB until it's interrupted. 
I made a few modifications to it last night, so make sure you get the latest snapshot version.

Let me know what you think would be the ideal default setting!

lynchaj, 2022-05-12:
Hi Andras
Would you please set the default -oit back to zero like the older versions were? Then if someone specifies a value it would use it. When I use the .jar file it doesn't pay attention to the -oit command line options so there is no way to set it. This means I am restarting FreeRouting many times to get it to fully optimize a PCB.

Alternatively, add a menu option to specify the oit value as a parameter from within the program GUI

Thanks, Andrew Lynch

andrasfuchs, 2022-05-13:
I'm hesitant to set `-oit` to `0` by default, because that would mean that freerouting would run indefinitely by default. This would prevent users to use it in automated scripting environments.

What do you mean that `-oit` is not set when you use it as a .jar? It definitely should work! What operating system are you using and what is your exact command line? I tested it on Windows and it works for me.

Just one another thought about the default value of `oit`. It would be great if others like you could also run some test, because according to my tests, the majority of the improvements by the route optimizer is done in the first 3 minutes. Running it for an extra 10 minutes does reduce the number of vias a little more, but that's less significant and this efficiency degrades greatly over time. I'm not convinced at the moment that it even matters really if we have 5% less vias on a board or the wire length is 10% less, to be honest.

I'm open to be convinced otherwise with real-world data, so please, don't hesitate to share your experience!

andrew64k, 2022-05-13:
I agree that 0 is not a good choice as people will not understand why it never finishes. -oit 0.001 does take a little more time to run but seems to provide a good result. With the default of 0.1 the output just seems to be unfinished.

Adding the option to the GUI would be nice and user friendly as it is an important function. May be a slider or some buttons along with the actual number...quick(.1)...short(.01)...standard/default(.001)...improved(.0001)....long(.00001)....other(manual)

On my current circuit (about 500 routes) the reduction from 150+ vias to <110 is ok (0.1) but only provides basic clean up. Using .01 does not take long and gets down to <80 which shows a big improvement in routing quality. Using .001 gets down to 75 adds more time but not too bad. Running .0001 provides more improvement (<70 vias) and takes a little longer but seems to have a more consistent result between runs. At this point (and beyond) I don't like some of the choices and revert some odd traces created to avoid vias.

So, if I had to pick a default that most people would be happy with, I would pick 0.001. It seems to provide decent output in a reasonable time (depending on CPU cores/speed). At worst (fastest) 0.01 would be ok. Personally I'll run it a little longer and then change stuff I don't like.

lynchaj, 2022-05-13:
Hi
FreeRouting will eventually finish with a setting of 0. It may take a while but for large and complicated boards with a lot of components it needs time to solve and optimize. As it is now it quits too soon and I cannot get the command line parameters to work. I am using Win10, 32-bit with the .jar file. I placed the -oit parameters in the shortcut and they seem to be ignored. I think the best solution would be a menu option so I can type in a value

Thanks, Andrew Lynch

andrasfuchs, 2022-05-13:
Sounds good to me, thank you @andrew64k for the testing and feedback.

@lynchaj I used the following command line with success while using [the latest snapshot .jar](https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT):
`java -jar freerouting-SNAPSHOT-20220513_222400.jar -de Mars-64.dsn -do Mars-64.ses -oit 0`

(Please reply to the other, more related issue #102)

closed 2022-06-06 by andrasfuchs