freerouting/freerouting #110: Non-ascii synbils in .dsn file

BOND12, 2022-04-22:
Greetings. I found a problem when trying to use freerouting with a .dsn file that contains non-ascii characters in its content. When I try to use freerouting with such a file, I get the following error:
[image]

log:
`2022-04-22 13:16:49.965 [main] INFO  Freerouting v1.5.0 (build-date: 2022-01-27)
2022-04-22 13:16:50.002 [main] INFO  Opening './testPCBSpecctraFile.dsn'...
WARNING: sun.reflect.Reflection.getCallerClass is not supported. This will impact performance.
2022-04-22 13:17:39.407 [main] ERROR Illegal character 'м' was found at position 4433
java.lang.Error: Illegal character 'м' was found at position 4433
        at app.freerouting.designforms.specctra.SpecctraFileScanner.next_token(SpecctraFileScanner.java:1404) ~[freerouting-1.5.0.jar:unspecified]
        at app.freerouting.designforms.specctra.ScopeKeyword.skip_scope(ScopeKeyword.java:26) ~[freerouting-1.5.0.jar:unspecified]
        at app.freerouting.designforms.specctra.Component.read_place_scope(Component.java:333) ~[freerouting-1.5.0.jar:unspecified]
        at app.freerouting.designforms.specctra.Component.read_scope(Component.java:59) ~[freerouting-1.5.0.jar:unspecified]
        at app.freerouting.designforms.specctra.Component.read_scope(Component.java:25) ~[freerouting-1.5.0.jar:unspecified]
        at app.freerouting.designforms.specctra.ScopeKeyword.read_scope(ScopeKeyword.java:85) ~[freerouting-1.5.0.jar:unspecified]
        at app.freerouting.designforms.specctra.ScopeKeyword.read_scope(ScopeKeyword.java:85) ~[freerouting-1.5.0.jar:unspecified]
        at app.freerouting.designforms.specctra.DsnFile.read(DsnFile.java:64) ~[freerouting-1.5.0.jar:unspecified]
        at app.freerouting.interactive.BoardHandling.import_design(BoardHandling.java:1030) ~[freerouting-1.5.0.jar:unspecified]
        at app.freerouting.gui.BoardFrame.read(BoardFrame.java:158) ~[freerouting-1.5.0.jar:unspecified]
        at app.freerouting.gui.MainApplication.create_board_frame(MainApplication.java:362) ~[freerouting-1.5.0.jar:unspecified]
        at app.freerouting.gui.MainApplication.main(MainApplication.java:86) ~[freerouting-1.5.0.jar:unspecified]
^Z
`
These characters from the kicad project get there, where I describe some things in Russian, for example, the denominations of components (units of measurements are written in Russian characters (in projects, this is a completely normal situation.)). I observed a similar situation if I use them in the names of the circuits. I note right away that this is completely acceptable for the kicad project. But when trying to use freerouting, problems are observed. So far, I've found a way around this problem by writing a program that will transliterate all Russian characters in a .dsn file into ascii characters. This way of using freerouting is quite possible at this time, although it requires additional steps.
Are there ways to fix this shortcoming in future versions of freerouting?
.dsn file (boards from an electrical point of view may be incorrect, but exported to .dsn is correct):
[dsn_files.zip](https://github.com/freerouting/freerouting/files/8540032/dsn_files.zip)

freerouting 1.5.0
OS Linux (Kubuntu 21.10), Windows 7, Windows 10.

BOND12, 2022-04-22:
I saw [message from [erichVK5](https://github.com/erichVK5)](https://github.com/freerouting/freerouting/issues/93#issuecomment-1079698329) about the proposed solution to this problem. I strongly discourage doing this, because if there are 2 chains in a .dsn file with a name containing the same number of non-ascii characters of the same length, freerouting will probably consider this as one chain, since they will be replaced with X , which will lead to trace errors.

BOND12, 2022-04-22:
As I expected, it does not work correctly if different names with the same number of characters are assigned to the same chain:
[]([image])

log:
`
2022-04-22 17:13:48.368 [main] INFO  Freerouting v1.5.1-SNAPSHOT (build-date: 2022-04-22)
2022-04-22 17:13:48.394 [main] INFO  Opening './testProjectFromFreeroutingBugTest.dsn'...
2022-04-22 17:13:48.865 [main] WARN  Illegal character 'ц' found at position 6609 replaced with 'X'.
2022-04-22 17:13:48.867 [main] WARN  Illegal character 'е' found at position 6610 replaced with 'X'.
2022-04-22 17:13:48.867 [main] WARN  Illegal character 'п' found at position 6611 replaced with 'X'.
2022-04-22 17:13:48.868 [main] WARN  Illegal character 'ь' found at position 6612 replaced with 'X'.
2022-04-22 17:13:48.870 [main] WARN  Illegal character 'ц' found at position 6654 replaced with 'X'.
2022-04-22 17:13:48.871 [main] WARN  Illegal character 'е' found at position 6655 replaced with 'X'.
2022-04-22 17:13:48.871 [main] WARN  Illegal character 'л' found at position 6656 replaced with 'X'.
2022-04-22 17:13:48.872 [main] WARN  Illegal character 'ь' found at position 6657 replaced with 'X'.
2022-04-22 17:13:48.873 [main] WARN  Illegal character 'ц' found at position 6927 replaced with 'X'.
2022-04-22 17:13:48.873 [main] WARN  Illegal character 'е' found at position 6928 replaced with 'X'.
2022-04-22 17:13:48.874 [main] WARN  Illegal character 'п' found at position 6929 replaced with 'X'.
2022-04-22 17:13:48.874 [main] WARN  Illegal character 'ь' found at position 6930 replaced with 'X'.
2022-04-22 17:13:48.875 [main] WARN  Illegal character 'ц' found at position 6973 replaced with 'X'.
2022-04-22 17:13:48.875 [main] WARN  Illegal character 'е' found at position 6974 replaced with 'X'.
2022-04-22 17:13:48.876 [main] WARN  Illegal character 'п' found at position 6975 replaced with 'X'.
2022-04-22 17:13:48.876 [main] WARN  Illegal character 'ь' found at position 6976 replaced with 'X'.
2022-04-22 17:13:49.476 [main] INFO  Opening 'testProjectFromFreeroutingBugTest.rules'...
2022-04-22 17:13:49.487 [main] ERROR File 'testProjectFromFreeroutingBugTest.rules' was not found.
2022-04-22 17:14:15.786 [Thread-0] INFO  Starting auto-routing...
2022-04-22 17:14:16.116 [Thread-0] INFO  Auto-routing was completed in 0,33 seconds.
2022-04-22 17:14:16.118 [Thread-0] INFO  Starting route optimization on 3 threads...
2022-04-22 17:14:16.339 [Thread-0] INFO  Route optimization was completed in 0,21 seconds.
`
Attached is the Kicad project that caused the error.
[testProjectFromFreeroutingBugTest.zip](https://github.com/freerouting/freerouting/files/8541704/testProjectFromFreeroutingBugTest.zip)

I have an idea how to fix this. I'll try to do it, if it works, I'll let you know.

BOND12, 2022-04-22:
[https://github.com/BOND12/freerouting/commit/3683ad3ec84d7bf983604c5362db6717f93a6269](https://github.com/BOND12/freerouting/commit/3683ad3ec84d7bf983604c5362db6717f93a6269l) - this commit had to fix this issue. Please, test it and comment.

andrasfuchs, 2022-04-22:
Thank you! 
Please create a pull request with your two commits. I (and others) can only test and comment that way.

erichVK5, 2022-04-22:
The specctra language reference says "The design file is a text file."

It was on that basis I treated non ASCII in the path at the top of the DSN file as something of an exception, and something that was not going to occur in the rest of the design file.

There is also an ASCII special character descriptor in the specctra standard, which implies only ASCII is assumed.

If some layout tools will be emitting non compliant DSN files, then something better than my patch is clearly required.

BOND12, 2022-04-22:
Files with non-ASCII characters can be obtained not only from kicad. They are also generated by P CAD for example (which is used at my work). So it won't help completely. Now I am trying to modify freerouting so that when uploading a file.dsn he checked all the characters and if they are > 127 then substituted the string illegalCharactersFound in their place. I think it will work. Transliteration performs approximately the same actions. I partially managed to do this, but it doesn't work fine for all files right now. While trying to fix it.

erichVK5, 2022-04-22:
maybe substituting the unicode for the offending non ASCII, i.e. something like "U+2160", would effect a fairly unique, and invertible, mapping?

BOND12, 2022-04-22:
PCAD generates the dsn file is encoded in windows-1251, so this is also not a complete solution.

erichVK5, 2022-04-22:
a standalone pre-processor utility may be the simplest option, which finds and replaces non ascii in the network net labels etc. The returned topology in the .ses file should be unaffected.

BOND12, 2022-04-22:
[https://github.com/BOND12/freerouting/commit/f970a5fc1e11ff233f5728c26ab195840c9ad8d2](https://github.com/BOND12/freerouting/commit/f970a5fc1e11ff233f5728c26ab195840c9ad8d2)
I have +- works in this version. I made it so that I substituted the ilch + replacement number for the lines. If the same character comes across, then the string value that was used earlier is taken.

BOND12, 2022-04-22:
This works, and successfully, but requires additional actions when used.

BOND12, 2022-04-22:
What do you say about my modification at all?

erichVK5, 2022-04-22:
You could still get collisions with a pair of nets something like

"NetilCh1"

and

"NetX"  where X is an illegal character

but it would be much less likely than with my solution

BOND12, 2022-04-22:
Well, what I wrote converts the string "xzx", for example, to the string "ilCh0ilCh1IlCh0" if x and z are invalid characters. Here with your modification, such a problem really was, I tested it.

erichVK5, 2022-04-22:
maybe the simplest thing is to more deliberately convert the entire offending module name, padstack name, pin-name, or net name, to an actual hash.

at what point is this better done outside of the autorouter application?

BOND12, 2022-04-22:
I manipulated the dsn file after exporting it from CAD. If you use another program for this operation, then at this stage you should do it.

BOND12, 2022-04-22:
I have created a merge request. If you don't mind my version of the solution to this problem, of course.

andrasfuchs, 2022-04-22:
Sorry for my lack of knowledge regarding this, but could you help me understand the depth of the problem?

I know that we have an old, poorly designed DSN reader logic, and I suppose that our SES writer is similar, so I'm not questioning if the problem you describe is legit. It is.

I have a few assumptions, could you help me confirm or correct them?
* (Unfortunately) we read DSN files as binary files instead of text files
* This means that we don't handle UTF-8 files, and we suppose that the file has one kind of ASCII encoding, but in reality it can be more than one (like GB2312 or Windows-1251) if it has special characters (like Chinese or Cyrillic)
* Since it's ASCII, there is no way to detect automatically which encoding the file uses unless the user defines it manually
* The "invalid" character strings can be part of a file path, component, via or net names stored in the DSN and ideally we would try to keep them as they are, but we can't because our DSN reader throws an error if the character's code is >127
* You guys implemented two workarounds: both of them replace the >127 characters to some other string which solves the DSN reader's problem, but it generates a SES file in the end with funny names where those >127 characters are replaced with something else

If these are true then I think we are trying to solve a thing that is already solved in general by using UTF-8.

So if the modern PCB editors can export DSN and import SES as UTF-8, then we should consider rewriting the DSN reader and SES writer classes to use UTF-8 to handle all kinds of special characters. It's a bigger piece of work, of course, but a long term solution instead of a quick fix.

As a quick workaround I think we should rework only the "special character detector" and we should keep the special characters and let them travel from the DSN to the SES untouched. This solution would require the user to be aware of the fact that the DSN and SES files must have the same encoding, but on the other hand the user would see the names in the final SES files as they were originally in the DSN (if they set the right encoding, of course).

What do you guys think?

erichVK5, 2022-04-23:
I thought about it a bit more.

The Specctra DSN spec is for ASCII.

The parser very reasonably assumes this and parses the ASCII text (not binary) DSN file char by char looking for tokens.

The simplest solution would be a pre-processor enabled at the command line, i.e. -na "non-ascii", and some code:

 parses the ASCII DSN line by line
 splits each line with whitespace
 checks each string for non ASCII
 replaces any non ASCII string with a hash, if not previously found in a lookup
 saves the string/hash lookup
 passes on a reassembled, now ASCII text DSN to the DSN parser

this takes care of non-ascii paths, nets, padstack names, and footprint names in one fell swoop.

everything else will be a DSN keyword, number, or paranthesis

a slightly cleverer way to do it would be reading it in like the s-expression tree that it is, and traverse the tokens, then reassemble a flat output file as text

you may have arrived at the same sort of solution, I will check it out

erichVK5, 2022-04-23:
Here's some code that converts either of the two example DSN files provided by BOND12, turning contiguous non ASCII into a String.hashCode() ASCII segment

This works pretty well, as far as I can tell. If it were to be incorporated into freerouting, it may be possible to interpose it between the p_input_stream

in

app.freerouting.designforms.specctra.DsnFile 

in

public static ReadResult read(java.io.InputStream p_input_stream, app.freerouting.interactive.IBoardHandling p_board_handling, app.freerouting.board.BoardObservers p_observers, app.freerouting.datastructures.IdNoGenerator p_item_id_no_generator, TestLevel p_test_level) 

before passing a new InputStream built from the text output of ASCIIfying code to

Scanner scanner = new SpecctraFileScanner(p_input_stream);

i.e. this would sanitise non ASCII with low risk of hash collisions before it gets to the DSN parser.

========================

import java.io.File;
import java.util.Scanner;
import java.util.ArrayList;
import java.lang.StringBuilder;

/**
 * Class for ASCIIfying malformed dsn-files that contains non-ASCII.
 */
public class ASCIIfy
{

// to check if a string consists of US-ASCII only
    private static boolean stringNotASCII(String input) {
        for (int i = 0; i < input.length(); i++) {
            int c = input.charAt(i);
            if (c > 0x7F) {
                return true;
            }
        }
        return false;
    }

    private static String ASCIIfyString(String input) {
        StringBuilder newString = new StringBuilder("");
        StringBuilder nonASCII = new StringBuilder("");
        boolean ascii = true;
        for (int i = 0; i < input.length(); i++) {
            char c = input.charAt(i);
            int val = c;
            if (val > 0x7F) {
                if (ascii == true) {
                    ascii = false;
                    nonASCII = new StringBuilder("X"); 
                }
                nonASCII.append(c);
            } else {
                if (ascii == false) {
                    ascii = true;
                    newString.append(nonASCII.toString().hashCode());
                    nonASCII = new StringBuilder("");
                }
                newString.append(c);
            }
        }
        if (ascii == false) {
            newString.append(nonASCII.toString().hashCode());
        }
        return newString.toString();
    }

    public static void main(String[] args)
    {
        //String fileName = "testProjectFromFreeroutingBugTest.dsn";
        String fileName = "RelayModule.dsn";
        try {
            File file = new File(fileName);
            Scanner scanner = new Scanner(file);
            scanner.useDelimiter(System.getProperty("line.separator"));
            while (scanner.hasNext()) {
                String currentLine = scanner.next();
                if (stringNotASCII(currentLine)) {
                    // System.out.println(currentLine); // to display visual diff
                    System.out.println(ASCIIfyString(currentLine));
                } else {
                    System.out.println(currentLine);
                }
            }
            scanner.close();
        } catch (Exception e) {
            e.printStackTrace();
        }
    }
}

BOND12, 2022-04-23:
I checked your program, everything works fine with it. Tracks are imported back into the kickad without errors. In general, your program gives a similar effect as mine, which I used to work with freerouting before (now I have built a similar functionality into freerouting).

In general, this technique will work in practice and works. But there will still be restrictions, for example, if non-ASCII characters fall into the designation of components, this can create problems when importing the .ses file back into CAD. Such projects are not usually found in practice, but, nevertheless, such a project can exist in nature and from the point of view of Kicad, it will have all the rights to life (I will attach an example so that you can make sure of this (there will be problems when importing a .ses file into it when using this method of character transformation kicad, tracing in freerouting is successful at the same time)).
[testProjectFromFreeroutingBugTest1.zip](https://github.com/freerouting/freerouting/files/8547640/testProjectFromFreeroutingBugTest1.zip)

The ideal solution would be to remove the restriction on the character set in the .dsn file with the algorithm of the program itself, and not to convert non-ASCII characters into something else. Moreover, without being tied to any encoding, but to sort them in the encoding of the operating system used, because encoding .the dsn of the file will depend on this.

But as a temporary solution, substituting something in place of non-ASCII characters will work.

erichVK5, 2022-04-23:
The freerouter DSN parser works 100% correctly for Specctra standards compliant DSN files.

Changing the existing, fit for purpose parser code to support non Specctra standards compliant DSN files would require quite a bit of work from developers with limited time.

Just because some EDA packages generate non Specctra standards compliant DSN files does not mean that significant resources should go into accommodating broken exports.

With not much extra effort, the code example above can save a substitution list, which can then be used to de-ASCIIfy the .ses file on the return trip to the EDA tool.

A standalone utility seems the simplest way to do this.

Interposing the same mechanism at the file load stage would also work, but this creates additional complexity in freerouter, especially if de-ASCIIfying is needed on saving, and the whole point of unix like environments is the ability to pipe inputs and outputs through other programs, such as a standalone sanitising utility. Freerouting is quite capable of being fed inputs via the command line, making a scripted solution with a standalone utility pretty low effort for someone using an EDA exporting non Specctra standards compliant DSN files.

andrasfuchs, 2022-05-12:
Hey guys, I made a fix for the identifier with non-ASCII characters bug. 
Could you download the latest snapshot release, and test it? I had only a limited number of files to try. 
Thank you!

BOND12, 2022-05-13:
My tests were successful.  At least I didn't notice any problems.

andrasfuchs, 2022-05-13:
That's good to hear, thank you @BOND12!

closed 2022-06-06 by andrasfuchs