freerouting/freerouting #117: Some issues building it on Ubuntu 22.04.

leoheck, 2022-06-07:
Have you tried to build it on Ubuntu 22.04 already?
I installed `openjdk-17-jre`

[image]

Please, if you need more info I am glad to share, but the reports with special flags are pretty big to put here I guess...

Here a plaintext
```bash
➜  freerouting git:(master) ./gradlew assemble
> Task :javadoc FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':javadoc'.
> Javadoc generation failed. Generated Javadoc options file (useful for troubleshooting): '/home/lheck/Dropbox/Documents4/freerouting/build/tmp/javadoc/javadoc.options'

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1s
6 actionable tasks: 4 executed, 2 up-to-date
➜  freerouting git:(master) ./gradlew assemble
> Task :javadoc FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':javadoc'.
> Javadoc generation failed. Generated Javadoc options file (useful for troubleshooting): '/home/lheck/Dropbox/Documents4/freerouting/build/tmp/javadoc/javadoc.options'

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 994ms

```

andrasfuchs, 2022-06-07:
I made a successful compilation on GitHub using `ubuntu-latest` which is Ubuntu 20.04 according [to the documentation](https://github.com/actions/virtual-environments).

[image]

Could you dig a little deeper and get a more detailed error message from the javadoc task? I'm not sure what could cause problems there.

I use `Java_Temurin-Hotspot_jdk/17.0.3-7/x64` as the JRE, that might be the issue?

szaffarano, 2022-06-07:
Hi @leoheck ,

`openjdk-17-jre` doesn't include `javac`, please try installing the whole `jdk`, i.e.:

```shell
$ sudo apt-get install openjdk-17-jdk
```

It should work.

leoheck, 2022-06-07:
Thanks, guys. Indeed, the JDK was missing.

To summarize, on Ubuntu 2022.04, this is the only dependency `sudo apt-get install openjdk-17-jdk` since JRE is in this package I suppose.

I still got a warning but it is good to go.

```bash
➜  freerouting git:(master) ./gradlew assemble       
Starting a Gradle Daemon, 1 incompatible and 1 stopped Daemons could not be reused, use --status for details

> Task :compileJava
/home/lheck/Dropbox/Documents4/freerouting/src/main/java/app/freerouting/gui/WindowNetSamples.java:106: warning: [removal] AccessControlException in java.security has been deprecated and marked for removal
        catch (java.security.AccessControlException e)
                            ^
1 warning

BUILD SUCCESSFUL in 10s
```