freerouting/freerouting #123: Components and traces are not bound to each other when exporting to easyEDA

michiel2005, 2022-06-23:
I'm using V1.6.1 of freerouting and V 6.5.5 of easyEDA
I'm running it on windows11 64 bit system x86 
[dsn and ses file.zip](https://github.com/freerouting/freerouting/files/8966799/dsn.and.ses.file.zip)
No use of command line arguments, only gui use
[image]
[image]
Load the ses file in easy eda and see that the routes are not bound to components.
I would expect that the parts are bound to the lines as it does with the newest version from freerouting.org.
Without this being fixed the freerouting is useless for me.

I can't find where the .log file is located

denjhang, 2022-06-24:
You can try the oldest freerouting version (V1.4.4), maybe EasyEDA is compatible. But even the oldest version of freerouting still has some network errors, mainly because the wire name is inexplicably enclosed in double quotes.

michiel2005, 2022-06-24:
Yes, after manually removing the quotation marks it binds with the components, I will report this to easyEDA.

denjhang, 2022-06-25:
I've detailed the problem to the EasyEDA developers, but they claim it's not their fault. Because using other auto-routing software, such as konekt electra 6.56 there is no problem.

andrasfuchs, 2022-06-25:
Thank you for your proactivity! Their reasoning still doesn't prove that
they follow the specification when they read our routed files.

I'll dive deeper into the official .ses file specification to double check
if we use the quotation marks correctly, and I will do some tests with
KiCad to see if it also has a problem with the most recent version of
freerouting.

Hopefully I can get back to you in two weeks.

andrasfuchs, 2022-07-01:
Here is the Specctra file specification document: https://cdn.hackaday.io/files/1666717130852064/specctra.pdf

After reading it through I would say that we are doing it right in freerouting, and they should be able to parse our output file, but I also see the reason why someone would say otherwise. I would need your opinion, if you have the time to investigate.

So, both input .DSN and output .SES file follow a similar structure and they are both part of the Specctra spec.

One important difference is, that the DSN file can have a parser configuration section at the beginning of it, that tells the parser what to expect. It has a property called `string_quote` and it is usually defined by the EDA software that created the .DSN file. It is usually the `"` character.

[image]

[image]

With this configuration it is obvious how to parse the input .DSN file regarding the names of the components: if they have the quote characters around them, that usually indicates that they have a space or special characters, so they should be parsed accordingly.

The problem is that the spec don't have this section for the output .SES files, so we can't tell the EDA software how to parse our files. They can suppose of course that that quotes are used and they are represented by the `"` character, but it is not possible to implicitly tell them that. That causes the confusion.

The whole thing is problematic, because component names can have spaces and special characters and in those cases we must use quotes in the input files. But how should we save them in the output .SES file? We save them now as quoted strings regardless if they have special characters or spaces, that is our safest bet.

We can argue that the <string> in the spec can have quotes anywhere and the parser must be prepared for this case, but since the spec allows that the quotes to be expressed with the `"`, `'` or `$` characters, it's not trivial.

[image]

[image]

[image]

[image]

[image]

What the EasyEDA developers could do if they wanted to have a robust parser, is that they could check for all the 3 types of quote characters when they parse a string in the .SES file. 
I suppose they don't check for any of them at the moment and that causes the problem, because we use them.

What do you think?

denjhang, 2022-07-01:
Through communication I found that EasyEDA developers seem to be lazy, they are not willing to actively fix bugs or make improvements.
Actually EasyEDA doesn't support any special symbols at all, and many other features are incomplete. My suggestion is that freerouting can provide a version specifically for EasyEDA users to be compatible

michiel2005, 2022-07-01:
Hi, thanks for your investigation.

I think it is done right in freerouting, but since easyEDA is not willing to fix this I should say there comes a option: "wrap output nets in double quotes".
In easyEDA it is not possible to have a space in the net name, so spaces won't be causing problems.
It is however possible to use: ", ' and $ in net names in easyEDA, so if they parse that away it could make problems for other users wo use that in their net names.

andrasfuchs, 2022-07-01:
Thanks for your feedback!

Alright, I will not expect the easyEDA developers to fix this then...

I would say that having an specific easyEDA option sounds like a workable solution, but I would prefer a way that is not EDA-specific, because that makes both the code and the user experience more complicated, and there is more chance to make an error.

What if I add an extra check before we save the component/net/etc. names into the .SES file, and only use quotes if they are justified, because the name has a space or special character?

It would mean that if the components didn't have any spaces or special characters to begin with, then they will not have quote characters in the output, so all is good. Hopefully that would cover the majority of the cases.
If they have, then they will not work with easyEDA, but that's the price we have to pay because of their faulty parser.

With this solution we would keep the ability to have spaces, special characters like Chinese and Cyrillic characters in the names, and work with KiCad and other EDAs.

Would it be an acceptable solution for you?

(Please note that I need to focus on other things too, and I still have a pull request here that awaits for my review, so don't expect this to be done too soon, but I will try to fix this in ~2 weeks.)

michiel2005, 2022-07-01:
That would be a great solution, that will at least always work with easyEDA.

But doesn't that introduce the problem for other eda parsers that some nets does have double quotes and some does not?
I would say to first check if there is a space in any net, and than wrap all net's in double quotes or not all of it in double quotes.
Or was that what you meant?
I don't know if it is possible.
Maybe it could also be a extra feature to control this with a command line argument that overwrites the check for special characters for the more advanced users.

andrasfuchs, 2022-07-01:
I've seen .DSN files where the quotes/not quotes were mixed, but I'm not sure if it can cause problems in .SES files for the EDAs.

I like your idea of checking all names first, and decide to use or not use quotes consistently for all names. I'll look into it, when I get there.

I have one issue with the command line argument: what if the names have spaces or non-ASCII characters and the command line argument specifies that the user doesn't want to use quotes? That would generate a non-standard output file that is not valid according to the spec, and it would most probably fail to load. Unless we remove the spaces as an extra step, but that would cause other issues...

andrasfuchs, 2022-07-05:
Hey guys, I created a fix that theoretically should work, but I can't test it properly, because I don't have easyEDA. 
Could you give it a go?

The .jar and binaries are [in the SNAPSHOT section of the releases page](https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT).

michiel2005, 2022-07-05:
I can't do it now because I will be on holiday  but as soo  as it is possible I will test it.

denjhang, 2022-07-05:
I'll test it, I have time.
Actually I use Chinese, but I never use Chinese in net, I only use English or numbers.

denjhang, 2022-07-05:
https://s3.bmp.ovh/imgs/2022/07/05/6af95ddcdfa64993.jpeg

denjhang, 2022-07-05:
[image]

denjhang, 2022-07-05:
At present, it seems that even importing DSN files cannot work properly.

andrasfuchs, 2022-07-05:
Thank you @denjhang for the test! 
Could you try the `-l en` command line argument? That should be fixed now.

I'm going to fix the exception on your screenshot in a few minutes.

denjhang, 2022-07-06:
DSN can now be imported normally.
I remember that the old version of freerouting had a circular cursor when auto-routing and optimizing vias. I thought that was very nice, but I don't know why it should be canceled?
It would be nice to have some visual cues when routing or optimizing vias. Without proper visual cues, I often think freerouting is stuck.

denjhang, 2022-07-06:
It still looks like there are annoying double quotes. This caused EasyEDA to think that the wiring was not done and there were a lot of DRC errors.
[image]

denjhang, 2022-07-06:
I never use double quotes when naming components or nets, just English characters (usually all caps) and numbers and spaces.

denjhang, 2022-07-07:
Currently it appears that only some nets have double quotes. This goes back to the case with V1.4.4.

closed 2022-07-13 by andrasfuchs

EasyEDA2021, 2022-07-14:
hi
we fixed this issue, compatible with this change at EasyEDA v6.5.8 @andrasfuchs @denjhang @michiel2005 @miho 
thank you

andrasfuchs, 2022-07-15:
That's great to hear, thank you @EasyEDA2021 for letting us know!