freerouting/freerouting #126: Merge KiCad plugin into this repo

andrasfuchs, 2022-07-04:
Although I created [a pull request a few months ago](https://github.com/random-builder/kicad_freerouting-plugin/pull/12), it looks like the owner of the [kicad_freerouting-plugin repo](https://github.com/random-builder/kicad_freerouting-plugin), @Andrei-Pozolotin doesn't maintain that project any more. 

It would be great to have its source in our repo instead, so that we could keep it up-to-date and make sure that it works with the latest KiCad and freerouting.

nikitalita, 2022-07-22:
I just updated my fork of the plugin to fix a lot of bugs and update to 1.6.2, you may find this useful: https://github.com/nikitalita/kicad_freerouting-plugin

andrasfuchs, 2022-07-27:
Excellent stuff, thank you @nikitalita! 👍 

I'll refer to your repo from now on in [our integrations/KiCad folder](https://github.com/freerouting/freerouting/tree/master/integrations/KiCad).

nikitalita, 2022-07-27:
I, uh, don’t really intend to maintain it going forward, I just had to update it to fix things. Its probably not a good idea to link to me.

andrasfuchs, 2022-07-27:
Alright, no worries, I'll integrate it then into this repo.

closed 2022-07-27 by andrasfuchs